kizniche/Mycodo #200: crontab configuration attempts to run from the wrong directory

lucasvickers, 2017-03-14:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0-dev
 
#### Problem Description
Current 5.0-dev branch produces the following log error during install:

```
~/Mycodo/install $ sudo ./setup.sh
...
#### Updating crontab entry
/bin/bash: /home/nibbler/install/crontab.sh: No such file or directory
/bin/bash: /home/nibbler/install/crontab.sh: No such file or directory
```

I was able to fix this by changing L86 and L87 in [upgrade_commands.sh](https://github.com/kizniche/Mycodo/blob/dev-5.0/mycodo/scripts/upgrade_commands.sh#L86-L87) to:
```
/bin/bash ${INSTALL_DIRECTORY}/Mycodo/install/crontab.sh mycodo --remove
/bin/bash ${INSTALL_DIRECTORY}/Mycodo/install/crontab.sh mycodo
```

kizniche, 2017-03-14:
Thanks. I haven't tried an install of 5.0, so I haven't caught any of these bugs yet. You're a pioneer!

closed 2017-03-14 by kizniche

lucasvickers, 2017-03-14:
happy to help :)

commit 0de5a455 (2017-03-14): Fix #200