kizniche/Mycodo #202: AttributeError: 'NoneType' object has no attribute 'stats_opt_out'

lucasvickers, 2017-03-14:
## Mycodo Issue Report:
  - 5.0-dev
 
#### Problem Description
After install running the daemon yields:
```
Mar 14 14:39:23 nibbler mycodo_daemon.py[27608]: File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 847, in <module>
Mar 14 14:39:23 nibbler mycodo_daemon.py[27608]: daemon_controller = DaemonController()
Mar 14 14:39:23 nibbler mycodo_daemon.py[27608]: File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 296, in __init__
Mar 14 14:39:23 nibbler mycodo_daemon.py[27608]: self.refresh_daemon_misc_settings()
Mar 14 14:39:23 nibbler mycodo_daemon.py[27608]: File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 521, in refresh_daemon_misc_settings
Mar 14 14:39:23 nibbler mycodo_daemon.py[27608]: self.opt_out_statistics = misc.stats_opt_out
Mar 14 14:39:23 nibbler mycodo_daemon.py[27608]: AttributeError: 'NoneType' object has no attribute 'stats_opt_out'
Mar 14 14:39:23 nibbler systemd[1]: mycodo.service: control process exited, code=exited status=1
Mar 14 14:39:23 nibbler systemd[1]: Failed to start Mycodo server.
Mar 14 14:39:23 nibbler systemd[1]: Unit mycodo.service entered failed state.
```

kizniche, 2017-03-14:
Can you open your Mycodo/databases/mycodo.db with a SQL editor (I use SQLiteStudio, free) and see if you have an entry in your Misc table?

lucasvickers, 2017-03-14:
Will do.  Just a heads up I renamed my `Mycodo` to `Mycodo-4` and have been installing 5.0-dev as if it was from scratch.  I believe you keep all the databases local to the `Mycodo` folder so I don't think this should cause issues.

kizniche, 2017-03-14:
It will look to what directory the files are executed from, so in that care your Install directory would be detected as Mycodo-4... I think I found the issue. Commit fix incoming.

lucasvickers, 2017-03-14:
All of the executions I'm manually starting up should happen in `Mycodo` which is the 5.0-dev folder.  Are you saying that the sqlite is in a different folder?

Either way, keeping things self contained to an install folder is ideal, that way we can start from scratch or bounce between versions.  On that note, you should probably move to virtualenv at some point, I'll open up a feature request.

Once you have the fix in place let me know and I'll try out.  thanks!

kizniche, 2017-03-14:
Whatever you name the folder, it will set as the install folder. So, if you name it Mycodo-4, it will detect that as the parent directory (install folder). So your databases it will read/write from/to will be Mycodo-4/databases. Testing a fix (uncommited) now, and will push if successful.

lucasvickers, 2017-03-14:
I did the following
- Install Mycodo4 in `Mycodo`
- Rename `Mycodo` to `Mycodo-4`
- Checkout 5.0-dev to `Mycodo`
- Install 5.0-dev from within `Mycodo`
- Run from within `Mycodo`

So shouldn't it be properly looking for things in `Mycodo`?

kizniche, 2017-03-14:
The [setup.sh script](https://github.com/kizniche/Mycodo/blob/04e60ad199e0f24c3426ff82010990252b0e38d0/install/setup.sh#L8) will look to what the parent directory is to determine the install root:

```INSTALL_DIRECTORY=$( cd "$( dirname "${BASH_SOURCE[0]}" )/.." && pwd -P )```

Same with the other scripts. Does this answer the question. I'm not exactly following the directory changes/install setup you're performing.

kizniche, 2017-03-14:
Found the issue, working on a fix.

lucasvickers, 2017-03-14:
I'm basically just saying I don't use upgrade, I just start from scratch each time.

If you are keeping anything outside of the `Mycodo` folder then it will remain persistent, but I think you keep everything (databases, etc) inside of that folder.

kizniche, 2017-03-14:
Yes, that is true. The logs (/var/logs/mycodo) are the only thing separate from the mycodo install directory.

lucasvickers, 2017-03-14:
Then, I just confused you, and the issue shouldn't have to do what I did.  Once you get your fix in place I'll test!

kizniche, 2017-03-14:
The issue is it's not creating the default entries in some tables when the database is created. This was due to a change a few days ago. I'm tracking down what needs to be changed to fix it. I should find it shortly.

kizniche, 2017-03-14:
That last commit should fix the database issue and get you moving on to the next issue 😆

lucasvickers, 2017-03-14:
Getting closer :)

```
Mar 14 16:38:16 nibbler mycodo_daemon.py[31090]: File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/util/compat.py", line 202, in raise_from_cause
Mar 14 16:38:16 nibbler mycodo_daemon.py[31090]: reraise(type(exception), exception, tb=exc_tb, cause=cause)
Mar 14 16:38:16 nibbler mycodo_daemon.py[31090]: File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1139, in _execute_context
Mar 14 16:38:16 nibbler mycodo_daemon.py[31090]: context)
Mar 14 16:38:16 nibbler mycodo_daemon.py[31090]: File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/default.py", line 450, in do_execute
Mar 14 16:38:16 nibbler mycodo_daemon.py[31090]: cursor.execute(statement, parameters)
Mar 14 16:38:16 nibbler mycodo_daemon.py[31090]: sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such table: misc [SQL: u'SELECT ...c_hide_a
Mar 14 16:38:16 nibbler systemd[1]: mycodo.service: control process exited, code=exited status=1
Mar 14 16:38:16 nibbler systemd[1]: Failed to start Mycodo server.
Mar 14 16:38:16 nibbler systemd[1]: Unit mycodo.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.
```

kizniche, 2017-03-14:
hmm. You should be having the database populate the correct tables with the latest updates. Try deleting your databases/mycodo.db file and restarting the web service "touch Mycodo/mycodo_flask.wsgi"

kizniche, 2017-03-14:
Can you paste the whole traceback so I can see where misc is being called from?

kizniche, 2017-03-14:
I'm going to try to get another Pi up and attempt a fresh install of 5.0-dev

lucasvickers, 2017-03-14:
Hang on! seems like it installed.  Give me a few to test.

closed 2017-03-14 by kizniche