kizniche/Mycodo #204: If mycodo.service already exists, install fails

lucasvickers, 2017-03-14:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0-dev, but noticed it in other versions
 
#### Problem Description
https://github.com/kizniche/Mycodo/blob/dev-5.0/install/setup.sh#L84

This always fails if there was a previous install, I get:
`
#### Enabling mycodo startup script
Failed to execute operation: File exists
`

I have to follow these instructions to get it removed:
https://superuser.com/questions/513159/how-to-remove-systemd-services

Any way to see if it already exists and either just leave it or remove it?

kizniche, 2017-03-14:
These two lines should come before that command:

```
systemctl disable mycodo.service
rm -rf /etc/systemd/system/mycodo.service
```

kizniche, 2017-03-14:
From [CHANGELOG.md](https://github.com/kizniche/Mycodo/blob/e89664e4cad0bb5ba61b1d124e7e2521d0ea0e55/CHANGELOG.md#419-2017-01-27) but I forgot to incorportate it into the setup. I'll push a fix shortly.

kizniche, 2017-03-14:
Might have found more errors in the setup.sh

commit dd5c945c (2017-03-14): Change directory in setup command (fixes #204)
closed 2017-03-14 by kizniche