kizniche/Mycodo #209: Relay conditionals wont trigger

pewi81, 2017-03-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.01
 
#### Problem Description
Please list: Gets error when using relay conditionals:
Trying to deactivate one relay when another turns on
Trying to activate one relay when another turns off

### Errors
Error: Actuate Relay: 'ConditionalActions' object has no attribute 'name'

========= Remote Traceback (1) =========
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/protocol.py", line 305, in _dispatch_request
    res = self._HANDLERS[handler](self, *args)
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/protocol.py", line 535, in _handle_call
    return self._local_objects[oid](*args, **dict(kwargs))
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 213, in exposed_relay_off
    return mycodo.relay_off(relay_id, trigger_conditionals)
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 561, in relay_off
    trigger_conditionals=trigger_conditionals)
  File "/var/www/mycodo/mycodo/controller_relay.py", line 328, in relay_on_off
    self.check_conditionals(relay_id, duration)
  File "/var/www/mycodo/mycodo/controller_relay.py", line 354, in check_conditionals
    timestamp, each_cond_action.id, each_cond_action.name)
AttributeError: 'ConditionalActions' object has no attribute 'name'

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Turning on the relay with the conditionals
 2. ...
 3. etc

kizniche, 2017-03-17:
Thanks for submitting this issue. I'll look into it when I get a chance.

kizniche, 2017-03-17:
I'm going to release 5.0.2 in a second so you can update. Let me know if the update doesn't resolve the issue.

commit d8ab1d41 (2017-03-17): Fix relay conditionals (closes #209)
closed 2017-03-17 by kizniche