kizniche/Mycodo #210: Relay conditionals wont trigger

pewi81, 2017-03-17:
#209 wasnt solved
Error report:
Error: Actuate Relay: 'ConditionalActions' object has no attribute 'if_relay_id'

========= Remote Traceback (1) =========
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/protocol.py", line 305, in _dispatch_request
    res = self._HANDLERS[handler](self, *args)
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/protocol.py", line 535, in _handle_call
    return self._local_objects[oid](*args, **dict(kwargs))
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 208, in exposed_relay_on
    return mycodo.relay_on(relay_id, duration, min_off_duration)
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 579, in relay_on
    min_off_duration=min_off_duration)
  File "/var/www/mycodo/mycodo/controller_relay.py", line 328, in relay_on_off
    self.check_conditionals(relay_id, duration)
  File "/var/www/mycodo/mycodo/controller_relay.py", line 354, in check_conditionals
    each_cond_action.if_relay_id,
AttributeError: 'ConditionalActions' object has no attribute 'if_relay_id'

kizniche, 2017-03-17:
Oh, I only solved part of it. sorry about that. Fixing...

kizniche, 2017-03-17:
Okay, a properly-tested version (5.0.3) has been released. I've confirmed relay conditionals work. Please let me know if you find any other errors.

pewi81, 2017-03-17:
Update 5.03 fixed the issue, many thanks, all working just fine now!

kizniche, 2017-03-17:
Thanks for finding the bugs! Also, if you ever have feature requests, create an issue with your thoughts :)

closed 2017-03-17 by kizniche