kizniche/Mycodo #211: Errors when adding LCD

zsole2, 2017-03-18:
## Mycodo Issue Report:
  - Specific Mycodo Version:  

5.0.3

#### Problem Description
Please list:   - what were you trying to do:  

add an LCD, without connecting it first, setting up  the new version to the same as the production system
 
### Errors
  - List any errors you encountered.

Error in the LCD Type field - Not a valid choice
Error in the Period field - Duration between calculating LCD output and applying to regulation must be between 5 and 86400 seconds.
Error in the Name field - This field is required.
Error in the Multiplexer Channel field - Number must be between 0 and 8.
Error in the I2C Address field - This field is required

### Steps to Reproduce the issue:
How can this issue be reproduced? 

I left all defaults intact, only added the two lines to show
 
### Additional Notes

The new 5.0.3 version looks really great! I have some errors in the log, but probably due to the fact that no sensors are connected, I plan to switch this SD card into a real environment soon to see if anything pops up...

kizniche, 2017-03-18:
I was able to reproduce the issue. Working on a fix now.

kizniche, 2017-03-18:
> I have some errors in the log

Can you show me these?

zsole2, 2017-03-18:
I intend to insert the SD card into my real system tomorrow. 
Right now I have time-out errors for the HTU21 sensor (it is not connected, so the behaviour is expected):

> 2017-03-18 17:53:18,414 - mycodo.sensors.htu21d - ERROR - HTU21DSensor raised an exception when taking a reading: 'no handle available'
2017-03-18 17:53:33,487 - mycodo.sensors.htu21d - ERROR - HTU21DSensor raised an exception when taking a reading: 'no handle available'
2017-03-18 17:53:48,559 - mycodo.sensors.htu21d - ERROR - HTU21DSensor raised an exception when taking a reading: 'no handle available'
2017-03-18 17:53:48,562 - mycodo.sensor_4 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-03-18 17:54:03,621 - mycodo.sensors.htu21d - ERROR - HTU21DSensor raised an exception when taking a reading: 'no handle available'
2017-03-18 17:54:18,678 - mycodo.sensors.htu21d - ERROR - HTU21DSensor raised an exception when taking a reading: 'no handle available'
2017-03-18 17:54:33,744 - mycodo.sensors.htu21d - ERROR - HTU21DSensor raised an exception when taking a reading: 'no handle available'
2017-03-18 17:54:33,746 - mycodo.sensor_4 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-03-18 17:54:47,307 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-18 17:54:47,316 - mycodo.pid_3 - WARNING - No data returned from influxdb

And this repeats in a continuous loop..

kizniche, 2017-03-18:
Ah, yes! Haha. If no sensor is connected, that would be expected.

kizniche, 2017-03-18:
I found the LCD issue. It was a dumb mistake that caused the issue when I refactored the LCD page. I'll have to figure out a way to fix it. It shouldn't be long.

kizniche, 2017-03-18:
Just released v5.0.4 that should fix the LCD issues.

closed 2017-03-18 by kizniche