kizniche/Mycodo #212: Setup error

zsole2, 2017-03-18:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 
5.0.3

#### Problem Description

When installing, I stopped with error at the same place. 
### Errors

This is setup.log from the second attempt:

[setup.log.txt](https://github.com/kizniche/Mycodo/files/852584/setup.log.txt)

### Steps to Reproduce the issue:

1. I had a fresh 4.1.16 installed just a few days ago, and nothing was changed from the initial stage, even a sensor was not added. 
2. Moved the Mycodo directory to a backup
3. Followed the install instructions, except that used 'v5.0.3.tar.gz' from github/releases
4. setup.sh stopped at the same place at two attempts of installation, see log above

### Additional Notes

After checking that influxd was up and running, I was able to finish installing the  system by completing the setup from  lines 76.

kizniche, 2017-03-18:
Yeah, this seems to be an issue with the time it takes to start influxdb. I initially had no sleep period, and then found the issue. I added a 15 seconds sleep period, but this wasn't long enough to allow it to start, so I changed the sleep time to 90 seconds. That seems to be good enough for Pis 2, and 3, but not Zero and 1s. I suspect you have a zero or Pi1 (pretty sure I remember you using a zero?).

So, unless I add a long wait, like 3-5 minutes, it will error at the point. I was considering adding a loop that if it fails, it will wait another 90 seconds, for a max of 3 tries.

What are your thoughts?

zsole2, 2017-03-18:
That what I was suspecting, I indeed use Pi Zero. 
Adding more sleep time seems to effect users with more powerful Pi's, so it does not look a too good idea. The loop might be better, and in that case, even the period might be less...

kizniche, 2017-03-18:
I found a solution that I'll add to the setup.sh script. I'll have to do some release reordering, but it should work out in the end. Do you think 5 minutes waiting is enough, or should I add more?

```bash
printf "\n#### Creating InfluxDB database and user\n"
# Attempt to connect to influxdb 5 times, sleeping 60 seconds every fail
for i in {1..5}; do
influx -execute "CREATE DATABASE mycodo_db" &&
influx -database mycodo_db -execute "CREATE USER mycodo WITH PASSWORD 'mmdu77sj3nIoiajjs'" &&
break || sleep 60; done
```

zsole2, 2017-03-18:
Probably 5 minutes will be OK, influxd was already running when I checked after first checking setup.log and identified where the problem may be...

kizniche, 2017-03-18:
I deleted the v5.0.0 and v5.0.3 releases, pushed the update for the setup.sh script, released v5.0.0, pushed the updates for the LCD issues, then released v5.0.4. So, users coming from v4.2.0 should get the new setup.sh script, and users currently on 5.x will be able to update to v5.0.4 and get the LCD fixes.

closed 2017-03-18 by kizniche

zsole2, 2017-03-18:
By the way, the whole setup issue came around when the v4.1.6 did not offer to upgrade to v5, but only v4.2.0. I thought this is the expected behaviour, and nothing indicates in the instruction how to upgrade from v4 to v5. Did I miss something?

zsole2, 2017-03-19:
Thanks for the clarification.

A few other notes on install issues: 

1) In the meantime, I did a fresh install on another SD-card. The for influxd loop worked fine, but the script stopped later on: 

```
 #### Compiling Translations
 /home/pi/Mycodo/mycodo/scripts/upgrade_commands.sh: line 18: pybabel: command not found
 
 *******************************
 *** ERROR: Install Aborted! ***
 *******************************
``` 

2) When upgrading from v5.0.3 to v5.0.4, the web interface still directs the user in case of an error to the log which is found at "Tools > Mycodo log > ..."

kizniche, 2017-03-19:
Thanks. Updated the reference to pybabel in the virtualenv and altered the upgrade notification.