kizniche/Mycodo #216: Anonymous Statistics Preview

kizniche, 2017-03-19:
I just wanted to show anyone who is interested what the anonymous statistics page looks like. I've erased the anonymous statistics ID of each machine (just to be safe), even though it can't identify anyone closer than the country origin. You can see there is information missing from certain users, as their version is older and doesn't report that information. This information is used to determine how much of the major features are used by users, and see if any configurations cause issues (high ram, startup time, etc.). You can check out the [mycodo stats viewer repository](https://github.com/kizniche/mycodo_stats_viewer) if you want to review the statistics viewer code, and if you want to review the Mycodo statistics code, most of it resides [here](https://github.com/kizniche/Mycodo/blob/fc1471e30a2cae0d25aa0c3519d3402fdda5167c/mycodo/utils/statistics.py).

[image]

Myke974, 2017-03-19:
I'm interested ! :) thank you !

kizniche, 2017-03-19:
Some interesting observations:

  - 22 is using 164 MB of ram
  - 44 is the first Pi ZeroW to to use Mycodo (with statistics enabled)
  - 45 took 44443 seconds for the daemon to start

zsole2, 2017-03-19:
Definitely interesting!

The Zero W should be me, it has just arrived and used Mycodo  to test it. It works as good as the Zero itself, at least on first sight.  

What is strange, though, that I usually get the following error in the log on fresh installs (I just did one now): 
 
> 2017-03-19 19:44:00,806 - mycodo.stats - ERROR - Could not send anonymous usage statistics: 'ascii' codec can't encode character u'\xe1' in position 1: ordinal not in range(128)
> Traceback (most recent call last):
>   File "/var/www/mycodo/mycodo/utils/statistics.py", line 279, in send_anonymous_stats
>     add_update_csv(STATS_CSV, 'country', geocoder.ip('me').country)
>   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/geocoder/api.py", line 418, in ip
>     return get(location, provider='ipinfo', **kwargs)
>   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/geocoder/api.py", line 144, in get
>     return options[provider][method](location, **kwargs)
>   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/geocoder/ipinfo.py", line 23, in __init__
>     self._initialize(**kwargs)
>   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/geocoder/base.py", line 135, in _initialize
>     self._json()
>   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/geocoder/base.py", line 142, in _json
>     value = getattr(self, key)
>   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/geocoder/ipinfo.py", line 45, in address
>     return '{0}, {1}, {2}'.format(self.city, self.state, self.country)
> UnicodeEncodeError: 'ascii' codec can't encode character u'\xe1' in position 1: ordinal not in range(128)

Doesn't this prevent sending the data? Telling the truth, I did not bother too much checking the logs lately, I had uptimes of 48 and 94 days on previous installs, so it is a testament of the usability of Mycodo!

kizniche, 2017-03-20:
I've never seen that error, and I'm not really sure why it still sent, unless the error only occasionally occurs. I committed an edit I think may fix the issue. I also updated the screenshot. Not much new except a few more devices and the heatmap is more useful.

kizniche, 2017-03-20:
I added another image to the main post, which includes 339 Mycodo installs on the list. This goes back to when statistics were first collected, with the earliest install on the list being May 23, 2016.

closed 2017-10-24 by kizniche