kizniche/Mycodo #217: HTU21D does not start after upgrade

zsole2, 2017-03-22:
## Mycodo Issue Report:
v5.0.8

#### Problem Description
Upgraded this morning. After a few minutes, I realized that no temp data are collected. As it helped in the past, I rebooted, and the temp readings normalized. 
 
### Errors
[mycodo.log.txt](https://github.com/kizniche/Mycodo/files/860633/mycodo.log.txt)
[influxd.log.txt](https://github.com/kizniche/Mycodo/files/860635/influxd.log.txt)
[mycodoupgrade.log.txt](https://github.com/kizniche/Mycodo/files/860658/mycodoupgrade.log.txt)

### Steps to Reproduce the issue:

### Additional Notes
Apparently the reboot solved the issue with HTU21D, which as far as I can see is indicated in line 346 of the mycodo log. The upgrade log shows no errors. 
After reboot, some errors are there, but then everything normalizes, after about 6-7 minutes, which coincides with the influxd log, this is the time needed to start it up on Pi Zero (in line with the issue the other day with the setup script).
The stat encoding is still not solved, but it the issue  probably lies with the geocoder module.
It would help reading the log if a few separator lines were present at the start of the daemon... A timestamp at the finish of the upgrade would be also nice...

kizniche, 2017-03-22:
> Apparently the reboot solved the issue with HTU21D

I'm not sure what happened there, but it appears it is an issue with the Pi Zero slow bootup and perhaps influxdb hadn't started yet.

> The stat encoding is still not solved

Yes, that's a particularly annoying bug because I don't know how many systems it's affecting.

> It would help reading the log if a few separator lines were present at the start of the daemon

I'll add these

> A timestamp at the finish of the upgrade would be also nice

This was included in 5.0.8, but you'll have to wait fir the next upgrade for it to take effect.

kizniche, 2017-03-22:
I created an issue (#218) to track the geocoder issue, but I also just released [v5.0.9](https://github.com/kizniche/Mycodo/releases/tag/v5.0.9) that updates geocoder from 1.15.0 to 1.21.0. Can you report back in that issue if it fixes it for you?

kizniche, 2017-03-22:
Reading geocoder's changelog, it appears it may have been fixed a few versions ago:

```Fixed Python2 encoding issues```

zsole2, 2017-03-22:
I'm afraid the reboot issue was not due to the slow start of influxd. The last section of mycodo log show about 7 mins delay, while the next to last section, after upgrade and before the reboot, is about 30 mins in length. Only then I decided to reboot.
I'll upgrade soon a see the other stuff...

kizniche, 2017-03-22:
I was meaning the demon rebooting. Have you tried restarting the daemon (```sudo service mycodo restart```) instead of rebooting the whole system?

zsole2, 2017-03-22:
It  appears that the upgrade was perfect, without any errors !!!! No encoding error, no influxd error... 

n addition, the statistic shows that it detected HU as country, can you check if  id No. nhYIezpq8Jxr was sent out around 19:00 Central European time?

kizniche, 2017-03-22:
Great! I'll have to check later, as the stats can only be viewed from a specific computer.

kizniche, 2017-03-22:
I'm finding some issues with the upgrade process. It's highly confusing when I add alembic (which upgrades the database when I need to change it), and I'm not sure if I'm breaking peoples installs or not. If it happens to you, just delete your database and a new one will be generated by the web UI. Hopefully v5.0.11 is a final solution to the headache I've been causing myself with the past few updates.

kizniche, 2017-03-22:
I think everything is in a working state now, but like I've said, I'm not sure if or how many databases I've messed up. I have an event I need to be at for the next several hours, so I can't do the testing I want to do. Let me know if you have any issues. If you do, you cam probably delete the only row in the alembic_version table, save the database, and restart the web UI, then the daemon, and it should fix itself. My head is spinning and I need to take a break.

zsole2, 2017-03-22:
Right now everything looks perfect in my setup with v5.0.10, so  I'll wait  for all reading to stabilize. Let's see if  others follow these new releases and report problems if they have any.

kizniche, 2017-03-23:
If you made it to 5.0.10 without issue, you are probably fine. This issue will affect people who are upgrading form an older version, such as 5.0.7. I didn't notice the issue at first because I was upgrading incrementally, and not going multiple versions before upgrading. Upgrades are a headache sometimes, because you have to ensure upgrades work flawlessly from any other minor/subversion.

kizniche, 2017-03-26:
Not really sure where to go with this issue. It'd guess it was a one-off event. What do you think?

zsole2, 2017-03-26:
Probably, I have upgraded 3 times since then, without problem....

zsole2, 2017-04-01:
The HTU21D sensor again gave me trouble on upgrade. This time it was from v5.0.10 to v5.0.17. See line 88 of [mycodo_log.txt](https://github.com/kizniche/Mycodo/files/887857/mycodo_log.txt).
After realizing the problem, I tried to restart the daemon, but it did not start, and unfortunately I was not at my PC, and did not save the exact failure info.
Then I rebooted, and it started OK, except for the 4-5 minutes that was needed for influxd to start.
The empty lines are really useful for reading the log, by the way... Could be good to have it when the daemon starts after reboot, too...

kizniche, 2017-04-01:
My HTU21D sensors (2) didn't work properly when I first tested them a few months ago. I'll see if I can get them working when I get home. I suspect it was due to using 5 volts instead of 3 volts power.

I'm still not sure what the issue is, but I suspect it's influxdb restarting after an influxdb upgrade. Check your upgrade log to see of your influxdb was indeed upgraded with your mycodo upgrade.

kizniche, 2017-04-01:
So I may need to add a helper script that checks if influxdb has started before continuing the upgrade process, sort of how the setup script does during an install to ensure it's running to be able to create the database and user.

kizniche, 2017-10-24:
I believe this may have been due to an influxdb issue. I'm going to close it since there have been a lot of patches to influxdb to resolve crashing issues. Feel free to reopen if the current version of influxdb still does this.

closed 2017-10-24 by kizniche