kizniche/Mycodo #219: Error update 5.0.11

Ericktz, 2017-03-23:
Previously Version 5.0.7, error apache 2, in attached file
[error.log.txt](https://github.com/kizniche/Mycodo/files/863614/error.log.txt)

kizniche, 2017-03-23:
Unfortunately, I wasn't able to prevent this. See [this comment](https://github.com/kizniche/Mycodo/issues/217#issuecomment-288518865) for an explanation of the issue. Hopefully this won't happen again.

Ericktz, 2017-03-23:
maybe it not the same problem... :(, 
[image]

```
pi@raspi:~ $ systemctl status mycodo.service
● mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled)
   Active: failed (Result: exit-code) since Thu 2017-03-23 11:06:55 CLST; 23s ago
  Process: 21297 ExecStart=/var/www/mycodo/env/bin/python /var/www/mycodo/mycodo/mycodo_daemon.py (code=exited, status=1/FAILURE)
/mycodo_daemon.py (code=exited, status=1/FAILURE)
pi@raspi:~ $ sudo python ~/Mycodo/mycodo/mycodo_daemon.py -d
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_daemon.py", line 46, in <module>
    from mycodo.controller_lcd import LCDController
  File "/home/pi/Mycodo/mycodo/controller_lcd.py", line 62, in <module>
    from databases.models import (
  File "/home/pi/Mycodo/mycodo/databases/__init__.py", line 6, in <module>
    from mycodo.mycodo_flask.extensions import db
  File "/home/pi/Mycodo/mycodo/mycodo_flask/extensions.py", line 11, in <module>
    from flask_sqlalchemy import SQLAlchemy
ImportError: No module named flask_sqlalchemy

```

in /var/log/mycodo/mycodolog.log:

```
2017-03-22 14:46:44,681 - mycodo - INFO - 56.352 MB ram in use
2017-03-23 00:35:28,721 - mycodo - ERROR - Timelapse ERROR
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 321, in run
    self.timelapse_check(each_camera, now)
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 788, in timelapse_check
    camera_record('timelapse', camera)
  File "/var/www/mycodo/mycodo/devices/camera.py", line 175, in camera_record
    img_edited = img_orig.copy()
AttributeError: 'NoneType' object has no attribute 'copy'
2017-03-23 00:39:55,963 - mycodo - INFO - Received command to terminate daemon
2017-03-23 00:39:56,209 - mycodo - INFO - All LCD controllers stopped
2017-03-23 00:39:56,286 - mycodo.sensor_1 - INFO - Deactivated in 77.1 ms
2017-03-23 00:39:56,307 - mycodo.sensor_2 - INFO - Deactivated in 97.1 ms
2017-03-23 00:39:56,318 - mycodo.timer_1 - INFO - Deactivated in 108.8 ms
2017-03-23 00:39:56,319 - mycodo - INFO - All Timer controllers stopped
2017-03-23 00:39:56,320 - mycodo - INFO - All Sensor controllers stopped
2017-03-23 00:39:56,320 - mycodo - INFO - All PID controllers stopped
2017-03-23 00:39:56,390 - mycodo.relay - INFO - Relay controller deactivated in 69.3 ms
2017-03-23 00:39:56,391 - mycodo - INFO - Mycodo terminated in 0.428 seconds
2017-03-23 00:44:49,768 - mycodo - INFO - Mycodo daemon v5.0.11 starting
2017-03-23 00:44:49,971 - mycodo.databases.utils - ERROR - Error raised in session_scope.  Session will be rolled back: db_uri='sqlite:////home/pi/Mycodo/mycodo/../databases/mycodo.db', error='(sqlite3.OperationalError) no such column: misc.hide_tooltips [SQL: u'SELECT misc.id AS misc_id, misc.dismiss_notification AS misc_dismiss_notification, misc.force_https AS misc_force_https, misc.hide_alert_info AS misc_hide_alert_info, misc.hide_alert_success AS misc_hide_alert_success, misc.hide_alert_warning AS misc_hide_alert_warning, misc.hide_tooltips AS misc_hide_tooltips, misc.language AS misc_language, misc.login_message AS misc_login_message, misc.relay_usage_cost AS misc_relay_usage_cost, misc.relay_usage_currency AS misc_relay_usage_currency, misc.relay_usage_dayofmonth AS misc_relay_usage_dayofmonth, misc.relay_usage_volts AS misc_relay_usage_volts, misc.relay_usage_report_gen AS misc_relay_usage_report_gen, misc.relay_usage_report_span AS misc_relay_usage_report_span, misc.relay_usage_report_day AS misc_relay_usage_report_day, misc.relay_usage_report_hour AS misc_relay_usage_report_hour, misc.stats_opt_out AS misc_stats_opt_out \nFROM misc\n LIMIT ? OFFSET ?'] [parameters: (1, 0)]'
2017-03-23 00:51:59,858 - mycodo - INFO - Mycodo daemon v5.0.11 starting
2017-03-23 00:52:00,221 - mycodo.databases.utils - ERROR - Error raised in session_scope.  Session will be rolled back: db_uri='sqlite:////home/pi/Mycodo/mycodo/../databases/mycodo.db', error='(sqlite3.OperationalError) no such column: misc.hide_tooltips [SQL: u'SELECT misc.id AS misc_id, misc.dismiss_notification AS misc_dismiss_notification, misc.force_https AS misc_force_https, misc.hide_alert_info AS misc_hide_alert_info, misc.hide_alert_success AS misc_hide_alert_success, misc.hide_alert_warning AS misc_hide_alert_warning, misc.hide_tooltips AS misc_hide_tooltips, misc.language AS misc_language, misc.login_message AS misc_login_message, misc.relay_usage_cost AS misc_relay_usage_cost, misc.relay_usage_currency AS misc_relay_usage_currency, misc.relay_usage_dayofmonth AS misc_relay_usage_dayofmonth, misc.relay_usage_volts AS misc_relay_usage_volts, misc.relay_usage_report_gen AS misc_relay_usage_report_gen, misc.relay_usage_report_span AS misc_relay_usage_report_span, misc.relay_usage_report_day AS misc_relay_usage_report_day, misc.relay_usage_report_hour AS misc_relay_usage_report_hour, misc.stats_opt_out AS misc_stats_opt_out \nFROM misc\n LIMIT ? OFFSET ?'] [parameters: (1, 0)]'
2017-03-23 11:06:55,223 - mycodo - INFO - Mycodo daemon v5.0.11 starting
2017-03-23 11:06:55,440 - mycodo.databases.utils - ERROR - Error raised in session_scope.  Session will be rolled back: db_uri='sqlite:////home/pi/Mycodo/mycodo/../databases/mycodo.db', error='(sqlite3.OperationalError) no such column: misc.hide_tooltips [SQL: u'SELECT misc.id AS misc_id, misc.dismiss_notification AS misc_dismiss_notification, misc.force_https AS misc_force_https, misc.hide_alert_info AS misc_hide_alert_info, misc.hide_al [... cut ...]
Failed: Error while running post-upgrade script.

The upgrade has failed: Attempting to revert moving the old Mycodo install.

Successfully reverted moving the old Mycodo install directory. Moved /var/Mycodo-backups/Mycodo-backup-2017-03-23_00-39-46-5.0.7 to /home/pi/Mycodo

#### Creating proper users, directories, and permissions
useradd: user 'mycodo' already exists
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `adm'.
The user `mycodo' is already a member of `video'.
Done.

```

kizniche, 2017-03-23:
This is the error from your initial log, that is the error I was referring to from the database issue:

```(sqlite3.OperationalError) no such column: misc.hide_tooltips```

> ImportError: No module named flask_sqlalchemy

This is strange. Somehow this python module didn't install.

> The upgrade has failed: Attempting to revert moving the old Mycodo install.

Since the database was messed up, the web UI couldn't start (technically, gave a 500 error. This prevents the upgrade from working.

pewi81, 2017-03-23:
I also get an 500 error after update..

kizniche, 2017-03-23:
See [this commant](https://github.com/kizniche/Mycodo/issues/217#issuecomment-288518865) for how to fix.

pewi81, 2017-03-23:
I have absolutely no idea what to do!

kizniche, 2017-03-23:
You have 2 options:

1. Delete ~/Mycodo/databases/mycodo.db and restart your Pi.
2. Follow the instructions for editing the database.

Deleting the database is easier, but you will have to set everything up again. Using SQLiteStudio isn't too difficult, but it is an interface if using it for the first time.

kizniche, 2017-03-23:
Here are specific instructions for using SQLiteStudio... (next comment)

kizniche, 2017-03-23:
1. Download ~/Mycodo/databases/mycodo.db to your computer
2. Select Database -> Add a Database and select mycodo.db
3. Double-click the newly-added database to show the tables
4. Double-click the "alembic_version" table
5. Select the "Data" tab
6. Select row 1 (there should only be one row in there)
7. Press the delete key to delete the selected row (or press the delete row icon)
8. Press Ctrl+Return to commit the row deletion (or press the green check mark icon to commit)
9. Upload mycodo.db, overwriting ~/Mycodo/databases/mycodo.db on the Pi

pewi81, 2017-03-23:
Many thanks, will give it a try!

zsole2, 2017-03-23:
I got the same error.
Removed the alembic version row, rebooted, still got the same error page.
Checking the daemon, it says:

> ● mycodo.service - Mycodo server
>    Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled)
>    Active: failed (Result: exit-code) since cs 2017-03-23 19:42:38 CET; 21min ago
>   Process: 638 ExecStart=/var/www/mycodo/env/bin/python /var/www/mycodo/mycodo/mycodo_daemon.py (code=exited, status=1/FAILURE)
> 
> márc 23 19:42:37 chz-112 python[638]: File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 202, in raise_from_cause
> márc 23 19:42:37 chz-112 python[638]: reraise(type(exception), exception, tb=exc_tb, cause=cause)
> márc 23 19:42:37 chz-112 python[638]: File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1139, in _execute_context
> márc 23 19:42:37 chz-112 python[638]: context)
> márc 23 19:42:37 chz-112 python[638]: File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 450, in do_execute
> márc 23 19:42:37 chz-112 python[638]: cursor.execute(statement, parameters)
> márc 23 19:42:37 chz-112 python[638]: sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: misc.hide_tooltips [SQL: u'SELECT misc.id AS misc_id, misc.dismiss_notification AS misc_dismiss_notification, misc.force_https AS misc_force_https, misc.hide_alert_info AS misc_hide_alert_info, misc.hide_alert_success AS misc_hide_alert_success, misc.hide_alert_warning AS misc_hide_alert_warning, misc.hide_tooltips AS misc_hide_tooltips, misc.language AS misc_language, misc.login_message AS misc_login_message, misc.relay_usage_cost AS misc_relay_usage_cost, misc.relay_usage_currency AS misc_relay_usage_currency, misc.relay_usage_dayofmonth AS misc_relay_usage_dayofmonth, misc.relay_usage_volts AS misc_relay_usage_volts, misc.relay_usage_report_gen AS misc_relay_usage_report_gen, misc.relay_usage_report_span AS misc_relay_usage_report_span, misc.relay_usage_report_day AS misc_relay_usage_report_day, misc.relay_usage_report_hour AS misc_relay_usage_report_hour, misc.stats_opt_out AS misc_stats_opt_out \nFROM misc\n LIMIT ? OFFSET ?'] [parameters: (1, 0)]
> márc 23 19:42:38 chz-112 systemd[1]: mycodo.service: control process exited, code=exited status=1
> márc 23 19:42:38 chz-112 systemd[1]: Failed to start Mycodo server.
> márc 23 19:42:38 chz-112 systemd[1]: Unit mycodo.service entered failed state.

Indeed, there is no misc.hide_tooltip row in my database!

zsole2, 2017-03-23:
And the version info is recreated, but it is empty.

pewi81, 2017-03-23:
I removed the alembic version row but still same problem, will try to delete the database..

zsole2, 2017-03-23:
Just checked my other system running, v5.0.10, and it has the hide_tooltip row in the misc table, and the alembic version is also filled.

kizniche, 2017-03-23:
One minute... I make a mistake in the instructions.

kizniche, 2017-03-23:
Instead of deleting the row, just erase the version (so the value is blank), and save.

kizniche, 2017-03-23:
I updated the instructions with the correct method to fix the database.

kizniche, 2017-03-23:
Actually, if you're not running 5.0.10 or 5.0.11, then the instructions may not work. Hmm. It's a multifaceted issue!

kizniche, 2017-03-23:
So, you can use the install instructions to reinstall Mycodo, then copy your database over and follow the database fix instructions, then reboot the Pi. That should fix it if the instructions haven't been working thus far.

kizniche, 2017-03-23:
Goes to show I should have tested the database upgrade system a little more before releasing version 5.0.0.

pewi81, 2017-03-23:
I deleted the database, rebooted, but still get the 500 error..

zsole2, 2017-03-23:
Copying back to Mycodo/databases might be prevented by folder permissions, at least this was the reason for my first failed attempt. 
After figuring this out, I managed to copy over my database from my other system (which I always wanted to be able to do, instead of setting everything up repeatedly), and it apparently works! Can this cause any adverse effects?

pewi81, 2017-03-23:
I cant get it to work, trying an clean installation! If this doesnt work, how do I reinstall the previous version I have in the backup folder?

kizniche, 2017-03-23:
Here are commands to perform the same actions, but with sqlite3:

```
sqlite3 ~/Mycodo/databases/mycodo.db
UPDATE alembic_version SET version_num='';
.exit
```

pewi81, 2017-03-23:
When I did this I instead get a 403 error: Forbidden. You don't have permission to access / on this server.

kizniche, 2017-03-23:
> how do I reinstall the previous version I have in the backup folder?

Copy/move the entire folder back in place of ~/Mycodo should be all that's needed.

> When I did this I instead get a 403 error...

When you did what?

pewi81, 2017-03-23:
After I applied:
sqlite3 ~/Mycodo/databases/mycodo.db
UPDATE alembic_version SET version_num='';
.exit

.. I get the 403 error!

kizniche, 2017-03-23:
DId you restart the servers or restart the Pi?

kizniche, 2017-03-23:
@pewi81 You never mentioned what Mycodo version you are using.

pewi81, 2017-03-23:
For now, none! Trying to go back to 5.0.5!

pewi81, 2017-03-23:
Still get the 403 after reboot!

kizniche, 2017-03-23:
Whether the commands I've been giving to @Ericktz and @zsole2 succeed depend on the Mycodo version. @pewi81, I need to know what version you were upgrading to to be able to properly help you.

pewi81, 2017-03-23:
I upgraded to the latest version!

kizniche, 2017-03-23:
The current latest version, 5.0.11? or a different current version at the time you were upgrading? Please give me hard numbers.

pewi81, 2017-03-23:
v.5.0.11 about 3 hours ago

kizniche, 2017-03-23:
So, if the instructions so far haven't worked for 5.0.11, there's one last think I can suggest before leaving the last resort of reinstall and start with a new database, is to run the following command, then reboot:

```sudo ~/Mycodo/mycodo/scripts/upgrade_post.sh```

kizniche, 2017-03-23:
Also, you can verify that you are indeed editing version_num to be blank, in the first row of the table alembic_version, by opening the database in SQLiteStudio?

pewi81, 2017-03-23:
Yes I tried that and it didnt work!

kizniche, 2017-03-23:
Then you're left with reinstalling.

pewi81, 2017-03-23:
Ok, copying my backup files from 5.0.5 to the Mycodo folder didnt work at first, but after applying: 
sudo ~/Mycodo/mycodo/scripts/upgrade_post.sh
I´m now back at 5.0.5 and everything seems to work fine! Thanks for your help!

pewi81, 2017-03-23:
I will run on 5.0.5 since its stable now, the only error I can find is when I enter the upgrade page and get the following message:

> There was an error that occurred during the upgrade.
> View the upgrade log at /var/log/mycodo/mycodoupgrade.log.
> Delete ~/Mycoco/.upgrade to acknowledge this message and restore upgrade functionality.

I cant find anything to remove but for now I dont have to upgrade, just curious on what might cause this error.

The last 30 lines from the upgrade log (when I tried to upgrade to 5.0.11 earlier) is:

> 
> The user `mycodo' is already a member of `gpio'.
> The user `mycodo' is already a member of `adm'.
> The user `mycodo' is already a member of `video'.
> 
> #### Reloading systemctl, Mycodo daemon, and apache2
> Restarting apache2 (via systemctl): apache2.service.
> WARNING: combining -O with -r or -p will mean that all downloaded content
> will be placed in the single file you specified.
> 
> converted 'http://127.0.0.1' (ANSI_X3.4-1968) -> 'http://127.0.0.1' (UTF-8)
> converted 'http://127.0.0.1/' (ANSI_X3.4-1968) -> 'http://127.0.0.1/' (UTF-8)
> --2017-03-23 18:18:43--  http://127.0.0.1/
> Connecting to 127.0.0.1:80... connected.
> HTTP request sent, awaiting response... 500 Internal Server Error
> 2017-03-23 18:18:55 ERROR 500: Internal Server Error.
> 
> Job for mycodo.service failed. See 'systemctl status mycodo.service' and 'journalctl -xn' for details.
> Failed: Error while running post-upgrade script.
> 
> 
> The upgrade has failed: Attempting to revert moving the old Mycodo install.
> 
> Successfully reverted moving the old Mycodo install directory. Moved /var/Mycodo-backups/Mycodo-backup-2017-03-23_18-11-38-5.0.5 to /home/pi/Mycodo
> 
> #### Creating proper users, directories, and permissions
> useradd: user 'mycodo' already exists
> The user `mycodo' is already a member of `gpio'.
> The user `mycodo' is already a member of `adm'.
> The user `mycodo' is already a member of `video'.
> Done.

kizniche, 2017-03-23:
It's what this whole git issue is about, the database issue. As I've said before, you need the correct database version. I've given the instructions how to upgrade with your current database. You have those 2 options, alter the database manually or start anew.

kizniche, 2017-03-23:
The upgrade fail message has a typo. It's supposed to be ~/Mycodo/.upgrade

Ericktz, 2017-03-23:
i do : 
```
sudo sqlite3 ~/Mycodo/databases/mycodo.db
UPDATE alembic_version SET version_num='';
.exit

```
and reboot, but same error ocurr, in apache2/error.log:

`[Thu Mar 23 18:01:21.469466 2017] [wsgi:error] [pid 1191:tid 1963279408] [remote 200.14.84.157:32374] OperationalError: (sqlite3.OperationalError) no such column: misc.hide_tooltips [SQL: u'SELECT count(*) AS count_1 \\nFROM (SELECT misc.id AS misc_id, misc.dismiss_notification AS misc_dismiss_notification, misc.force_https AS misc_force_https, misc.hide_alert_info AS misc_hide_alert_info, misc.hide_alert_success AS misc_hide_alert_success, misc.hide_alert_warning AS misc_hide_alert_warning, misc.hide_tooltips AS misc_hide_tooltips, misc.language AS misc_language, misc.login_message AS misc_login_message, misc.relay_usage_cost AS misc_relay_usage_cost, misc.relay_usage_currency AS misc_relay_usage_currency, misc.relay_usage_dayofmonth AS misc_relay_usage_dayofmonth, misc.relay_usage_volts AS misc_relay_usage_volts, misc.relay_usage_report_gen AS misc_relay_usage_report_gen, misc.relay_usage_report_span AS misc_relay_usage_report_span, misc.relay_usage_report_day AS misc_relay_usage_report_day, misc.relay_usage_report_hour AS misc_relay_usage_report_hour, misc.stats_opt_out AS misc_stats_opt_out \\nFROM misc) AS anon_1']`

then i try this

> sudo ~/Mycodo/mycodo/scripts/upgrade_post.sh

but at the final log : 

```
equirement already up-to-date: pycparser in /usr/local/lib/python2.7/dist-packages (from cffi>=1.1->bcrypt->-r /home/pi/Mycodo/install/requirements.txt (line 7))
Requirement already up-to-date: decorator in /usr/local/lib/python2.7/dist-packages (from ratelim->geocoder==1.21.0->-r /home/pi/Mycodo/install/requirements.txt (line 16))
Building wheels for collected packages: cffi
  Running setup.py bdist_wheel for cffi ... done
  Stored in directory: /root/.cache/pip/wheels/7c/c4/64/889d3d2de89188ed0fcb0b313106338f06127874911eeb4c96
Successfully built cffi
Installing collected packages: argparse, SQLAlchemy, Flask-SQLAlchemy, Flask-WTF, geocoder, imutils, cffi
  Found existing installation: argparse 1.2.1
    Not uninstalling argparse at /usr/lib/python2.7, as it is in the standard library.
  Found existing installation: SQLAlchemy 0.9.8
    Uninstalling SQLAlchemy-0.9.8:
      Successfully uninstalled SQLAlchemy-0.9.8
  Found existing installation: Flask-SQLAlchemy 1.0
    Uninstalling Flask-SQLAlchemy-1.0:
      Successfully uninstalled Flask-SQLAlchemy-1.0
  Found existing installation: Flask-WTF 0.12
    Uninstalling Flask-WTF-0.12:
      Successfully uninstalled Flask-WTF-0.12
  Found existing installation: geocoder 1.15.0
    Uninstalling geocoder-1.15.0:
      Successfully uninstalled geocoder-1.15.0
  Found existing installation: imutils 0.4.0
    Uninstalling imutils-0.4.0:
      Successfully uninstalled imutils-0.4.0
  Found existing installation: cffi 1.9.1
    Uninstalling cffi-1.9.1:
      Successfully uninstalled cffi-1.9.1
Successfully installed Flask-SQLAlchemy-2.1 Flask-WTF-0.14.2 SQLAlchemy-1.0.13 argparse-1.4.0 cffi-1.10.0 geocoder-1.21.0 imutils-0.3.10

#### Upgrading database with alembic
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.

#### Removing statistics file
rm: cannot remove ‘/home/pi/Mycodo/databases/statistics.csv’: No such file or directory

#### Enabling mycodo startup script
Removed symlink /etc/systemd/system/multi-user.target.wants/mycodo.service.
Removed symlink /etc/systemd/system/mycodo.service.
Created symlink from /etc/systemd/system/mycodo.service to /home/pi/Mycodo/install/mycodo.service.
Created symlink from /etc/systemd/system/multi-user.target.wants/mycodo.service to /home/pi/Mycodo/install/mycodo.service.

#### Compiling Translations
compiling catalog mycodo_flask/translations/ko/LC_MESSAGES/messages.po to mycodo_flask/translations/ko/LC_MESSAGES/messages.mo
compiling catalog mycodo_flask/translations/es/LC_MESSAGES/messages.po to mycodo_flask/translations/es/LC_MESSAGES/messages.mo
compiling catalog mycodo_flask/translations/fr/LC_MESSAGES/messages.po to mycodo_flask/translations/fr/LC_MESSAGES/messages.mo
#### Updating crontab entry
Removing crontab entry ...
Appending to crontab:
-----------------------------------------------
@reboot /usr/local/bin/pigpiod &-----------------------------------------------
To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh mycodo --remove
[... 15 lines cut ...]
2017-03-23 17:52:17 ERROR 500: Internal Server Error.

Job for mycodo.service failed. See 'systemctl status mycodo.service' and 'journalctl -xn' for details.
```

in 'sudo systemctl status mycodo.service -l' : 

```
pi@raspi:~ $ sudo systemctl status mycodo.service -l
7● mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled)
   Active: failed (Result: exit-code) since Thu 2017-03-23 18:01:04 CLST; 13min ago
  Process: 637 ExecStart=/var/www/mycodo/env/bin/python /var/www/mycodo/mycodo/mycodo_daemon.py (code=exited, status=1/FAILURE)

Mar 23 18:01:04 raspi python[637]: File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 202, in raise_from_cause
Mar 23 18:01:04 raspi python[637]: reraise(type(exception), exception, tb=exc_tb, cause=cause)
Mar 23 18:01:04 raspi python[637]: File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1139, in _execute_context
[... cut ...] eport_span AS misc_relay_usage_report_span, misc.relay_usage_report_day AS misc_relay_usage_report_day, misc.relay_usage_report_hour AS misc_relay_usage_report_hour, misc.stats_opt_out AS misc_stats_opt_out \nFROM misc\n LIMIT ? OFFSET ?'] [parameters: (1, 0)]
Mar 23 18:01:04 raspi systemd[1]: mycodo.service: control process exited, code=exited status=1
Mar 23 18:01:04 raspi systemd[1]: Failed to start Mycodo server.
Mar 23 18:01:04 raspi systemd[1]: Unit mycodo.service entered failed state.
```

kizniche, 2017-03-23:
What is the output from these commands?

```
sudo sqlite3 ~/Mycodo/databases/mycodo.db
DELETE FROM alembic_version;
.exit

source ~/Mycodo/env/bin/activate
cd ~/Mycodo/databases
alembic upgrade head
```

Ericktz, 2017-03-24:
```
pi@raspi:~$ sudo sqlite3 ~/Mycodo/databases/mycodo.db
SQLite version 3.8.7.1 2014-10-29 13:59:56
Enter ".help" for usage hints.
sqlite> DELETE FROM alembic_version;
sqlite> .exit
pi@raspi:~$ 
pi@raspi:~$ source ~/Mycodo/env/bin/activate
(env) pi@raspi:~$ cd ~/Mycodo/databases
(env) pi@raspi:~/Mycodo/databases$ alembic upgrade head
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.
INFO  [alembic.runtime.migration] Running upgrade  -> 059a47f950b8, add option to show/hide tooltips
Traceback (most recent call last):
  File "/usr/local/bin/alembic", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/alembic/config.py", line 479, in main
    CommandLine(prog=prog).main(argv=argv)
  File "/usr/local/lib/python2.7/dist-packages/alembic/config.py", line 473, in main
    self.run_cmd(cfg, options)
  File "/usr/local/lib/python2.7/dist-packages/alembic/config.py", line 456, in run_cmd
    **dict((k, getattr(options, k, None)) for k in kwarg)
  File "/usr/local/lib/python2.7/dist-packages/alembic/command.py", line 254, in upgrade
    script.run_env()
  File "/usr/local/lib/python2.7/dist-packages/alembic/script/base.py", line 416, in run_env
    util.load_python_file(self.dir, 'env.py')
  File "/usr/local/lib/python2.7/dist-packages/alembic/util/pyfiles.py", line 93, in load_python_file
    module = load_module_py(module_id, path)
  File "/usr/local/lib/python2.7/dist-packages/alembic/util/compat.py", line 75, in load_module_py
    mod = imp.load_source(module_id, path, fp)
  File "alembic/env.py", line 70, in <module>
    run_migrations_online()
  File "alembic/env.py", line 65, in run_migrations_online
    context.run_migrations()
  File "<string>", line 8, in run_migrations
  File "/usr/local/lib/python2.7/dist-packages/alembic/runtime/environment.py", line 817, in run_migrations
    self.get_context().run_migrations(**kw)
  File "/usr/local/lib/python2.7/dist-packages/alembic/runtime/migration.py", line 323, in run_migrations
    step.migration_fn(**kw)
  File "/home/pi/Mycodo/databases/alembic/versions/059a47f950b8_add_option_to_show_hide_tooltips.py", line 21, in upgrade
    batch_op.add_column(sa.Column('hide_tooltips', sa.BOOLEAN))
  File "/usr/lib/python2.7/contextlib.py", line 24, in __exit__
    self.gen.next()
  File "/usr/local/lib/python2.7/dist-packages/alembic/operations/base.py", line 299, in batch_alter_table
    impl.flush()
  File "/usr/local/lib/python2.7/dist-packages/alembic/operations/batch.py", line 82, in flush
    batch_impl._create(self.impl)
  File "/usr/local/lib/python2.7/dist-packages/alembic/operations/batch.py", line 246, in _create
    op_impl.create_table(self.new_table)
  File "/usr/local/lib/python2.7/dist-packages/alembic/ddl/impl.py", line 194, in create_table
    self._exec(schema.CreateTable(table))
  File "/usr/local/lib/python2.7/dist-packages/alembic/ddl/impl.py", line 118, in _exec
    return conn.execute(construct, *multiparams, **params)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 914, in execute
    return meth(self, multiparams, params)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/sql/ddl.py", line 68, in _execute_on_connection
    return connection._execute_ddl(self, multiparams, params)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 968, in _execute_ddl
    compiled
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1146, in _execute_context
    context)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1341, in _handle_dbapi_exception
    exc_info
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/util/compat.py", line 202, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1139, in _execute_context
    context)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/default.py", line 450, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'\nCREATE TABLE _alembic_batch_temp (\n\tid INTEGER NOT NULL, \n\tdismiss_notification BOOLEAN, \n\tforce_https BOOLEAN, \n\thide_alert_info BOOLEAN, \n\thide_alert_success BOOLEAN, \n\thide_alert_warning BOOLEAN, \n\tlanguage TEXT, \n\tlogin_message TEXT, \n\trelay_usage_cost FLOAT, \n\trelay_usage_currency TEXT, \n\trelay_usage_dayofmonth INTEGER, \n\trelay_usage_volts INTEGER, \n\trelay_usage_report_gen BOOLEAN, \n\trelay_usage_report_span TEXT, \n\trelay_usage_report_day INTEGER, \n\trelay_usage_report_hour INTEGER, \n\tstats_opt_out BOOLEAN, \n\thide_tooltips BOOLEAN, \n\tPRIMARY KEY (id), \n\tCHECK (dismiss_notification IN (0, 1)), \n\tCHECK (force_https IN (0, 1)), \n\tCHECK (hide_alert_info IN (0, 1)), \n\tCHECK (hide_alert_success IN (0, 1)), \n\tCHECK (hide_alert_warning IN (0, 1)), \n\tCHECK (relay_usage_report_gen IN (0, 1)), \n\tCHECK (stats_opt_out IN (0, 1)), \n\tCHECK (hide_tooltips IN (0, 1)), \n\tCHECK (hide_tooltips IN (0, 1)), \n\tUNIQUE (id)\n)\n\n']
(env) pi@raspi:~/Mycodo/databases$ 

```

etiology, 2017-03-24:
You should try the alembic command as admin or ```chmod``` the DB so that you have write access. 

```(env) pi@raspi:~/Mycodo/databases$ **sudo** alembic upgrade head```

Ericktz, 2017-03-24:
```
pi@raspi:~$ sudo sqlite3 ~/Mycodo/databases/mycodo.db
SQLite version 3.8.7.1 2014-10-29 13:59:56
Enter ".help" for usage hints.
sqlite> DELETE FROM alembic_version;
sqlite> .exit
pi@raspi:~$ source ~/Mycodo/env/bin/activate
(env) pi@raspi:~$ cd ~/Mycodo/databases
(env) pi@raspi:~/Mycodo/databases$ sudo alembic upgrade head
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.
INFO  [alembic.runtime.migration] Running upgrade  -> 059a47f950b8, add option to show/hide tooltips
```

after reboot the deamon startup and no error in login ni live ny anyohter save one, in live graph get erro 500 and error.log of apache:

```

pi@raspi:~$ tail -2000 /var/log/apache2/error.log
[Fri Mar 24 01:25:58.845511 2017] [ssl:warn] [pid 1176:tid 1995685888] AH01906: 127.0.1.1:443:0 server certificate is a CA certificate (BasicConstraints: CA == TRUE !?)
[Fri Mar 24 01:25:58.845684 2017] [ssl:warn] [pid 1176:tid 1995685888] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Fri Mar 24 01:25:58.846514 2017] [wsgi:warn] [pid 1176:tid 1995685888] mod_wsgi: Compiled for Python/2.7.8.
[Fri Mar 24 01:25:58.846554 2017] [wsgi:warn] [pid 1176:tid 1995685888] mod_wsgi: Runtime using Python/2.7.9.
[Fri Mar 24 01:25:58.848514 2017] [mpm_event:notice] [pid 1176:tid 1995685888] AH00489: Apache/2.4.10 (Raspbian) OpenSSL/1.0.1t mod_wsgi/4.3.0 Python/2.7.9 configured -- resuming normal operations
[Fri Mar 24 01:25:58.848635 2017] [core:notice] [pid 1176:tid 1995685888] AH00094: Command line: '/usr/sbin/apache2'
[Fri Mar 24 01:31:20.059767 2017] [mpm_event:notice] [pid 1176:tid 1995685888] AH00491: caught SIGTERM, shutting down
[Fri Mar 24 01:31:37.034851 2017] [ssl:warn] [pid 1106:tid 1995472896] AH01906: 127.0.0.1:443:0 server certificate is a CA certificate (BasicConstraints: CA == TRUE !?)
[Fri Mar 24 01:31:37.040960 2017] [ssl:warn] [pid 1106:tid 1995472896] AH01909: 127.0.0.1:443:0 server certificate does NOT include an ID which matches the server name
[Fri Mar 24 01:31:37.117655 2017] [ssl:warn] [pid 1110:tid 1995472896] AH01906: 127.0.0.1:443:0 server certificate is a CA certificate (BasicConstraints: CA == TRUE !?)
[Fri Mar 24 01:31:37.117781 2017] [ssl:warn] [pid 1110:tid 1995472896] AH01909: 127.0.0.1:443:0 server certificate does NOT include an ID which matches the server name
[Fri Mar 24 01:31:37.161432 2017] [wsgi:warn] [pid 1110:tid 1995472896] mod_wsgi: Compiled for Python/2.7.8.
[Fri Mar 24 01:31:37.161530 2017] [wsgi:warn] [pid 1110:tid 1995472896] mod_wsgi: Runtime using Python/2.7.9.
[Fri Mar 24 01:31:37.166134 2017] [mpm_event:notice] [pid 1110:tid 1995472896] AH00489: Apache/2.4.10 (Raspbian) OpenSSL/1.0.1t mod_wsgi/4.3.0 Python/2.7.9 configured -- resuming normal operations
[Fri Mar 24 01:31:37.166282 2017] [core:notice] [pid 1110:tid 1995472896] AH00094: Command line: '/usr/sbin/apache2'
[Fri Mar 24 01:33:06.678861 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358] [2017-03-24 01:33:06,666] ERROR in app: Exception on /graph [GET]
[Fri Mar 24 01:33:06.679020 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358] Traceback (most recent call last):
[Fri Mar 24 01:33:06.679041 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1982, in wsgi_app
[Fri Mar 24 01:33:06.679061 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]     response = self.full_dispatch_request()
[Fri Mar 24 01:33:06.679080 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1614, in full_dispatch_request
[Fri Mar 24 01:33:06.679098 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]     rv = self.handle_user_exception(e)
[Fri Mar 24 01:33:06.679116 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1517, in handle_user_exception
[Fri Mar 24 01:33:06.679133 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]     reraise(exc_type, exc_value, tb)
[Fri Mar 24 01:33:06.679151 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1612, in full_dispatch_request
[Fri Mar 24 01:33:06.679169 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]     rv = self.dispatch_request()
[Fri Mar 24 01:33:06.679186 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1598, in dispatch_request
[Fri Mar 24 01:33:06.679206 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]     return self.view_functions[rule.endpoint](**req.view_args)
[Fri Mar 24 01:33:06.679260 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[Fri Mar 24 01:33:06.679278 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]     return func(*args, **kwargs)
[... cut ...]
[Fri Mar 24 01:33:06.679326 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 1020, in dict_custom_colors
[Fri Mar 24 01:33:06.679344 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358]     'name': relay.name,
[Fri Mar 24 01:33:06.679359 2017] [wsgi:error] [pid 1115:tid 1945596976] [remote 192.168.20.104:42358] AttributeError: 'NoneType' object has no attribute 'name'
pi@raspi:~$ 
```
after this i put in progress the new update 5.0.12...

etiology, 2017-03-24:
```python
            for each_set in each_graph.relay_ids.split(','):
                relay_unique_id = each_set.split(',')[0]
                relay = Relay.query.filter_by(
                    unique_id=relay_unique_id).first()
                if (index < len(each_graph.relay_ids.split(',')) and
                        len(colors) > index_sum + index):
                    color = colors[index_sum+index]
                else:
                    color = '#FF00AA'
                total.append({
                    'unique_id': relay_unique_id,
                    'name': relay.name,
                    'measure': 'relay duration',
                    'color': color})
                index += 1
            index_sum += index
```

The relay wasn't found using the ```relay_unique_id```. It may be useful to drop in some debug logging here. Its been awhile since I've gone over the code but in case it is t known you can drop in debugging statements like this:

```python
from flask import current_app

# then later when you want to log something
current_app.logger.debug("This is shown when your app.config['LOGGING'] level is set to DEBUG")
current_app.logger.info("this is an info statement")
current_app.logger.warn("A warning")
current_app.logger..error("an error")
```

I try and avoid depending on string tokenization and parsing to extract keys but for the moment that's how the config is working right now. I'd log the string before it's parsed and after it to see what's going on here. You could also run your debugger and mark this spot 
Likely there is a unit test that calls this area so you can debug it using an ide on better hardware than the pi.

kizniche, 2017-03-24:
@Ericktz Can you attach mycodo.db? You can delete the users table if you're worried about presenting your password hash, but keep in mind there are no passwords stored in plain text.

> [Fri Mar 24 01:52:58.463232 2017] [wsgi:error] [pid 3295:tid 1937904688] [remote 192.168.20.104:32882]     controller=gettext("Relay"))
> [Fri Mar 24 01:52:58.463247 2017] [wsgi:error] [pid 3295:tid 1937904688] [remote 192.168.20.104:32882] UnicodeEncodeError: 'ascii' codec can't encode character u'\\xe9' in position 3: ordinal not in range(128)

This is a strange error. Never seen this at this part of the code.

> The relay wasn't found using the relay_unique_id.

@etiology is correct here. I now see that situation can arise after saving a graph with a relay and then deleting that relay. I'll investigate later how to properly deal with this scenario.

Ericktz, 2017-03-24:
the db was renamed adding .txt at the end
[mycodo.db.txt](https://github.com/kizniche/Mycodo/files/868289/mycodo.db.txt)

kizniche, 2017-03-24:
I see the issue. I'll post a screenshot. I'll investigate later why this happened. In the meantime, you can delete the crazy entry in graph 2's relay_ids, save, upload, and it should resolve your issue. Graph with ID 3 is how relay IDs are supposed to be saved to relay_ids.

[image]

kizniche, 2017-03-24:
If you can give any insight on the differences between how you set up graphs 2 and 3, it would be helpful in tracking down the issue. There was obviously something different about how you configured them for one to work correctly and the other not to.

kizniche, 2017-03-24:
Reproducing the issue with exact steps is what I'm looking for, specifically.

kizniche, 2017-03-24:
I notice you're also using the Spanish language in Mycodo (I'm inquisitive, so I was looking for other issues in your database). How do the translations look? Any issues you've found?

Ericktz, 2017-03-24:
uff 
Well, it will be difficult to reproduce the problem, I have no memory of why I had two graphics for the RelayBomb, (since I have only one relay running, the other I stopped using it because it was not necessary yet), in fact I think Remember that I had removed that graph (the con 2) when I created it from the frontend.
I've followed the steps, I've removed the slot that was in relay_ids, and now there are no problems in the live graphics.
But on the relay page, they continue to throw an error 500 when adding a new relay, and / or deleting an existing relay.
```
[Fri Mar 24 12:40:27.268267 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882] [2017-03-24 12:40:27,266] ERROR in app: Exception on /relay [POST]
[Fri Mar 24 12:40:27.268392 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882] Traceback (most recent call last):
[Fri Mar 24 12:40:27.268413 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1982, in wsgi_app
[Fri Mar 24 12:40:27.268433 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]     response = self.full_dispatch_request()
[Fri Mar 24 12:40:27.268451 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1614, in full_dispatch_request
[Fri Mar 24 12:40:27.268470 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]     rv = self.handle_user_exception(e)
[Fri Mar 24 12:40:27.268488 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1517, in handle_user_exception
[Fri Mar 24 12:40:27.268507 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]     reraise(exc_type, exc_value, tb)
[Fri Mar 24 12:40:27.268525 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1612, in full_dispatch_request
[Fri Mar 24 12:40:27.268543 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]     rv = self.dispatch_request()
[Fri Mar 24 12:40:27.268605 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1598, in dispatch_request
[Fri Mar 24 12:40:27.268624 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]     return self.view_functions[rule.endpoint](**req.view_args)
[Fri Mar 24 12:40:27.268641 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[Fri Mar 24 12:40:27.268657 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]     return func(*args, **kwargs)
[Fri Mar 24 12:40:27.268673 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 708, in page_relay
[Fri Mar 24 12:40:27.268689 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]     flaskutils.relay_add(form_add_relay)
[Fri Mar 24 12:40:27.268705 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]   File "/var/www/mycodo/mycodo/flaskutils.py", line 1717, in relay_add
[Fri Mar 24 12:40:27.268721 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882]     controller=gettext("Relay"))
[Fri Mar 24 12:40:27.268737 2017] [wsgi:error] [pid 1208:tid 1963521072] [remote 200.14.84.157:32882] UnicodeEncodeError: 'ascii' codec can't encode character u'\\xe9' in position 3: ordinal not in range(128)
```
As for the translation into Spanish, there are still texts that are not translated, for example in pages:
 Live
· Sensor acquisition intervals
· The text for "Measurements / Timstamps" translation would be better -> "Valores ​​/ Último Registro"

Graph
· The days and names of the variables of the sensors, when passing the cursor over the graphs.
· Texts: Reorder and (relay duration), by expanding the edition of a graph.

Graph-async
· Descriptive text
· Submit button

Sensor
· The list of accepted sensors
· Descriptive text by passing the cursor above I2C Address, and I2C Bus,
· The text "Pre Relay Duration."

Relay
· In the box of Amperes, "current drawing", wrongly translated, should read "Corriente"

Usage
· Leave "Generated Fri Mar 24 13:01:21 2017" Should read: "Generated: Friday 24 Mar 2017 - 13:01:21"

Lcd
· descriptive text

Camera
·
Time-lapse: Running -> Time-lapse: Enabled
Start time: 2017-03-18 11:54:12 -> Start: XX
End time: 2017-03-28 11:54:12 -> Final: XX
Interval: 86400.0 seconds -> Intervals
Next Capture: 2017-03-25 11:54:12 -> Next capture
Next capture number: 7 -> Total catches

Export
· Untranslated descriptive text

Help
· All untranslated

Setting / general
· Interval box, (monthly, weekly, daily)

[... 29 lines cut ...]
[Fri Mar 24 13:32:52.592255 2017] [wsgi:error] [pid 1173:tid 1938437168] [remote 200.14.84.157:0]     response = self.full_dispatch_request()
[Fri Mar 24 13:32:52.592272 2017] [wsgi:error] [pid 1173:tid 1938437168] [remote 200.14.84.157:0]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1614, in full_dispatch_request
[... cut ...]
When I add a sensor, I get an error message on the front end:

`Error: Agregar Sensor: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE displayorder SET sensor=? WHERE displayorder.id = ?'] [parameters: ('1,2,None', 1)]`

The same when try to change the language:

`Error: Modificar Configuracion General: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE misc SET language=?, relay_usage_cost=?, relay_usage_dayofmonth=? WHERE misc.id = ?'] [parameters: (u'en', 0.05, u'15', 1)]`

Ericktz, 2017-03-24:
OK, I solved the problem:
The problem that it could not eliminate relays and change language,  was that when uploading the new database, it was with user and group pi, so I had to change them to mycodo:
`pi@raspi:~/Mycodo/databases $ sudo chown mycodo:mycodo mycodo.db`
Now everything seems to be working well, at the moment :D

kizniche, 2017-03-24:
> I had removed that graph (the con 2) when I created it from the frontend.

Well, you're right about Graph 2 not being displayed. It seems this may have been a bug from a lower version of 5.0, that has been resolve by now. Here is your display order, not showing Graph 2, but graph 2 still existing in the graph table (you should delete the table entry).

[image]

> UnicodeEncodeError: 'ascii' codec can't encode character u'\\xe1' in position 2: ordinal not in range(128)

Not sure about this error

kizniche, 2017-03-24:
Thanks for the language issues. I'll look at them later when I have a chance.

kizniche, 2017-03-24:
> UnicodeEncodeError: 'ascii' codec can't encode character u'\xe1' in position 2: ordinal not in range(128)

I've done some testing and figured out this issue. This was something I noticed a while ago but didn't think it was a prevalent as it is, but it is far-reaching and disastrous if not using the English language. I'll see if I can fix it all and push 5.0.13 by the end of the day.

[2 more replies]

closed 2017-03-26 by kizniche