kizniche/Mycodo #220: [Suggestion] Time interval units

Ericktz, 2017-03-24:
It would be easier to interpret task time intervals in other units, for example when setting a timer, or timelapse, where you want to run in intervals of days-hours, not just seconds, for example in my case , I have a time lapse every day, 86400.0 seconds, it would be better to have put X day X Hour, X min, etc, and that the system made the calculation of how many seconds correspond.

kizniche, 2017-03-24:
The issues I see with this is this would require either:

1. Removing the ```seconds``` row of the database in favor of ```days```, ```hours```, ```minutes```, and ```seconds``` rows.

or

2. Converting days, hours, minutes, and seconds to seconds, then converting back for display purposes. This would inevitably change what you entered to different values. For instance, input: 70 seconds, save, page loads: 1 minute, 10 seconds.

etiology, 2017-03-25:
Which database table is that coming from?  If it's [this part](https://github.com/kizniche/Mycodo/blob/1af3a2c9f4c0975a1c2e36b986653f45ceb7cb92/mycodo/databases/models/method.py#L25) then what about using datetime's [timedelta](https://docs.python.org/3/library/datetime.html#timedelta-objects) object?  

If you want to create a a new method entry using **Hours**, **minutes**, and **second** you can create an alternative constructor to handle that:
```python 
class MethodData(CRUDMixin, db.Model):
    __tablename__ = "method_data"
    id = db.Column(db.Integer, unique=True, primary_key=True)
    method_id = db.Column(db.Integer, db.ForeignKey('method.id'), default=None)
    time_start = db.Column(db.Text, default=None)
    time_end = db.Column(db.Text, default=None)
    duration_sec = db.Column(db.Float, default=None)
    relay_id = db.Column(db.Integer, db.ForeignKey('relay.id'), default=None)
    relay_state = db.Column(db.Text, default=None)
    relay_duration = db.Column(db.Float, default=None)
    setpoint_start = db.Column(db.Float, default=None)
    setpoint_end = db.Column(db.Float, default=None)
    amplitude = db.Column(db.Float, default=None)
    frequency = db.Column(db.Float, default=None)
    shift_angle = db.Column(db.Float, default=None)
    shift_y = db.Column(db.Float, default=None)
    x0 = db.Column(db.Float, default=None)
    y0 = db.Column(db.Float, default=None)
    x1 = db.Column(db.Float, default=None)
    y1 = db.Column(db.Float, default=None)
    x2 = db.Column(db.Float, default=None)
    y2 = db.Column(db.Float, default=None)
    x3 = db.Column(db.Float, default=None)
    y3 = db.Column(db.Float, default=None)

    def __reper__(self):
        return "<{cls}(id={s.id})>".format(s=self, cls=self.__class__.__name__)
    
    @classmethod
    def create_with_timedelta(cls, duration_delta, **kwargs):
        """ create a MethodData model using a timedelta object for duration """
        config = {k:v for k, v in kwargs.items() if hasattr(cls, v)}  # or something like this... 
        return cls(duration_sec=duration_delta.seconds, **config).save()
``` 

Then you can just create your timedelta and pass into the constructor:
```python
    # example of endpoint using a flask_restful resource class:
    from flask_restful import Resource
    from flask import request
    import datetime as dt

    class ConfigureMethod(Resource):
        """ An endpoint for processing a UI submission that sets the method criteria """
        def post(self):
            """ Creates a MethodData model from a JSON message """
            duration = dt.timedelta(hours=request.json.get('hours'),
                                                    minute=request.json.get('minutes'),
                                                    seconds=request.json.get('seconds))
            method = MethodData.create_with_timedelta(duration_delta=duration)

```

etiology, 2017-03-25:
I guess you could even just create a constructor with Hours, minutes, second:
```python
    from datetime import timedelta

    # a method on MethodData
    @classmethod
    def from_time(cls, hours=None, minutes=None, seconds, **kwargs):
        """ creates a MethodData model with a duration explicitly set """
        config = {k:v for k, v in kwargs.items() if k not in ['hours', 'minutes', 'seconds']}
        return cls(duration_sec=timedelta(hours=hours, 
                                                                  minutes=minutes,
                                                                   seconds=seconds).seconds).save()
```

etiology, 2017-03-25:
A few properties on your model could handle your conversions back to the UI.  Your model could easily have **duration_hours**, **duration_minutes**, **duration_seconds** without messing with the DB.  Same with the relay times

Ericktz, 2017-03-30:
I think it is not necessary to modify the database model, just modify the frontend pages, it would only be necessary three selection boxes with the option: hour, minutes, seconds. And give a function to the submit button to convert that to seconds, and continue using the same database

closed 2020-07-23 by kizniche