kizniche/Mycodo #221: Server error on saving Sensor Conditionals

cohe4ko, 2017-03-24:
## Mycodo Issue Report:
  - Mycodo Version: 5.0.11
 

### Errors
Server error 500  

 [Fri Mar 24 23:54:52.518760 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590] ========= Remote Traceback (1) =========
[Fri Mar 24 23:54:52.518775 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590] Traceback (most recent call last):
[Fri Mar 24 23:54:52.518790 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590]   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 305, in _dispatch_request
[Fri Mar 24 23:54:52.518808 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590]     res = self._HANDLERS[handler](self, *args)
[Fri Mar 24 23:54:52.518824 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590]   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 535, in _handle_call
[Fri Mar 24 23:54:52.518841 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590]     return self._local_objects[oid](*args, **dict(kwargs))
[Fri Mar 24 23:54:52.518856 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590]   File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 198, in exposed_refresh_sensor_conditionals
[Fri Mar 24 23:54:52.518886 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590]     cond_id)
[Fri Mar 24 23:54:52.518901 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590]   File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 550, in refresh_sensor_conditionals
[Fri Mar 24 23:54:52.518918 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590]     return self.controller['Sensor'][sensor_id].setup_sensor_conditionals(cond_mod, cond_id)
[Fri Mar 24 23:54:52.518934 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590] TypeError: setup_sensor_conditionals() takes at most 2 arguments (3 given)
[Fri Mar 24 23:54:52.518950 2017] [wsgi:error] [pid 825:tid 1929831472] [remote:41590] 

### Steps to Reproduce the issue:
 1. Add new Conditional to Sensor 
 2. Save Conditional

kizniche, 2017-03-24:
v5.0.13 was just released. The changelog shows this issue resolved in v5.0.12.

kizniche, 2017-03-24:
Thanks for the report, but please try to upgrade to the latest version before submitting issues. I try to be on top of major issues like this.

closed 2017-03-24 by kizniche

cohe4ko, 2017-03-24:
I installed the whole system only yesterday. I could not suggest that you made 2 updates for 24 hours ;)
You're great :)

kizniche, 2017-03-24:
haha Yes, I've been making a lot of updates recently because v5.0 just came out. There were a torrent of issues that came in, and a number of issues I caught before anyone noticed ;)

cohe4ko, 2017-03-24:
There was one more bug - when I tried to add the new live graph with sensors and relays - after that action /graph gets server error always. Only editing of sqllite base helps in this case.

kizniche, 2017-03-24:
Upgrade to 5.0.13, and if the issue still exists, create a new issue for it.