kizniche/Mycodo #223: Relay clicking every minute.

cohe4ko, 2017-03-25:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.13
 
#### Problem Description
Timers and Conditionals have one common issue.
Every minute it has relay clicking off-on. It's not a problem for ventilation. But for lighting with metal-halogen lamps - it becomes a huge issue.

kizniche, 2017-03-25:
It shouldn't do that. Can you provide exact steps to reproduce the issue? I'll investigate later.

cohe4ko, 2017-03-25:
Something like this:
[image]
[image]

etiology, 2017-03-25:
What about creating a system for using a dimmer instead?  I would not have it be completely controlled by software but if you can have a dimmer control circuit that accepts a gradient value for light output then it could handle the implementation. 

Leaving it to software to directly control the light is a bad idea. This shound be an isolated circuit that takes instruction periodically from the server.

cohe4ko, 2017-03-25:
I don't think that adding one more controller to the system - it's a good idea.
And why software should not control the light?

cohe4ko, 2017-03-25:
Previously I wrote the python script that controlled whole system (watering, light, windows (open/close), ventilation) basing on temperature, humidity, lighting conditions. And that script was reliable enough. but without stats, graphs, and money counting and without GUI.

kizniche, 2017-03-25:
How I set up my lights is to have a timer set to turn on, for example, from 8:00 - 20:00, and then another timer to turn off at 20:01. This ensures the lights are always on during that period (even after a power failure/return), and the off timer will turn the light off, of course.

kizniche, 2017-03-25:
I have to go for now, but I can review all this in more detail later.

etiology, 2017-03-25:
Well think of all of the issues that can happen with that communication. You can have lag or software crashes etc. 

Think of how your laptop work. If it's a MacBook you have a System Management Controller that handle the management of power in the system. Now this controls anything from battery charging to sleep behaviors. 
Well software has some control but it's a conversation, not a direct control. If the batter is in danger of overcharging the SMC deals with it. It does not require asking the operating system for permission not to catch on fire. 

It's a form of encapsulation which is a common patter in software design. Software developer learned it from other areas of study and we see it in nature all of the time because it's a sure way of insulating risk and minimizing dependence.

cohe4ko, 2017-03-25:
I understand you and you're right. But it depends on the project. My project it's small (50 m2) greenhouse near my house.

cohe4ko, 2017-03-25:
And my relays by default (on power failure) are off :) And I'm using it only on not very critical systems. the worse result's that I can get here - is the killed cucumbers and tomatoes. or large electricity bill :)

kizniche, 2017-03-25:
Looking at the conditional you set up and the graph (screenshots), you're saying the relay is turning off for a fraction of a second before going back on? Have you confirmed this, or are you speculating based on the graphs?

I ask because the expected behavior is this:

1. Relay 1 turns on for 65 seconds.
2. 60 seconds pass and conditional instructs Relay 1 to turn on for 65 seconds.
3. Relay 1 logs long long the relay had already been on for, then, if on, just leaves Relay 1 on and resets the timer for another 65 seconds.

The relay controller, when in debug mode, should indicate something to the effect of "Relay 1 instructed to turn on for 65 seconds, but it's already on for 65 seconds. Logging how long it's been on (60 seconds) and resetting timer for 65 more seconds".

cohe4ko, 2017-03-25:
I confirm that relay clicks once a minute for a fraction of a second. I'm not basing only on the graph. Graphs are for only for illustration

kizniche, 2017-03-25:
I'll see if I can fix that, shortly.

cohe4ko, 2017-03-25:
Thank you very much ;)

kizniche, 2017-03-25:
I've verified some strange behavior, which could explain what you've been experiencing. Log below. Still investigating.

```
[Sensor Conditional: Conditional Name (6)]                                                              [28/193]temperature: 43.85 (> 0.0 set value).
Conditional Action (4): relay
  Turn relay 5 on for 15.0 seconds.
2017-03-25 17:37:37,208 - mycodo.relay - DEBUG - Relay 5 (T 223) is already on for a duration of 10.0 seconds (w
ith 0.0 seconds remaining). Recording the amount of time the relay has been on (10.0 sec) and updating the on du
ration to 15.0 seconds.
2017-03-25 17:37:37,293 - mycodo.relay - DEBUG - Relay 5 (T 223) turned off.
2017-03-25 17:37:37,405 - mycodo.relay - DEBUG - Relay 5 (T 223) turned off.
2017-03-25 17:37:37,511 - mycodo.relay - DEBUG - Relay 5 (T 223) turned off.
2017-03-25 17:37:37,624 - mycodo.relay - DEBUG - Relay 5 (T 223) turned off.
2017-03-25 17:37:37,736 - mycodo.relay - DEBUG - Relay 5 (T 223) turned off.
2017-03-25 17:37:37,849 - mycodo.relay - DEBUG - Relay 5 (T 223) turned off.
2017-03-25 17:37:37,962 - mycodo.relay - DEBUG - Relay 5 (T 223) turned off.
```

kizniche, 2017-03-25:
I think I just found the issue. I'm running a long test to see if anything strange happens, but it's been working flawlessly the past 10 or so iterations.

cohe4ko, 2017-03-26:
Is there some way to make the following scenario:
from 5am till 8pm if sensor lux is less than 1000 turn on the light.
or:
if lux less then 1000 turn o relay 1 (and relay one has a time period when it can be turned on)

cohe4ko, 2017-03-26:
Or maybe there is some way to use PID but with continuously on and off, without increasing periods?

kizniche, 2017-03-29:
Does the latest release resolve this issue?

kizniche, 2017-03-31:
Have you had a chance to upgrade to the latest version and see if it resolved this issue?

cohe4ko, 2017-03-31:
Yes, I had. But I didn't test it at my greenhouse.

kizniche, 2017-04-11:
Has this issue been fixed?

closed 2017-05-24 by kizniche