kizniche/Mycodo #224: TSL2561 not working

cohe4ko, 2017-03-25:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.13
 
#### Problem Description
I have not original (not Adafruit) tsl2561 with 0x23 i2c address.
I can't set this address in setting.
Even when I tried to hardcode this address in tsl2561.py - the device doesn't work.

kizniche, 2017-03-25:
Have you been able to run any code to get it working (non-Mycodo)?

kizniche, 2017-03-25:
My TSL2561 sensor is working with Mycodo 5.0.13, btw

cohe4ko, 2017-03-25:
This code works for me:
```
import smbus
bus = smbus.SMBus(1)
data = bus.read_i2c_block_data(0x23,0x20) # 0x23- i2c addres, 0x20 - ONE_TIME_HIGH_RES_MODE_1
lux = ((data[1] + (256 * data[0])) / 1.2)
print (lux)
```

kizniche, 2017-03-25:
> Even when I tried to hardcode this address in tsl2561.py

How were you ding this? Can you paste your code?

cohe4ko, 2017-03-25:
I found it here:
https://github.com/ControlEverythingCommunity/TSL2561/blob/master/Python/TSL2561.py
and then here:
http://www.raspberrypi-spy.co.uk/2015/03/bh1750fvi-i2c-digital-light-intensity-sensor/
As I understand it's some modification.

cohe4ko, 2017-03-25:
Oh I found what the problem:
there is two different sensors: TLS2561 and BH1750. 
I bought BH1750 with a marking of TLS2561. China is China :)

kizniche, 2017-03-25:
Ah. That's easy then. I can just add the BH1750 to Mycodo.

cohe4ko, 2017-03-25:
That's would be great! :)

kizniche, 2017-03-25:
Can you test this code? It appears more feature-rich ;)

https://gist.github.com/oskar456/95c66d564c58361ecf9f

cohe4ko, 2017-03-25:
Yes, It works. but you need to make some way to chose sensitivity. 
`python test.py 
Sensitivity: 69
Low Res  Light Level : 3.33 lx
HighRes  Light Level : 0.00 lx
HighRes2 Light Level : 0.00 lx
--------
Sensitivity: 79
Low Res  Light Level : 5.82 lx
HighRes  Light Level : 0.00 lx
HighRes2 Light Level : 0.00 lx
--------
Sensitivity: 89
Low Res  Light Level : 5.17 lx
HighRes  Light Level : 0.00 lx
HighRes2 Light Level : 0.00 lx
--------
Sensitivity: 99
Low Res  Light Level : 4.65 lx
HighRes  Light Level : 0.00 lx
HighRes2 Light Level : 0.00 lx
--------
Sensitivity: 109
Low Res  Light Level : 6.33 lx
HighRes  Light Level : 0.00 lx
HighRes2 Light Level : 0.00 lx
--------
Sensitivity: 119
Low Res  Light Level : 5.80 lx
HighRes  Light Level : 0.00 lx
HighRes2 Light Level : 0.00 lx
--------
`

kizniche, 2017-03-25:
No prob. I got that covered. I'll push a new release for you to upgrade to to test in a few.

kizniche, 2017-03-25:
The ```HighRes``` and ```HighRes2``` Light Levels are at 0. Is that because your light levels are very low where your sensor is? Can you shine a light on it and rerun the code?

cohe4ko, 2017-03-25:
It's night here :)
And my greenhouse with raspberry pi is far from me. 
I just remotely turned on lighting there and the light from lamps don't shine to sensor directly. that's why is such situation.

kizniche, 2017-03-25:
I think I did it. I'm testing all I can right now, without having the sensor. I'll push an update (5.0.14) before I head off to dinner.

cohe4ko, 2017-03-25:
That's great :) Thank you once more.

kizniche, 2017-03-26:
I just released v5.0.14 and performed an upgrade with one of my dev systems without error. Upgrade when you can and let me know if the new sensor works.

cohe4ko, 2017-03-26:
One more issue - there is no bh1750 sensor in list of available sensors in live graphs.

cohe4ko, 2017-03-26:
Sensor works properly.

kizniche, 2017-03-26:
Good to hear! I'll include the update to fix the graph selection in the next release. In the meantime, you can change line [824 of flaskutils.py](https://github.com/kizniche/Mycodo/blob/20491c52a6f41070dc8f4e56f99cee6311d184a4/mycodo/flaskutils.py#L824) from:

```python
        if each_sensor.device in ['CHIRP', 'TSL2561']:
```

to

```python
        if each_sensor.device in ['BH1750', 'CHIRP', 'TSL2561']:
```

Then restart apache (```sudo /etc/init.d/apache2 restart```) and that should fix it.

kizniche, 2017-03-26:
I'm actually quite surprised the sensor worked on the first try ;)

kizniche, 2017-03-26:
I even included the ability to select resolution and sensitivity, which made it more complex.

kizniche, 2017-03-26:
Closing this issue. Reopen it if the code edit above doesn't work or, when the next release comes, it doesn't fix the display list for graphs/exports.

closed 2017-03-26 by kizniche