kizniche/Mycodo #225: USB Camera defaults issue

cohe4ko, 2017-03-25:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.13
 
#### Problem Description
I use external USB logitech web camera.
It works properly with fswebcam
`fswebcam -r 1280x960 --no-banner $filename` 
But when I use opencv driver there are many settings and no way to use defaults. 
I tried maybe half an hour to find proper figures but there is so many combinations :)

kizniche, 2017-03-25:
Try these settings:

[image]

cohe4ko, 2017-03-25:
Much better :) 
Thank you.

closed 2017-03-25 by cohe4ko

kizniche, 2017-03-25:
I should make those the default values when a camera is added.

cohe4ko, 2017-03-25:
It would be perfect.

kizniche, 2017-03-26:
I'll do some testing and consider adding fswebcam to the list of cam library options.