kizniche/Mycodo #226: Issue with saving of conditional settings

cohe4ko, 2017-03-25:
## Mycodo Issue Report:
  - Specific Mycodo Version:5.0.13
 
#### Problem Description
Doesn't saves email and camera selection: 
[image]

kizniche, 2017-03-25:
That functionality is current;y not working. I added the framework, but didn't finish the stuff in the backend. I figured if I allowed it to save, it would give people a false sense of security that it would work. I'll get around to working on that soon (days, weeks in future).

cohe4ko, 2017-03-25:
Ok. Thank you. 
What about conditionals with two or more conditions? if 1- then if2 - then if3 -> do something.

kizniche, 2017-03-25:
That doesn't exist in Mycodo. Maybe in the future.

kizniche, 2017-04-02:
I'm trying to think of a way to implement a new conditional section of Mycodo that allows multiple conditions with ```and``` and ```or``` operators. I still need to figure out how to group them.

For example:

```A and (B or C) and D```

is different than

```(A and B) or (C and D)```

cohe4ko, 2017-04-05:
Maybe you can do this by making this statements manually.
the first step - create few simple conditions:
temp1 = if temperature on sensor1 less than 20C. 
hum1 = if humidity on sensor2 greater than 80%
The second step - manually make formula:
complexCondition = if (hum1 and temp1)

cohe4ko, 2017-04-05:
and I think it can be very good idea to add as a statement relay status. because I can turn on/off relay from outside Mycodo. And this can be one more trigger in use scenario.

closed 2017-04-11 by kizniche