kizniche/Mycodo #228: Stream via picamera

Drakthule, 2017-03-26:
## Mycodo Issue Report:
5.0.14
 
#### Problem Description
Time lapse and still capture seems to work okay, however whenever i try to start the stream using picamera library nothing appears in the browser window. no error of any sort it just does nothing.i checked the daemon log, doesnt seem to indicate any errors.

the camera in question worked fine back in 4.x.x, its listed below
https://www.amazon.com/gp/product/B01ER2SKFS/ref=oh_aui_detailpage_o08_s00?ie=UTF8&psc=1

mycodo log listed below

> 2017-03-25 22:03:11,641 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-03-25 22:03:12,129 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:12,166 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:12,200 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:12,228 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:12,230 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.
2017-03-25 22:03:17,257 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:17,284 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:17,308 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:17,332 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:17,334 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.
2017-03-25 22:03:20,476 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-03-25 22:03:22,366 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:22,389 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:22,440 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:22,467 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:03:22,469 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.
2017-03-25 21:27:26,882 - mycodo - INFO - Mycodo daemon v5.0.14 starting
2017-03-25 21:27:27,374 - mycodo - INFO - Anonymous statistics enabled
2017-03-25 21:27:27,756 - mycodo.relay - INFO - Relay 1 setup on pin 26 and turned OFF (OFF=HIGH)
2017-03-25 21:27:27,757 - mycodo.relay - INFO - Relay 2 setup on pin 6 and turned OFF (OFF=HIGH)
2017-03-25 21:27:27,758 - mycodo.relay - INFO - Relay 3 setup on pin 19 and turned OFF (OFF=HIGH)
2017-03-25 21:27:27,759 - mycodo.relay - INFO - Relay 4 setup on pin 13 and turned OFF (OFF=HIGH)
2017-03-25 21:27:27,760 - mycodo.relay - INFO - Relay controller activated in 94.2 ms
2017-03-25 21:27:27,761 - mycodo - INFO - All activated timer controllers started
2017-03-25 21:27:28,025 - mycodo.sensor_1 - INFO - Activated in 204.2 ms
2017-03-25 21:27:28,263 - mycodo.sensor_2 - INFO - Activated in 176.5 ms
2017-03-25 21:27:28,264 - mycodo - INFO - All activated sensor controllers started
2017-03-25 21:27:28,265 - mycodo - INFO - All activated PID controllers started
2017-03-25 21:27:28,919 - mycodo.lcd_1 - INFO - Activated in 567.1 ms
2017-03-25 21:27:28,920 - mycodo - INFO - All activated LCD controllers started
2017-03-25 21:27:28,921 - mycodo - INFO - Mycodo daemon v5.0.14 started in 2.039 seconds
2017-03-25 21:27:28,930 - mycodo - INFO - 38.692 MB ram in use
2017-03-25 22:05:28,301 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:05:28,353 - mycodo.influxdb - ERROR - ERROR parsing last influx measurement
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/utils/influx.py", line 112, in read_last_influxdb
    number = len(last_measurement['series'][0]['values'])
KeyError: 'series'
2017-03-25 22:36:08,478 - mycodo.timer_3 - INFO - Activated in 42.0 ms
2017-03-25 22:36:10,352 - mycodo.timer_1 - INFO - Activated in 56.8 ms
2017-03-25 22:36:13,696 - mycodo.timer_2 - INFO - Activated in 47.9 ms
2017-03-25 22:36:35,404 - mycodo.timer_1 - INFO - Deactivated in 139.1 ms
2017-03-25 22:38:01,904 - mycodo.timer_2 - INFO - Deactivated in 61.7 ms
2017-03-25 22:38:21,328 - mycodo.timer_1 - INFO - Activated in 46.4 ms
2017-03-25 22:38:24,644 - mycodo.timer_2 - INFO - Activated in 89.5 ms
2017-03-26 01:18:39,997 - mycodo.timer_2 - INFO - Deactivated in 96.1 ms
2017-03-26 01:18:44,115 - mycodo.timer_2 - INFO - Activated in 81.3 ms
[image]

kizniche, 2017-03-26:
I haven't been able to test streaming since the 5.0 release because my only pi cam has been in a timelapse for the past 3 months (and is projected to be for another 2 months). Unless I get another pi cam, I won't be able to properly test. I'll see what I can do without testing.

kizniche, 2017-03-26:
I think I sound the camera stream issue. I can't use my camera to test, so I may ask for your assistance in helping getting it work.

kizniche, 2017-03-26:
I pushed an update that should restore streaming functionality. If you want to test this, open [camera.html](https://github.com/kizniche/Mycodo/blob/60e4666134ff4b76befef5766cd51f5f1409a760/mycodo/mycodo_flask/templates/pages/camera.html#L90-L95) and add the following code as it is and where it appears in the linked file, then save, and restart the web interface (```sudo /etc/init.d/apache2 restart```).

```html
        {%- if each_camera.stream_started -%}
          <div style="padding-bottom: 1em">
            {{_('Video Stream')}}:
            <img style="max-width: 100%" src="{{url_for('general_routes.video_feed')}}">
          </div>
        {%- endif -%}
```

Drakthule, 2017-03-27:
after wiping camera html and replacing it with the new commit restarting apache and even rebooting, it hasnt made any difference sadly. could you possibly reccomend me a decent webcam that works over usb? the pi camera has a little habit of crashing when fiddled with so wouldnt mind replacing it

kizniche, 2017-10-18:
I finally have my only Pi camera free from the long timelapse that it's been used for, so I can begin testing this issue. Sorry, I would have gotten to this earlier if I hadn't had my camera in use.

kizniche, 2017-11-09:
I'm going to attempt to get the following working in Mycodo: https://github.com/miguelgrinberg/flask-video-streaming

kizniche, 2017-11-10:
I've tested the new code to work streaming USB cameras and I have the code in for the Pi Camera but have yet to test it. I'll hook it up to test soon, and if ti works, which I suspect it will, I'll try to roll out a new release this weekend.

kizniche, 2017-11-10:
> could you possibly reccomend me a decent webcam that works over usb?

I have a Logitech C910 USB camera, but there are a lot that should work with opencv, which is used for streaming now, as well as image/timelapse captures.

kizniche, 2017-11-10:
Just tested and streaming is working again with the Pi Camera.

kizniche, 2017-11-10:
Alright!! I got streaming working for both the Pi Camera and USB cameras AND you can now have more than one camera streaming at once. Very Cool!

closed 2017-11-10 by kizniche