kizniche/Mycodo #231: Error 500 when a sensor are deleted

Ericktz, 2017-03-30:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.15
 
#### Problem Description
error in graph page, when one graph is created and the sensor are removed.

### Errors
```
[Wed Mar 29 22:06:22.400804 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684] [2017-03-29 22:06:22,395] ERROR in app: Exception on /graph [GET]
[Wed Mar 29 22:06:22.402919 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684] Traceback (most recent call last):
[Wed Mar 29 22:06:22.402958 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1982, in wsgi_app
[Wed Mar 29 22:06:22.402979 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]     response = self.full_dispatch_request()
[Wed Mar 29 22:06:22.402997 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1614, in full_dispatch_request
[Wed Mar 29 22:06:22.403015 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]     rv = self.handle_user_exception(e)
[Wed Mar 29 22:06:22.403033 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1517, in handle_user_exception
[Wed Mar 29 22:06:22.403051 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]     reraise(exc_type, exc_value, tb)
[Wed Mar 29 22:06:22.403068 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1612, in full_dispatch_request
[Wed Mar 29 22:06:22.403086 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]     rv = self.dispatch_request()
[Wed Mar 29 22:06:22.403104 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1598, in dispatch_request
[Wed Mar 29 22:06:22.403122 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]     return self.view_functions[rule.endpoint](**req.view_args)
[Wed Mar 29 22:06:22.403140 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[Wed Mar 29 22:06:22.403157 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]     return func(*args, **kwargs)
[Wed Mar 29 22:06:22.403175 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 318, in page_graph
[Wed Mar 29 22:06:22.403192 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]     dict_colors = dict_custom_colors()
[Wed Mar 29 22:06:22.403209 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 1016, in dict_custom_colors
[Wed Mar 29 22:06:22.403228 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684]     'name': sensor.name,
[Wed Mar 29 22:06:22.403247 2017] [wsgi:error] [pid 7601:tid 1962898480] [remote 192.168.20.104:24684] AttributeError: 'NoneType' object has no attribute 'name'
```

### Steps to Reproduce the issue:
1. add a sensor 
2. activate the sensor
3. create a new graph for this sensor
4. eliminate the sensor
5. in graph page error 500

kizniche, 2017-03-30:
Try deleting all your graphs and readd them and see if the issue persists.

kizniche, 2017-03-30:
I have a fix for this. It will be in the next release. Thanks for the bug report.

commit 5e174f03 (2017-03-30): Fix issue with graph page crashing when non-existent sensor referenced (closes #231)
closed 2017-03-30 by kizniche