kizniche/Mycodo #232: Graph error from version 5.0.14  Internal Server Error

molexuse, 2017-03-31:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.16
 
#### Problem Description
 Graph error from version 5.0.14, add graph whith relay.
 Internal Server Error in graph page.
 Update to 5.0.15 and 5.0.16  not resolve this problem.
 
### Errors
[Fri Mar 31 23:30:45.944704 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870] [2017-03-31 23:30:45,943] ERROR in app: Exception on /graph [GET]
[Fri Mar 31 23:30:45.944832 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870] Traceback (most recent call last):
[Fri Mar 31 23:30:45.944854 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
[Fri Mar 31 23:30:45.944874 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]     response = self.full_dispatch_request()
[Fri Mar 31 23:30:45.944893 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
[Fri Mar 31 23:30:45.944912 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]     rv = self.handle_user_exception(e)
[Fri Mar 31 23:30:45.944929 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
[Fri Mar 31 23:30:45.944948 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]     reraise(exc_type, exc_value, tb)
[Fri Mar 31 23:30:45.944965 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
[Fri Mar 31 23:30:45.944984 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]     rv = self.dispatch_request()
[Fri Mar 31 23:30:45.945002 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
[Fri Mar 31 23:30:45.945021 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]     return self.view_functions[rule.endpoint](**req.view_args)
[Fri Mar 31 23:30:45.945039 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[Fri Mar 31 23:30:45.945089 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]     return func(*args, **kwargs)
[Fri Mar 31 23:30:45.945108 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 318, in page_graph
[Fri Mar 31 23:30:45.945124 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]     dict_colors = dict_custom_colors()
[Fri Mar 31 23:30:45.945140 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 1034, in dict_custom_colors
[Fri Mar 31 23:30:45.945157 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870]     if relay.name is not None:
[Fri Mar 31 23:30:45.945173 2017] [wsgi:error] [pid 1313:tid 1930089520] [remote 192.168.1.26:42870] AttributeError: 'NoneType' object has no attribute 'name'

kizniche, 2017-03-31:
This is a known/duplicate issue. Delete all your graphs and it should resolve the issue.

kizniche, 2017-03-31:
If you cannot access the graph page, then deleting the database (~/Mycodo/databases/mycodo.db), then visiting the web UI, then restarting, will regenerate the database.

kizniche, 2017-03-31:
This is actually the same issue that I thought I had fixed, but my fix was improper. I'm committing a fix that should work and pushing v5.0.17 now.

kizniche, 2017-03-31:
Try upgrading to 5.0.17 before messing with your database. You should be able to upgrade without touching it and be alright. If not, delete the database and start over from scratch (unless you have the ability to edit the database and delete individual entries from your graph table).

closed 2017-03-31 by kizniche

molexuse, 2017-03-31:
Sen`c for fast reaction. 
I upgrade to 5.0.17 and its all ok. graph page work!!!

kizniche, 2017-03-31:
Great! Thanks for the bug report!

commit af15b2d7 (2017-03-31): Fix issue with graph page crashing when non-existent sensor referenced (closes #232)