kizniche/Mycodo #233: Enhancement - Sensor Request: BMP280

Ryobidelacroix, 2017-04-01:
Hi Kyle, 

I have a handful of spare BMP280 sensors lying around. I was wondering if it would be difficult/compatible to add a BMP280 sensor (as compared to a BM**E**280 sensor) to the list of available sensors within Mycodo? 

5.0.17 is looking great! Thanks again for the great work. 

Thanks!

kizniche, 2017-04-01:
Can you test this code?

https://github.com/vitally/BMP280

Ryobidelacroix, 2017-04-01:
That appears to be working

kizniche, 2017-04-01:
Great! I'll add a BMP280 sensor module when I get home later and release v5.0.18.

kizniche, 2017-04-02:
I just added the BMP280 module. I'm going to release v5.0.18 shortly. Upgrade your system and let if know if the new module works after you get a chance to test it.

Ryobidelacroix, 2017-04-02:
My BMP280's are showing an I2C address of 0x76, however adding the sensor in Mycodo will only allow for a 0x77 address.

kizniche, 2017-04-02:
I just pushed a fix with the newest release (v5.0.19). See if that works.

Ryobidelacroix, 2017-04-02:
I have updated to 5.0.20 and I'm getting an error message when trying to activate the sensor:
"Could not activate Sensor controller with ID 7: _'SensorController' object has no attribute 'i2c_address'"_
and _"Sensor controller with ID 7 not found"_

Just out of curiosity I tried adding the sensor as a BM**E**280, and it is returning data - albeit erroneous data.

kizniche, 2017-04-03:
I just pushed a fix that should fix the issue an released v5.0.21

Ryobidelacroix, 2017-04-03:
This is working as expected now. Thanks!

closed 2017-04-03 by Ryobidelacroix