kizniche/Mycodo #234: 5.0.15 crashes without error files, after update to 5.0.21 no data measurement

Ericktz, 2017-04-03:
well, I having this problem some days ago, specifically when I noticed that haven't record after April 2, at 14:55 (this was the last record), when i looking in .../apache2/error.log:
```

[Mon Apr 03 11:13:39.974396 2017] [ssl:warn] [pid 31741:tid 1995915264] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Mon Apr 03 11:13:39.975207 2017] [wsgi:warn] [pid 31741:tid 1995915264] mod_wsgi: Compiled for Python/2.7.8.
[Mon Apr 03 11:13:39.975262 2017] [wsgi:warn] [pid 31741:tid 1995915264] mod_wsgi: Runtime using Python/2.7.9.
[Mon Apr 03 11:13:39.980044 2017] [mpm_event:notice] [pid 31741:tid 1995915264] AH00489: Apache/2.4.10 (Raspbian) OpenSSL/1.0.1t mod_wsgi/4.3.0 Python/2.7.9 configured -- resuming normal operations
[Mon Apr 03 11:13:39.980211 2017] [core:notice] [pid 31741:tid 1995915264] AH00094: Command line: '/usr/sbin/apache2'
[Mon Apr 03 11:13:53.275563 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:13:53,274 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:14:39.436906 2017] [wsgi:error] [pid 31745:tid 1929565232] 2017-04-03 11:14:39,434 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:14:53.275712 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:14:53,275 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:15:39.272320 2017] [wsgi:error] [pid 31745:tid 1929565232] 2017-04-03 11:15:39,271 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:15:53.204503 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:15:53,204 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:16:31.613511 2017] [wsgi:error] [pid 31745:tid 1929565232] 2017-04-03 11:16:31,612 URL for 'inject_mycodo_version' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:16:53.204607 2017] [wsgi:error] [pid 31745:tid 1929565232] 2017-04-03 11:16:53,204 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:17:29.664581 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:17:29,664 URL for 'inject_mycodo_version' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:17:35.866504 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:17:35,865 URL for 'inject_mycodo_version' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:17:48.113610 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:17:48,112 URL for 'inject_mycodo_version' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:17:50.608388 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:17:50,607 URL for 'inject_mycodo_version' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:17:53.098659 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:17:53,098 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:18:52.275791 2017] [wsgi:error] [pid 31745:tid 1929565232] 2017-04-03 11:18:52,275 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:18:53.099319 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:18:53,098 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:19:51.907238 2017] [wsgi:error] [pid 31745:tid 1929565232] 2017-04-03 11:19:51,906 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:19:53.092142 2017] [wsgi:error] [pid 31745:tid 1963119664] 2017-04-03 11:19:53,091 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:20:37.181015 2017] [wsgi:error] [pid 31745:tid 1929565232] 2017-04-03 11:20:37,180 URL for 'inject_mycodo_version' raised and error: Connection refused. Is the daemon running?
[Mon Apr 03 11:20:51.069507 2017] [mpm_event:notice] [pid 31741:tid 1995915264] AH00491: caught SIGTERM, shutting down
```

i follow the instructions in [this site](https://sysadmin.compxtreme.ro/how-to-deal-with-rsa-server-certificate-is-a-ca-certificate-basicconstraints-ca-true-problem/). 
and next I proceed with an upgrade for 5.0.21 with error:

```
...
Requirement already up-to-date: pycparser in /usr/local/lib/python2.7/dist-packages (from cffi>=1.1->bcrypt->-r /home/pi/Mycodo/install/requirements.txt (line 7))
Requirement already up-to-date: decorator in /usr/local/lib/python2.7/dist-packages (from ratelim->geocoder==1.21.0->-r /home/pi/Mycodo/install/requirements.txt (line 16))
Installing collected packages: argparse
  Found existing installation: argparse 1.2.1
    Not uninstalling argparse at /usr/lib/python2.7, outside environment /home/pi/Mycodo/env
Successfully installed argparse-1.4.0

#### Upgrading database with alembic
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.

#### Removing statistics file
rm: cannot remove '/home/pi/Mycodo/databases/statistics.csv': No such file or directory

#### Enabling mycodo startup script
Removed symlink /etc/systemd/system/multi-user.target.wants/mycodo.service.
Removed symlink /etc/systemd/system/mycodo.service.
Created symlink from /etc/systemd/system/mycodo.service to /home/pi/Mycodo/install/mycodo.service.
Created symlink from /etc/systemd/system/multi-user.target.wants/mycodo.service to /home/pi/Mycodo/install/mycodo.service.

#### Compiling Translations
compiling catalog mycodo_flask/translations/ko/LC_MESSAGES/messages.po to mycodo_flask/translations/ko/LC_MESSAGES/messages.mo
compiling catalog mycodo_flask/translations/es/LC_MESSAGES/messages.po to mycodo_flask/translations/es/LC_MESSAGES/messages.mo
compiling catalog mycodo_flask/translations/fr/LC_MESSAGES/messages.po to mycodo_flask/translations/fr/LC_MESSAGES/messages.mo
#### Updating crontab entry
Removing crontab entry ...
Appending to crontab:
-----------------------------------------------
@reboot /usr/local/bin/pigpiod &-----------------------------------------------
To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh mycodo --remove

#### Creating proper users, directories, and permissions
useradd: user 'mycodo' already exists
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `adm'.
The user `mycodo' is already a member of `video'.

[... 73 lines cut ...]
[Mon Apr 03 11:21:41.719868 2017] [wsgi:error] [pid 1193:tid 1963062320]     sensor_period)).raw
[Mon Apr 03 11:21:41.719886 2017] [wsgi:error] [pid 1193:tid 1963062320]   File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 325, in query
[Mon Apr 03 11:21:41.719906 2017] [wsgi:error] [pid 1193:tid 1963062320]     expected_response_code=expected_response_code
[Mon Apr 03 11:21:41.719923 2017] [wsgi:error] [pid 1193:tid 1963062320]   File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 241, in request
[Mon Apr 03 11:21:41.719941 2017] [wsgi:error] [pid 1193:tid 1963062320]     raise e
[Mon Apr 03 11:21:41.726057 2017] [wsgi:error] [pid 1193:tid 1963062320] ConnectionError: HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=SELECT+last%28value%29%0A++++++++++++++++++++++++++++++++++FROM+humidity%0A++++++++++++++++++++++++++++++++++WHERE+device_id%3D%27d7286512-c3cb-4b42-8786-dc87115ad1a7%27%0A++++++++++++++++++++++++++++++++++++++++AND+time+%3E+now%28%29+-+30m%0A+++++++++++++++++++++++++++++++&db=mycodo_db (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x6bd35ef0>: Failed to establish a new connection: [Errno 111] Connection refused',))
[Mon Apr 03 11:21:41.738159 2017] [wsgi:error] [pid 1193:tid 1937896496] 2017-04-03 11:21:41,731 URL for 'last_data' raised and error: HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=SELECT+last%28value%29%0A++++++++++++++++++++++++++++++++++FROM+dewpoint%0A++++++++++++++++++++++++++++++++++WHERE+device_id%3D%27d7286512-c3cb-4b42-8786-dc87115ad1a7%27%0A++++++++++++++++++++++++++++++++++++++++AND+time+%3E+now%28%29+-+30m%0A+++++++++++++++++++++++++++++++&db=mycodo_db (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x [... cut ...]
[Mon Apr 03 11:21:41.746907 2017] [wsgi:error] [pid 1193:tid 1929507888] ConnectionError: HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=SELECT+last%28value%29%0A++++++++++++++++++++++++++++++++++FROM+temperature%0A++++++++++++++++++++++++++++++++++WHERE+device_id%3D%27d7286512-c3cb-4b42-8786-dc87115ad1a7%27%0A++++++++++++++++++++++++++++++++++++++++AND+time+%3E+now%28%29+-+30m%0A+++++++++++++++++++++++++++++++&db=mycodo_db (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x6bdb2c10>: Failed to establish a new connection: [Errno 111] Connection refused',))
```

Ericktz, 2017-04-03:
I do not know how it happened but while I was writing the problem, I go back to check out and all be normal again :/

kizniche, 2017-04-03:
> Connection refused. Is the daemon running?

This is returned when the web UI (frontned) cannot communicate with the daemon (backend). If this is happening, the daemon is most likely not running and "Mycodo" in the top-left of the web UI will be red, because this is the "check" that's occurring and failing.

> Job for mycodo.service failed. See 'systemctl status mycodo.service' and 'journalctl -xn' for details.
> Failed: Error while running post-upgrade script.
> The upgrade has failed: Attempting to revert moving the old Mycodo install.

The upgrade failed because the mycodo daemon could not be restarted, and the install was reverted back. An upgrade cannot occurr if the mycodo daemon isn't functioning properly, so always make sure it's running and responding when performing an upgrade.

> ConnectionError: HTTPConnectionPool(host='localhost', port=8086)

This error occurs when influxdb (measurement database) is not responding, and typically happens at startup. Measurements cannot be retrieved or written while influxdb is starting, and until it fully starts, this is the error it returns. Once started, these errors go away.

kizniche, 2017-04-03:
There's nothing abnormal in that log, and only web UI (frontend) errors will appear, there, not the daemon. You will need to refer to the daemon log for those errors (through web UI or /var/log/mycodo/mycodo.log).

Ericktz, 2017-04-03:
The problem is that there was nothing strange in that logfile:
```
2017-03-31 13:08:31,893 - mycodo - INFO - 47.076 MB ram in use
2017-04-01 13:08:32,081 - mycodo - INFO - 47.076 MB ram in use
2017-04-02 13:08:32,167 - mycodo - INFO - 47.076 MB ram in use
2017-04-03 11:21:31,587 - mycodo - INFO - Mycodo daemon v5.0.21 starting
2017-04-03 11:21:32,094 - mycodo - INFO - Anonymous statistics enabled
```

It is as if he had not registered the error that caused him to stop executing

kizniche, 2017-04-03:
The worst kind of bugs. How long had it been running before the issue?

Ericktz, 2017-04-03:
Had been running since Friday, March 31, and from what I saw in graphics, the last measurement was recorded on April 2 at 2:45 pm, there I think that the daemon stopped and did not register anymore

kizniche, 2017-04-03:
I was meaning, that you have been running Mycodo for weeks and this is the first time the daemon stopped? The code hasn't changed that much in the last few updates that I think would start to cause daemon issues, so I think this may be operating system-related and not Mycodo. I would test some more and if it continually happens, then you should try to narrow down what part of Mycodo is causing it, be selectively turning off (or on) features until you can isolate the problem area.

Ericktz, 2017-04-03:
No no sorry, I was having problems with the daemon for some time ago, but I did not find out what the cause was, I rebooted the raspberry and everything was fixed, I always look for updates to keep the latest version of mycodo. Now I will be more attentive, it usually happened every 3-4 days that worked perfect and then stopped, I suspected that it was DHT-22 sensor problems (I have the AM2302, but I understand they use the same library), but I have not Confirm it. Apparently my problem are similar to  #198

kizniche, 2017-04-03:
I use both the DHT22 and AM2315 and those systems haven't had issues, even running for weeks between updates. So, I would be hard-pressed to believe it's Mycodo. Power supplies can be an issue, specifically those that don't filter out spikes caused by things such as refrigerators, generators, motors, etc. I would change variables (connected sensors, controllers activated, power supplies, locations power supply is plugged in at your location, etc.) and see if you can isolate the issue.

Ryobidelacroix, 2017-04-03:
I'm also having this intermittent issue with 5.0.21. 

When using `sudo service mycodo status -l` I get: 

● mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled)
   Active: failed (Result: exit-code) since Mon 2017-04-03 22:51:47 ACST; 8h ago
  Process: 31674 ExecStop=/var/www/mycodo/env/bin/python /var/www/mycodo/mycodo/mycodo_client.py -t (code=exited, status=1/FAILURE)
  Process: 535 ExecStart=/var/www/mycodo/env/bin/python /var/www/mycodo/mycodo/mycodo_daemon.py (code=exited, status=0/SUCCESS)
 Main PID: 1161 (code=killed, signal=ABRT)

Apr 03 22:51:46 raspberrypi systemd[1]: mycodo.service: main process exited, code=killed, status=6/ABRT
Apr 03 22:51:47 raspberrypi python[31674]: Traceback (most recent call last):
Apr 03 22:51:47 raspberrypi python[31674]: File "/var/www/mycodo/mycodo/mycodo_client.py", line 180, in <module>
Apr 03 22:51:47 raspberrypi python[31674]: daemon_control = DaemonControl()
Apr 03 22:51:47 raspberrypi python[31674]: File "/var/www/mycodo/mycodo/mycodo_client.py", line 55, in __init__
Apr 03 22:51:47 raspberrypi python[31674]: raise Exception("Connection refused. Is the daemon running?")
Apr 03 22:51:47 raspberrypi python[31674]: Exception: Connection refused. Is the daemon running?
Apr 03 22:51:47 raspberrypi systemd[1]: mycodo.service: control process exited, code=exited status=1
Apr 03 22:51:47 raspberrypi systemd[1]: Unit mycodo.service entered failed state.
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

I'll see if I can start the Daemon in debug mode, change power supplies and see how it goes.

kizniche, 2017-04-03:
Not sure about those errors. Perhaps a fresh install could fix it.

closed 2017-10-24 by kizniche