kizniche/Mycodo #235: Not proper relay stats

cohe4ko, 2017-04-05:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.21
 
#### Problem Description
Relay is on whole time but on stats it shown on only for few seconds
[image]

kizniche, 2017-04-05:
Please be a little more descriptive. I don't understand the issue. Use the issue template to submit a steps to reproduce the issue and explain in detail what the issue is.

cohe4ko, 2017-04-05:
I use sensor conditional. If the temperature is higher than 20C - turn on Relay4. Conditional works well. Relay turns on if temperature greater than 20C. but on the graphs I see that relay was on only for 2-5 seconds. It's not true.

kizniche, 2017-04-05:
What is your sensor conditional period?

cohe4ko, 2017-04-05:
[image]

kizniche, 2017-04-05:
Following your steps, I could not replicate the issue. My grapes show 60 second outputs every 60 seconds. Is relay 4 configured to be used anywhere else in the system?

cohe4ko, 2017-04-05:
Yes, It is configured on the other sensor with same conditions and on that sensor temperature is higher too.

kizniche, 2017-04-05:
Turn the other conditional off and test.

cohe4ko, 2017-04-05:
Done - the graph now is ok.

kizniche, 2017-04-05:
What is the period of your second conditional you just turned off?

cohe4ko, 2017-04-05:
but that is wrong behavior. Or not?

cohe4ko, 2017-04-05:
[image]

kizniche, 2017-04-05:
I've never tested more than one conditional using the same relay.

kizniche, 2017-04-05:
You're inherently bound to have conflicts. I'll check it out and see if there's anything I can do about it. Most likely not because it all depends when the conditionals are activated, which will determine their overlap. If one conditional activates the relay, it will record whatever duration the relay has been on, and turn it on for the new duration.

cohe4ko, 2017-04-05:
ok. thank you.

kizniche, 2017-04-05:
I'm not sure if this will break the system, but try this... Create two relays, using the same pin, and set each one up using their own conditional. I'd make sure you're physically at the location where the devices are actuating before attempting this experimental configuration.

kizniche, 2017-04-05:
Alternatively, create this setup to test:

Relay 4: the actual pin for relay 4
Relay 5: pin to free GPIO (unconnected)
Relay conditional: when relay 5 on for 80 sec, relay 4 on for 80 sec.
Sensor conditional 1: relay 4
Sensor conditional 2: relay 5

closed 2017-05-24 by kizniche