kizniche/Mycodo #236: "Internal Server Error" - When attempting to access 'Sensors' page.

Ryobidelacroix, 2017-04-08:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.21
 
#### Problem Description
Please list:

  - Clicking on 'sensors' tab will show error message: 

"**Internal Server Error**
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."

 
### Errors
  - **I will include crash logs as text files**, however, possible cause shown in Daemon log:

"ERROR - Error raised in session_scope.  Session will be rolled back: db_uri='sqlite:////home/pi/Mycodo/mycodo/../databases/mycodo.db', error='(sqlite3.OperationalError) database is locked [SQL: u'SELECT alembic_version.version_num AS alembic_version_version_num \nFROM alembic_version\n LIMIT ? OFFSET ?'] [parameters: (1, 0)]'"

[LOG_Http.txt](https://github.com/kizniche/Mycodo/files/907290/LOG_Http.txt)
[LOG_Daemon.txt](https://github.com/kizniche/Mycodo/files/907291/LOG_Daemon.txt)

Rebooting does not resolve. 

This is a recent fresh install (2-3 days old)

kizniche, 2017-04-09:
A database lock error occurs when a process is writing to the SQL database file and requires exclusive access to it. This normally takes a fraction of a second and the database is then able to be opened by other processes.

I don't know what happened to your system or database to cause this, but this is the first issue of this happening persistently. I'm not sure how to fix it other than to perhaps google the issue or reinstall Mycodo. One thing I'm pretty sure of, though, is that it is not a Mycodo-specific issue.

kizniche, 2017-04-09:
Your daemon log indicates this error happened once. The next daemon restart it did not happen (i.e. the issue was not present).

Are you saying you can only not access the Sensor page?

kizniche, 2017-04-09:
Looking at your HTTP log, the issue appears to be the code not recognizing any DS18B20 sensor(s) you previously added. Looking into the offending code and working on a fix.

kizniche, 2017-04-09:
I just committed a fix for this issue and released [v5.0.22](https://github.com/kizniche/Mycodo/releases/tag/v5.0.22). Update and let me know if the issue persists. Thanks for the bug report.

Ryobidelacroix, 2017-04-09:
Hi Kyle, 

That's correct; Only the sensor page is affected. It seems I missed a chunk of my Daemon log when pasting it into the txt file. I can confirm it was occurring more than once. 

Thanks for the fix, I'll update when I get home and let you know how it goes.

leoneo, 2017-04-09:
Ok i have find the problem. I have the problem when i change the name of RPI temperature by "température" the graph page don't like "é" special caracter

kizniche, 2017-04-09:
You're replying in the wrong issue, @leoneo 

They are named similarly, though ;)

leoneo, 2017-04-09:
sorry

Ryobidelacroix, 2017-04-09:
Issue resolved!

closed 2017-04-09 by Ryobidelacroix