kizniche/Mycodo #237: "Internal Server Error" - When attempting to access 'Graph' page.

leoneo, 2017-04-09:
## Mycodo Issue Report:
  - Specific Mycodo Version: Specific Mycodo Version: 5.0.22
 
#### Problem Description
Please list:

    Clicking on 'Live Graphs' and 'Asynchronous Graphs' tab will show error message:

"500 Internal Server Error
Internal Server Error

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."
 
### Errors

HTML Log
[Sun Apr 09 07:17:15.984235 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
[Sun Apr 09 07:17:15.984282 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]     rv = self.dispatch_request()
[Sun Apr 09 07:17:15.984328 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
[Sun Apr 09 07:17:15.984375 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]     return self.view_functions[rule.endpoint](**req.view_args)
[Sun Apr 09 07:17:15.984422 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[Sun Apr 09 07:17:15.984469 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]     return func(*args, **kwargs)
[Sun Apr 09 07:17:15.984513 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 303, in page_graph
[Sun Apr 09 07:17:15.984560 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]     sensor_choices = flaskutils.choices_sensors(sensor)
[Sun Apr 09 07:17:15.984608 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]   File "/var/www/mycodo/mycodo/flaskutils.py", line 749, in choices_sensors
[Sun Apr 09 07:17:15.984656 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478]     unit=MEASUREMENT_UNITS[each_measurement]['unit'])
[Sun Apr 09 07:17:15.984701 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478] UnicodeEncodeError: 'ascii' codec can't encode character u'\\xe9' in position 8: ordinal not in range(128)
[Sun Apr 09 09:35:32.440863 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450] [2017-04-09 09:35:32,435] ERROR in app: Exception on /graph [GET]
[Sun Apr 09 09:35:32.441142 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450] Traceback (most recent call last):
[Sun Apr 09 09:35:32.441208 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
[Sun Apr 09 09:35:32.441263 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]     response = self.full_dispatch_request()
[Sun Apr 09 09:35:32.441315 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
[Sun Apr 09 09:35:32.441368 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]     rv = self.handle_user_exception(e)
[Sun Apr 09 09:35:32.441497 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
[Sun Apr 09 09:35:32.441551 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]     reraise(exc_type, exc_value, tb)
[Sun Apr 09 09:35:32.441597 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
[Sun Apr 09 09:35:32.441645 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]     rv = self.dispatch_request()
[Sun Apr 09 09:35:32.441690 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
[Sun Apr 09 09:35:32.441739 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]     return self.view_functions[rule.endpoint](**req.view_args)
[Sun Apr 09 09:35:32.441786 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[Sun Apr 09 09:35:32.441833 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]     return func(*args, **kwargs)
[Sun Apr 09 09:35:32.441878 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 303, in page_graph
[Sun Apr 09 09:35:32.441984 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]     sensor_choices = flaskutils.choices_sensors(sensor)
[Sun Apr 09 09:35:32.442038 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]   File "/var/www/mycodo/mycodo/flaskutils.py", line 749, in choices_sensors
[Sun Apr 09 09:35:32.442119 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450]     unit=MEASUREMENT_UNITS[each_measurement]['unit'])
[Sun Apr 09 09:35:32.442167 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450] UnicodeEncodeError: 'ascii' codec can't encode character u'\\xe9' in position 8: ordinal not in range(128)

This is a recent fresh install (4-5 days old)

kizniche, 2017-04-09:
Do you happen to have a language other than English set in the settings?

leoneo, 2017-04-09:
yes french

leoneo, 2017-04-09:
i have change now in english but the error still

kizniche, 2017-04-09:
I'm unable to replicate the issue.

I've seen this error before when dealing with translations, but I've since fixed the issues I knew of relating to this. What sensors do you have configured?

leoneo, 2017-04-09:
DHT22
BMP180
RPi Temperature

leoneo, 2017-04-09:
I have delet RPI Temperature and now Graphs work ;)

kizniche, 2017-04-09:
I have a system set up with those sensors and I don't have this issue when English or French is set.

I'm also not sure why deleting the RPi Temperature sensor would cause that issue for you. Can you confirm that if you add it again the error returns?

leoneo, 2017-04-09:
Ok i have find the problem. I have the problem when i change the name of RPI temperature by "température" the graph page don't like "é" special caracter

kizniche, 2017-04-09:
Thanks for figuring out the issue. I'll take a look at it closer later and try to figure out a fix.

leoneo, 2017-04-09:
Thank you for your Mycodo system ;)

kizniche, 2017-04-10:
I just released version 5.0.23 that incorporates the fixes for this issue. After you upgrade, let me know if it fixes the issue you were having with using special characters.

closed 2017-04-11 by kizniche

jeffyboz, 2017-04-13:
> [Sun Apr 09 07:17:15.984375 2017] [wsgi:error] [pid 749:tid 1954960432] [remote 192.168.1.56:15478] return self.view_functionsrule.endpoint <x-msg://1/**req.view_args>
> [Sun Apr 09 09:35:32.441739 2017] [wsgi:error] [pid 749:tid 1946571824] [remote 192.168.1.56:14450] return self.view_functionsrule.endpoint <x-msg://1/**req.view_args>

I just upgraded to 5.0.26 and now i’m getting an internal server error when I try to look at live graph:

[Wed Apr 12 06:25:20.437314 2017] [ssl:warn] [pid 662:tid 3070030352] AH01906: 127.0.0.1:443:0 server certificate is a CA certificate (BasicConstraints: CA == TRUE !?)
[Wed Apr 12 06:25:20.440219 2017] [ssl:warn] [pid 662:tid 3070030352] AH01909: 127.0.0.1:443:0 server certificate does NOT include an ID which matches the server name
[Wed Apr 12 06:25:20.453409 2017] [wsgi:warn] [pid 662:tid 3070030352] mod_wsgi: Compiled for Python/2.7.8.
[Wed Apr 12 06:25:20.456044 2017] [wsgi:warn] [pid 662:tid 3070030352] mod_wsgi: Runtime using Python/2.7.9.
[Wed Apr 12 06:25:20.468289 2017] [mpm_event:notice] [pid 662:tid 3070030352] AH00489: Apache/2.4.10 (Raspbian) OpenSSL/1.0.1t mod_wsgi/4.3.0 Python/2.7.9 configured -- resuming normal operations
[Wed Apr 12 06:25:20.498034 2017] [core:notice] [pid 662:tid 3070030352] AH00094: Command line: '/usr/sbin/apache2'
[Wed Apr 12 18:17:26.008695 2017] [ssl:warn] [pid 619:tid 3069514256] AH01906: 127.0.0.1:443:0 server certificate is a CA certificate (BasicConstraints: CA == TRUE !?)
[Wed Apr 12 18:17:26.015539 2017] [ssl:warn] [pid 619:tid 3069514256] AH01909: 127.0.0.1:443:0 server certificate does NOT include an ID which matches the server name
[Wed Apr 12 18:17:27.024067 2017] [ssl:warn] [pid 624:tid 3069514256] AH01906: 127.0.0.1:443:0 server certificate is a CA certificate (BasicConstraints: CA == TRUE !?)
[Wed Apr 12 18:17:27.028055 2017] [ssl:warn] [pid 624:tid 3069514256] AH01909: 127.0.0.1:443:0 server certificate does NOT include an ID which matches the server name
[Wed Apr 12 18:17:27.066177 2017] [wsgi:warn] [pid 624:tid 3069514256] mod_wsgi: Compiled for Python/2.7.8.
[Wed Apr 12 18:17:27.066439 2017] [wsgi:warn] [pid 624:tid 3069514256] mod_wsgi: Runtime using Python/2.7.9.
[Wed Apr 12 18:17:27.143949 2017] [mpm_event:notice] [pid 624:tid 3069514256] AH00489: Apache/2.4.10 (Raspbian) OpenSSL/1.0.1t mod_wsgi/4.3.0 Python/2.7.9 configured -- resuming normal operations
[Wed Apr 12 18:17:27.144277 2017] [core:notice] [pid 624:tid 3069514256] AH00094: Command line: '/usr/sbin/apache2'
[Wed Apr 12 18:50:26.565750 2017] [wsgi:error] [pid 627:tid 3036439600] 2017-04-12 18:50:26,554 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 18:50:26.695283 2017] [wsgi:error] [pid 627:tid 3019662384] 2017-04-12 18:50:26,669 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 18:51:26.373460 2017] [wsgi:error] [pid 627:tid 3011273776] 2017-04-12 18:51:26,346 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 18:51:26.718182 2017] [wsgi:error] [pid 627:tid 3002885168] 2017-04-12 18:51:26,695 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:00:20.221563 2017] [wsgi:error] [pid 627:tid 3036439600] 2017-04-12 19:00:20,199 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:01:21.074579 2017] [wsgi:error] [pid 627:tid 3036439600] 2017-04-12 19:01:21,072 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:02:20.673090 2017] [wsgi:error] [pid 627:tid 3019662384] 2017-04-12 19:02:20,662 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:03:21.112013 2017] [wsgi:error] [pid 627:tid 3011273776] 2017-04-12 19:03:21,109 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:04:21.131455 2017] [wsgi:error] [pid 627:tid 3028050992] 2017-04-12 19:04:21,127 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:05:21.609042 2017] [wsgi:error] [pid 627:tid 3019662384] 2017-04-12 19:05:21,576 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:06:20.550561 2017] [wsgi:error] [pid 627:tid 3011273776] 2017-04-12 19:06:20,521 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:07:20.349068 2017] [wsgi:error] [pid 627:tid 3002885168] 2017-04-12 19:07:20,337 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:08:20.542341 2017] [wsgi:error] [pid 627:tid 3036439600] 2017-04-12 19:08:20,514 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:09:20.387164 2017] [wsgi:error] [pid 627:tid 3028050992] 2017-04-12 19:09:20,353 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:10:20.274430 2017] [wsgi:error] [pid 627:tid 3002885168] 2017-04-12 19:10:20,250 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:11:21.407705 2017] [wsgi:error] [pid 627:tid 3019662384] 2017-04-12 19:11:21,394 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:12:21.227679 2017] [wsgi:error] [pid 627:tid 3019662384] 2017-04-12 19:12:21,213 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:13:20.128826 2017] [wsgi:error] [pid 627:tid 3028050992] 2017-04-12 19:13:20,092 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:14:20.305357 2017] [wsgi:error] [pid 627:tid 3011273776] 2017-04-12 19:14:20,283 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:15:20.499442 2017] [wsgi:error] [pid 627:tid 3019662384] 2017-04-12 19:15:20,479 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:16:21.003104 2017] [wsgi:error] [pid 627:tid 3036439600] 2017-04-12 19:16:20,990 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:19:21.251648 2017] [wsgi:error] [pid 627:tid 3002885168] 2017-04-12 19:19:21,216 URL for 'past_data' raised and error: 'series'
[Wed Apr 12 19:26:21.666381 2017] [wsgi:error] [pid 627:tid 3002885168] 2017-04-12 19:26:21,662 URL for 'past_data' raised and error: 'series'
[... cut ...]
[Wed Apr 12 20:11:18.855642 2017] [wsgi:error] [pid 664:tid 3002983472] [remote 10.0.1.3:59820]     dict_colors = dict_custom_colors()
[Wed Apr 12 20:11:18.855924 2017] [wsgi:error] [pid 664:tid 3002983472] [remote 10.0.1.3:59820]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 1068, in dict_custom_colors
[Wed Apr 12 20:11:18.856080 2017] [wsgi:error] [pid 664:tid 3002983472] [remote 10.0.1.3:59820]     pid_measure = each_set.split(',')[1]
[Wed Apr 12 20:11:18.856250 2017] [wsgi:error] [pid 664:tid 3002983472] [remote 10.0.1.3:59820] IndexError: list index out of range

kizniche, 2017-04-13:
Although the title may be similar to your issue, these are two entirely different issues.

jeffyboz, 2017-04-13:
Ok, I just saw this too 

kizniche <https://github.com/kizniche> commented 12 days ago <https://github.com/kizniche/Mycodo/issues/232#issuecomment-290824274>
If you cannot access the graph page, then deleting the database (~/Mycodo/databases/mycodo.db), then visiting the web UI, then restarting, will regenerate the database.

I’ll try to search issues before replying next time.

Thanks!

kizniche, 2017-04-13:
I'm working on finding the commands to fix your issue. Don't delete your databases yet.

kizniche, 2017-04-13:
In the meantime, please start a new issue, and I'll delete these comments to keep the issues separate.

jeffyboz, 2017-04-13:
Ah, snap - too late

kizniche, 2017-04-13:
okay, that should fix your issue then. I was going to get the commands to delete only the graphs.

kizniche, 2017-04-13:
Also, typically, if an issue has been closed, it has been fixed, especially if there are commit references like midway up this issue.