kizniche/Mycodo #241: AM2315 reading issue

Sabare3, 2017-04-16:
## Mycodo Issue Report:
5.0.27
 
#### Problem Description
Please list:

Issues with reading humidity and temperature from AM2315
 
### Errors

Checking the system log I found this

"2017-04-16 11:28:56,719 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable"
 
### Additional Notes

Sensor will read occasionally and then will start to show this error

kizniche, 2017-04-16:
Is that the entirety of what is in the log for this error?

Sabare3, 2017-04-16:
This is what the entirety is:

"2017-04-16 11:28:29,997 - mycodo.sensor_2 - INFO - Deactivated in 33.1 ms
2017-04-16 11:28:36,650 - mycodo.sensor_2 - INFO - Activated in 112.1 ms
2017-04-16 11:28:56,719 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-04-16 11:29:16,766 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-04-16 11:29:36,806 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-04-16 11:29:36,806 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-04-16 11:29:56,850 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-04-16 11:30:16,895 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-04-16 11:30:36,940 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-04-16 11:30:36,941 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected."

It continues to loop the error and stops the iteration.

kizniche, 2017-04-16:
I have an AM2315 sensor that's been running for months without error. Do you have an "atypical" setup, like an exceptionally large number of sensors connected, devices that could cause power spikes on the sensor power line, long wire (longer than the wire supplied with the sensor), etc.?

Sabare3, 2017-04-16:
This is the only sensor in addition to a relay. I suspected it was from my old power supply not providing enough, but after attaching it to a 5V 2.5A power supply today it is showing the same issue.

kizniche, 2017-04-16:
What you're seeing above is the sensor module attempting to read the sensor 3 times, and then failing to get a reading. Once the AM2315 stops responding, I found the only way to get it working again is to unplug it and plug it back in. When my AM2315 would become unresponsive, it was when it was powered by the same power supply as my K30 CO2 sensor that would consume large amounts of current to power its LED to measure the CO2, and would eventually disrupt the AM2315. When I removed the CO2 sensor from the same power line, all issues stopped.

kizniche, 2017-04-16:
Try experimenting with reducing the wire length, adding a capacitor to the power line, and disabling your relay form actuating and see if you still have the issue.

kizniche, 2017-04-16:
Did you have this issue prior to version 5.x?

Sabare3, 2017-04-16:
I had some issues when I was running version 4.x with the AM2315. It was able to run a DH11 without any problems. I've only had the am2315 for the past couple weeks, it ran fine for about the first 2 days before it started to show this error. I never thought to check the log until after I completely reinstalled the raspian jessie OS and Mycodo.

kizniche, 2017-04-16:
Mine would last about a day before becoming unresponsive (when connected with the CO2 sensor).

kizniche, 2017-04-16:
I would add a capacitor to your sensor power line to smooth out any voltage irregularities. I suspect that may be what is causing the issue, if wire length is not an issue.

Sabare3, 2017-04-16:
I'm pretty new to this type of stuff. What size of a capacitor should I use for this sensor?

kizniche, 2017-04-16:
A 100 nF capacitor should be sufficient.

Sabare3, 2017-04-16:
Thank you. I'll let you know if adding the cap works!

kizniche, 2017-04-16:
Cool. If you have a multimeter that can measure capacitance, you may be able to find one on an old piece of electronic circuit board that you could desolder, if you don't want to buy any.

Sabare3, 2017-04-19:
Hello again. I installed a 100nF smoothing capacitor to no avail. I think I'll order another sensor to see if it's an issue with my current one.

kizniche, 2017-04-20:
I would try to reduce the number of variables as much as you can. Does it work with just the Pi and the sensor attached with the shortest wire, directly to the Pi or a breadboard? Have you tried both 3.3 and 5 volts?

lucasvickers, 2017-05-02:
I'm having the same issue.

Initially I thought it was the wire length since I extended so I now have it running to a breadboard.  I also have a second sensor that I'm using.  I continually get the error below, however it does get the reading every other time or so.  If you look at the timestamps on the live data it's usually just a bit out of date.

```
2017-05-01 20:43:31,215 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-05-01 20:43:46,242 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-05-01 20:44:31,457 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-05-01 20:44:46,541 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-05-01 20:45:16,711 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
```

I don't have a cap, but I don't have a K30 on the line anymore.  I noticed when I type `sudo i2cdetect -y 1` it only shows up about 50% of the time.  I have tried 3v and 5v, although I have not yet tried a logic translator or debugged low level digital signals.  I did a little digging and found the following comment on [this](http://www.switchdoc.com/am2315-encased-i2c-temperature-and-humidity-sensor/) website:

> The manufacturer of the device puts the sensor into a sleep mode except when it is going to be used to avoid self heating of the sensor. That makes sense, however, it means you have to go through a funny sequence to make sure the device is awake and responding. It doesn’t respond to an Arduino I2C scan nor a Raspberry Pi I2C scan, i2cdetect.
> 
> You can run the i2cdetect -y 1 twice quickly and it will often pick the sensor up on the second run.

Do you compensate for this sleeping?  Is there any way we can add some more delay to test?

The sensor uses 0.5ma during reading, I would be surprised if it's having power issues.

kizniche, 2017-05-02:
Mycodo uses the [tentacle_pi AM2315 module](https://github.com/lexruee/am2315). I've had an AM2315 running flawlessly for months straight, and I only get about 2 errors per day from that sensor.

kizniche, 2017-05-02:
I don't have access to my development environment but I did find some other [AM2315 python module](https://github.com/camrex/am2315-python-api) that can be test.ed

kizniche, 2017-05-02:
I also suspect it could be an error in my code. Change line 105 of Mycodo/Mycodo/sensors/am2315.py from ```logger.error``` to ```logger.exception``` and restart the mycodo daemon to see the traceback in the log.

lucasvickers, 2017-05-02:
```
2017-05-01 22:25:05,783 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/am2315.py", line 102, in read
    self._dew_point, self._humidity, self._temperature = self.get_measurement()
TypeError: 'int' object is not iterable
```

kizniche, 2017-05-02:
So it looks like it [Lines 88-91](https://github.com/kizniche/Mycodo/blob/61a5c1359c257e474a4b6233b7e84dc3e0cc13f5/mycodo/sensors/am2315.py#L88-L92) of am2315.py:

```python
        if crc_check != 1:
            return 1
        else:
            dew_pt = dewpoint(temperature, humidity)
            return dew_pt, humidity, temperature
```

This shows the CRC check is failing. I'll clean up the error later to be more descriptive of the issue.

kizniche, 2017-05-02:
[Here](https://github.com/lexruee/am2315/blob/20813fef045fe632ac57ebabbce0a106278b6811/src/am2315_ext.c#L71-L75) is the relevant code in the AM2315 module that returns the measurements and the CRC.

lucasvickers, 2017-05-02:
This makes more sense and I'll go ahead and open a new issue to track a change request.

closed 2017-10-24 by kizniche