kizniche/Mycodo #242: API or something to write stats to database

cohe4ko, 2017-04-22:
##  Mycodo Version: 5.0.27
 
#### Problem Description
I'm looking for some way to write manual relay on/off actions to Mycodo database.
I have some external scripts which use the same relays as Mycodo. I need some way to add data to Mycodo database.

kizniche, 2017-04-24:
[Influxdb](https://www.influxdata.com) is used for measurement storage. [L226-L231 of config.txt](https://github.com/kizniche/Mycodo/blob/61a5c1359c257e474a4b6233b7e84dc3e0cc13f5/mycodo/config.py#L226-L231) are the credentials to write entries to the database. For relays, the measurement is [duration_sec](https://github.com/kizniche/Mycodo/blob/61a5c1359c257e474a4b6233b7e84dc3e0cc13f5/mycodo/config.py#L64-L65) and the tag [device_id](https://github.com/kizniche/Mycodo/blob/61a5c1359c257e474a4b6233b7e84dc3e0cc13f5/mycodo/utils/influx.py#L50-L52) must be the relay's unique_id (UUID when you hover the mouse over the specific relay ID).

kizniche, 2017-04-24:
See they're [example](https://github.com/influxdata/influxdb#insert-some-data) for inserting data. There are a number of other ways to enter data.

kizniche, 2017-04-24:
Their [documentation](https://docs.influxdata.com/influxdb/v1.2/guides/writing_data/) is also very good.

kizniche, 2017-04-24:
Replacing RELAY_UNIQUE_ID, try:

```
curl -i -XPOST 'http://localhost:8086/write?db=mycodo_db' --data-binary 'duration_sec,device_id=RELAY_UNIQUE_ID, value=5.5'
```

kizniche, 2017-04-24:
[API client libraries](https://docs.influxdata.com/influxdb/v1.2/tools/api_client_libraries/) for your language of choice.

cohe4ko, 2017-04-24:
curl -i -XPOST 'http://localhost:8086/write?db=mycodo_db' --data-binary 'duration_sec,device_id=5779513b-b38c-489c-89d8-af5baddc7775, value=5.5'

gives me responce:
HTTP/1.1 400 Bad Request
Content-Type: application/json
Request-Id: bb31286e-28c3-11e7-b025-000000000000
X-Influxdb-Version: 1.2.2
Date: Mon, 24 Apr 2017 07:57:58 GMT
Content-Length: 118

{"error":"unable to parse 'duration_sec,device_id=5779513b-b38c-489c-89d8-af5baddc7775, value=5.5': missing tag key"}

kizniche, 2017-04-24:
I don't use the HTTP API. You'll have to read the docs.

kizniche, 2017-04-24:
Looks like you have an extra comma. Look at my example and you'll see yours isn't modeled exactly like it.

cohe4ko, 2017-04-24:
Is there some way to write period of "Relay on" to the database for Mycodo and it should execute it?

kizniche, 2017-04-24:
I'm not exactly understanding what you're asking. My example writes a relay duration of 5.5 seconds to the database.

cohe4ko, 2017-04-24:
For now, in my own script, I manually turn on Relay and writing to the Mycodo database relay and duration_sec for stats. 
Maybe I can change approach and write to the database in some other way and give relay control to Mycodo?

kizniche, 2017-04-24:
I don't know. You're on your own outside of Mycodo.

cohe4ko, 2017-04-24:
I mean maybe there is some way to give Mycodo command "turn on relay1 for 30sec"?

kizniche, 2017-04-24:
For a list of available commands:

```~/Mycodo/env/bin/python ~/Mycodo/Mycodo/mycodo_client.py --help```

cohe4ko, 2017-04-24:
pi@greenhouse:~ $ python ~/Mycodo/mycodo/mycodo_client.py --help
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_client.py", line 29, in <module>
    import rpyc
ImportError: No module named rpyc

kizniche, 2017-04-24:
That's not the command I gave. If you used my command it would have worked.

kizniche, 2017-04-24:
Your code has a high chance of breaking if you don't use the env I specified. If a python prerequisite changes after an upgrade, your command had a chance of erroring because of an unmet dependency, while my command will not.

closed 2017-05-24 by kizniche