kizniche/Mycodo #243: Can't increase display period for graph

cohe4ko, 2017-04-22:
## Mycodo 5.0.27
 
#### Problem Description
Can't find any way to increase period of stats on live graphs. Maximum period is set for 86400 sec. Is there some way to increase this? Or there some issues with this?

kizniche, 2017-04-24:
I'm not understanding the issue. What specifically are you referring to? is it  ```x-Axis (minutes)``` or ```Refresh (seconds)```?

cohe4ko, 2017-04-24:
x-Axis (minutes)

kizniche, 2017-04-24:
There is no limitation I'm aware of. Is there an error you're receiving?

cohe4ko, 2017-04-24:
I can see only 86400 last seconds. Where can I change this range?
[image]

kizniche, 2017-04-24:
Second or minutes? Didn't you say you were talking about the x-axis?

kizniche, 2017-04-24:
Expand the graph options so I can see everything.

cohe4ko, 2017-04-24:
In stats on /graph page, I see only last 24hours. 
[image]

kizniche, 2017-04-24:
You have x-axis set to 1440 minutes

cohe4ko, 2017-04-24:
oops :) sorry for disturbing

closed 2017-04-24 by cohe4ko

kizniche, 2017-04-24:
No problem. I like issues that I don't have to change the code to fix ;)

cohe4ko, 2017-04-24:
I didn't understand that "min" means "minutes" (I thought "minimum") - maybe you can change title to "x-Axis (minutes)"

kizniche, 2017-04-24:
Yeah, I was thinking that. I'll change it later when I get home.