kizniche/Mycodo #245: enhancement : add  nrf24l01 wireless communication

reymart081678, 2017-05-01:
## Mycodo Issue Report:
  - Specific Mycodo Version: NA
 
#### Problem Description
Please list:

  - what were you trying to do
       - need to communicate to remote arduino with nrf24l01 wireless communication module
  - specific setup details that are involved
       - remote monitoring 
 
### Errors
    NA

### Steps to Reproduce the issue:
   NA 
### Additional Notes
   NA

kizniche, 2017-05-02:
Can you test the [pynrf24 module](https://github.com/jpbarraca/pynrf24)?

kizniche, 2017-11-30:
I'm hesitant to begin integrating wireless protocols for sensor measurements, because there are so many to choose from. I think I better option would be to allow the user to create their own script to acquire a measurement, then just use the "Linux Command" Input to execute the script. This is the most universal way to get this working and virtually all wireless protocols can be used this way. The user just has to figure out the hardware/software for communication, then add the Mycodo Input.

closed 2019-10-07 by kizniche