kizniche/Mycodo #246: AM2315 sensor should intelligently handle CRC failure

lucasvickers, 2017-05-02:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.27
 
#### Problem Description
The CRC checksum fails on the AM2315 and it does not distinguish CRC error from a sensor error.

### Errors
```
2017-05-01 20:44:46,541 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
2017-05-01 20:45:16,711 - mycodo.sensors.am2315 - ERROR - AM2315Sensor raised an exception when taking a reading: 'int' object is not iterable
```

### Steps to Reproduce the issue:
I am using a standard length sensor attached to a breadboard.
 
### Additional Notes
Per comments in #241, it appears the checksum is failing and this is not distinguishable from other errors.  I would suggest you tried something like:
- Attempt to read sensor
  - If major error, fail
  - If CRC error, try again 3 - 5 times
     - If CRC error continues 3 - 5 times, log error and fail

kizniche, 2017-05-02:
If you can replace the code in your module with this new code and test (restarting the daemon after saving), let me know the output. I would test, but my AM2315 is running on a production Pi, not my development Pi. If it works reliably, I'll remove the individual CRC check fail error log lines (and leave the "all measurements returned failed CRC").

lucasvickers, 2017-05-02:
This functionally works.  I could see an argument for not always logging an ERROR unless all 3 steps fail, but all in all this works.

closed 2017-05-24 by kizniche

elongo, 2020-06-04:
Hi @lucasvickers , @kizniche,
I see you guys have experience on this topic...
I have two AM2315 sensors wired to a RPi (creating a new I2C bus for the second sensor, as it is not possible to change the address of the sensor). I have worked with the same sensor in other cases, where I only have one sensor connected to the Pi, and normally it works quite well.

However this time (with two sensors wired), the readings are failing alternately. i.e.:  sometimes they read OK with both CRC 1, and in the next iteration they have CRC -1 (with Temp = 0, Humidty = 0). Sometimes both sensors return CRC 1, sometimes both CRC -1, sometimes one of them CRC 1 and the other one CRC -1, and so on.

Any clue why this is happening, and how to fix it? 

thanks...
e

kizniche, 2020-06-05:
Have you tried putting the sensor on a software I2C bus, as recommended in the manual?

elongo, 2020-06-09:
Hi @kizniche, kind you point me out the manual and this section? I think I haven't tried this... I have just plugged the sensors in I2C busses in RPi.

I appreciate the link to the documentation.

Thank you.

e

elongo, 2020-06-09:
I have this manual (link below). Which syntax/translation is crap. Do you have any better version, that could help me?
https://cdn-shop.adafruit.com/datasheets/AM2315.pdf

kizniche, 2020-06-09:
https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#am2315

elongo, 2020-06-09:
awesome, man. Thanks for the fix. Actually the bus that I created initially
was very similar to the one in the documentation you shared, but with some
differences.

My original line of code was:
dtoverlay=i2c-gpio,bus=4,i2c_gpio_delay_us=1,i2c_gpio_sda=23,i2c_gpio_scl=24

and in the new doc is:
dtoverlay=i2c-gpio,i2c_gpio_sda=23,i2c_gpio_scl=24,i2c_gpio_delay_us=4

I have implemented this, and for now the sensor works much better (perhaps
perfect). Before it was returning crc -1 after maximum three minutes. Now
after more than 40 minutes is still returning crc 1.

I hope it keeps going like this.

Many thanks.

e