kizniche/Mycodo #247: Sensor Conditional Save error

hideouspenguinboy, 2017-05-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.7
 
#### Problem Description
  - what were you trying to do
  - specific setup details that are involved
Please list: When attempting to save a relay conditional action, or save an email address to the email conditional information in the DHT22 sensor, I get 500 "Internal Server Error -The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."

I was attempting to have the sensor conditional email me and turn on a heater/AC if the temp reached a high or low.

 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
[Tue May 02 21:44:56.711744 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242] [2017-05-02 21:44:56,710] ERROR in app: Exception on /sensor [POST]
[Tue May 02 21:44:56.711850 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242] Traceback (most recent call last):
[Tue May 02 21:44:56.711870 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
[Tue May 02 21:44:56.711889 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     response = self.full_dispatch_request()
[Tue May 02 21:44:56.711907 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
[Tue May 02 21:44:56.711926 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     rv = self.handle_user_exception(e)
[Tue May 02 21:44:56.711943 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
[Tue May 02 21:44:56.711962 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     reraise(exc_type, exc_value, tb)
[Tue May 02 21:44:56.711979 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
[Tue May 02 21:44:56.711997 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     rv = self.dispatch_request()
[Tue May 02 21:44:56.712014 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
[Tue May 02 21:44:56.712033 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     return self.view_functions[rule.endpoint](**req.view_args)
[Tue May 02 21:44:56.712051 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[Tue May 02 21:44:56.712069 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     return func(*args, **kwargs)
[Tue May 02 21:44:56.712086 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 834, in page_sensor
[Tue May 02 21:44:56.712104 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     flaskutils.conditional_action_mod(form_conditional_actions, 'modify')
[Tue May 02 21:44:56.712121 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/mycodo/flaskutils.py", line 1689, in conditional_action_mod
[Tue May 02 21:44:56.712139 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     cond.id)
[Tue May 02 21:44:56.712155 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/mycodo/flaskutils.py", line 2132, in check_refresh_conditional
[Tue May 02 21:44:56.712173 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     control.refresh_sensor_conditionals(sensor_id, cond_mod, cond_id)
[Tue May 02 21:44:56.712190 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/mycodo/mycodo_client.py", line 130, in refresh_sensor_conditionals
[Tue May 02 21:44:56.712208 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     sensor_id, cond_mod, cond_id)
[Tue May 02 21:44:56.712257 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/lib/python2.7/site-packages/rpyc/core/netref.py", line 196, in __call__
[Tue May 02 21:44:56.712276 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     return syncreq(_self, consts.HANDLE_CALL, args, kwargs)
[Tue May 02 21:44:56.712293 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/lib/python2.7/site-packages/rpyc/core/netref.py", line 71, in syncreq
[Tue May 02 21:44:56.712310 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     return conn.sync_request(handler, oid, *args)
[Tue May 02 21:44:56.712325 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/lib/python2.7/site-packages/rpyc/core/protocol.py", line 441, in sync_request
[Tue May 02 21:44:56.712342 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     raise obj
[Tue May 02 21:44:56.712356 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242] TypeError: setup_sensor_conditionals() takes at most 2 arguments (3 given)
[Tue May 02 21:44:56.712372 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242] 
[Tue May 02 21:44:56.712387 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242] ========= Remote Traceback (1) =========
[Tue May 02 21:44:56.712402 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242] Traceback (most recent call last):
[Tue May 02 21:44:56.712417 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 305, in _dispatch_request
[Tue May 02 21:44:56.712434 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     res = self._HANDLERS[handler](self, *args)
[Tue May 02 21:44:56.712450 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 535, in _handle_call
[Tue May 02 21:44:56.712466 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     return self._local_objects[oid](*args, **dict(kwargs))
[Tue May 02 21:44:56.712482 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 198, in exposed_refresh_sensor_conditionals
[Tue May 02 21:44:56.712498 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     cond_id)
[Tue May 02 21:44:56.712513 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]   File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 547, in refresh_sensor_conditionals
[Tue May 02 21:44:56.712530 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242]     return self.controller['Sensor'][sensor_id].setup_sensor_conditionals(cond_mod, cond_id)
[Tue May 02 21:44:56.712546 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242] TypeError: setup_sensor_conditionals() takes at most 2 arguments (3 given)
[Tue May 02 21:44:56.712562 2017] [wsgi:error] [pid 868:tid 1937540144] [remote 192.168.1.12:16242] 

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Open up conditionals
 2. change email/relay info
 3. save changes
 4. error
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?
I do not code in this language, but I would be willing to give access to the pi via remote.

kizniche, 2017-05-03:
Your version of Mycodo is severely out-of-date. This was fixed in the current release. Mycodo has an automatic upgrade system in the top-right menu. If an upgrade doesn't work, you may have to delete your ~/Mycodo directory and install again.

closed 2017-05-03 by kizniche

hideouspenguinboy, 2017-05-03:
I'm the dumbest. Updating. (but isn't it such an easy ticket to close!)