kizniche/Mycodo #248: Timer

hideouspenguinboy, 2017-05-03:
## Mycodo Issue Report:
  - Specific Mycodo Version:
5.0.27
 
#### Problem Description
Please list:
  - what were you trying to do
  - specific setup details that are involved
 Trying to use a timer to turn lights on for a period of time. 
ID: 1,Active
Relay ID- (2)light 1
State- on
Start Time- 18:00
End Time- 06:00

### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
Relay stays on past it's timer. No errors in log.

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Set up timer per above
 2. Start time always reverts to 930 after saving. Change to 18:00 again and save.
 3. Activate.
 4. Wait and cry.
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?
If the clock can't go past 23:59:59, it should have an error message much like the PID:Method does. 
If this require you to have a separate off timer, it should be listed in the manual, though it would be weird to list it as a timer if it didn't switch the state to it's opposite state at the end, unless I'm misunderstanding it's use-case.

kizniche, 2017-05-03:
The period timer (from x to y time) only does as you instruct. It will maintain the relay state for that duration. Handy for when the power goes out, when the power returns and Mycodo starts back up, it will look at the duration and determine if the relay should be on or off.

This is not true of the single time timer (at x time), as this will only trigger at that specific time, and if a power outage occurs, will not trigger when Mycodo starts up because that time had passed.

In order to be the most flexible, the timers ***only*** do as you instruct them. Therefore if you do not instruct it to turn off after a duration of being on, it will not do so.

kizniche, 2017-05-03:
It also properly handles going past midnight. If there is an issue with your duration, an error will occur.

closed 2017-05-03 by kizniche

kizniche, 2017-05-03:
I have confirmed the start time doesn't initially save, but as you pointed out it can be modified after creation. I'll fix this in the next release.

hideouspenguinboy, 2017-05-03:
In that case I would just add a little blurb to make it more clear, as timers tend to turn "state" for the duration, and turn "opposite state." Realizing that it will continually check to make sure it is in the "state" for the specified time makes perfect sense and I'll adjust my setting accordingly.  Thank you VERY much for your work with the project and your continued support.

kizniche, 2017-05-03:
Those last two commits should fix the saving issue and add a note about the functionality.

Thanks for the bug report.