kizniche/Mycodo #249: Sensor Conditional Relay On Forever

hideouspenguinboy, 2017-05-05:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.27
 
#### Problem Description
Please list:
  - what were you trying to do
  - specific setup details that are involved
 Sensor conditional set to turn on for 240 seconds if temp reached above 26.6 degrees. The relay stayed on until turned off manually. Just trying to keep the temp under 26.6 degrees until my understanding of PIDs is better.

### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
Asides from the relay not switching off, these are the only errors in the logs I could find.
[Fri May 05 06:25:05.792739 2017] [ssl:warn] [pid 703:tid 1995489280] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Fri May 05 06:25:05.793558 2017] [wsgi:warn] [pid 703:tid 1995489280] mod_wsgi: Compiled for Python/2.7.8.
[Fri May 05 06:25:05.793597 2017] [wsgi:warn] [pid 703:tid 1995489280] mod_wsgi: Runtime using Python/2.7.9.
[Fri May 05 06:25:05.795024 2017] [mpm_event:notice] [pid 703:tid 1995489280] AH00489: Apache/2.4.10 (Raspbian) OpenSSL/1.0.1t mod_wsgi/4.3.0 Python/2.7.9 configured -- resuming normal operations
[Fri May 05 06:25:05.795128 2017] [core:notice] [pid 703:tid 1995489280] AH00094: Command line: '/usr/sbin/apache2'

### Steps to Reproduce the issue:
How can this issue be reproduced?
I'm unsure how it happened as these settings have worked fine before. I will leave with the current settings and just pay WAY more attention since this is in production.
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-05-05:
Can you provide more detailed steps to reproduce the issue, including specific sensor and conditional settings?

kizniche, 2017-05-05:
Also, the relays are controlled by the daemon, so the daemon log is where you will want to be looking for errors, not the HTTP log.

hideouspenguinboy, 2017-05-05:
DHT22, GPIO 23, Power and Pre Relay disabled

IF Temperature (C) IS Greater Than 26.6 PERIOD 120
Relay 10(AC) TURN On SEC 240

The daemon log just didn't have anything for the time period, which was between 5/4 at night and 5/5 in the morning:
2017-05-03 16:29:17,323 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-05-03 16:29:17,362 - mycodo.pid_1 - WARNING - No data returned from influxdb
2017-05-03 17:20:17,554 - mycodo.pid_1 - INFO - Deactivated in 166.5 ms
2017-05-03 17:20:21,290 - mycodo.pid_1 - INFO - Activated in 128.8 ms
2017-05-04 11:00:43,692 - mycodo.timer_5 - INFO - Deactivated in 115.8 ms
2017-05-04 12:04:47,404 - mycodo.pid_1 - INFO - Deactivated in 65.8 ms
2017-05-04 15:17:17,257 - mycodo - INFO - 64.756 MB ram in use
2017-05-05 10:55:37,637 - mycodo.timer_5 - INFO - Activated in 54.2 ms
2017-05-05 11:05:02,280 - mycodo.timer_6 - INFO - Deactivated in 73.9 ms
2017-05-05 11:22:39,172 - mycodo.timer_6 - INFO - Activated in 108.2 ms

hideouspenguinboy, 2017-05-06:
This did not repeat last night.

kizniche, 2017-05-07:
What I think is happening is because your relay on duration is much greater than your period, the relay has no chance to turn off.

Scenario:
Time 0 seconds: Measure 28, turn relay on for 240 seconds
Time 120 seconds: Measure 27.7, log relay duration of 120 seconds, update relay on for 240 seconds
Time 240 seconds: Measure 27.6, log relay duration of 120 seconds, update relay on for 240 seconds
Time 360 seconds: Measure 27.5, log relay duration of 120 seconds, update relay on for 240 seconds
...and so on.

Naturally, if the on duration is longer than the period, and the conditional is continually being True, then you won't see the relay turn off, in the above scenario.

hideouspenguinboy, 2017-05-07:
Because the duration value does not reset and is additive?

kizniche, 2017-05-07:
The relay will never be on for more than the duration you set. See that at 120 seconds after turning on, the conditional is triggered again, causing the 120 seconds to be recorded and the relay to be updated ot be on for 240 seconds. There is no addition. If the temperature never falls below the conditional setpoint, then the relay will effectively be on forever, or until the temperature falls below the setpoint for longer than 240 seconds.

hideouspenguinboy, 2017-05-07:
Right, as was my intention.  The problem is that after the temperature fell under that condition, the AC stayed on for hours and the heater had to fight the AC to stay above my temperature minimum.

kizniche, 2017-05-07:
Strange. I haven't observed this behavior. If you turn on debugging in the daemon, there will be a lot of log lines that should indicate what's going on. You can do this by executing the following:

```bash
sudo service mycodo stop &&
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py -d
```

Then monitor the daemon log when the issue arises.

hideouspenguinboy, 2017-05-07:
Awesome, I will do this! I've worked in QA and I felt terrible putting up a bug that I can't reproduce or give any real data on.  Hopefully this will find something... or not and I'll find out it's a hardware problem with the relay.

kizniche, 2017-05-07:
Keep in mind there may be a ***lot*** of log lines, so I would only do this if you can dedicate a session to testing while you enable debug mode. Otherwise, just watch your log size (i.e. free space from the System information page).

kizniche, 2017-05-07:
With relays, I'll just comment that with an inductive load (current spikes from motor/compressor turning on) such as an AC unit, you will want to have a relay with a current rating 10 times (roughly) over the current draw of the AC unit. I've seen both mechanical and solid state relays "stick" in the on position due to exceeding the max rating. However, this would be obvious because you would not be able to turn the relay off from Mycodo because it would be stuck on, regardless of the signal coming from the Pi GPIO.

hideouspenguinboy, 2017-05-07:
Aaaaand the system crashed when I entered that. I'll reboot and see where it goes from there.
(And yeah, my lights were all on one and I thought the 10amp relays could deal with the 7 amps of current. I was wrong)

kizniche, 2017-05-08:
Resistive loads such as lights don't need the 10x amp requirement as inductive loads do, because they don't usually have large current spikes when initially powered. Though, that will depend on what type of lights you use, but as far as I know, they can be treated as a resistive load.

kizniche, 2017-06-30:
Just checking in on this issue, if you have any new info or if we can close it. Thanks.

hideouspenguinboy, 2017-06-30:
It never repeated after I started looking, but I did get errors stating the
amperage on the switch was rated too high to turn on. I'm guessing it dealt
more with my hardware than your software. Sorry I didn't get back to you
sooner!

kizniche, 2017-06-30:
You just made me recall that I never made the Max Allowable Amps option to be configured! If you look in [config.py](https://github.com/kizniche/Mycodo/blob/b7c5945ad18be782b4b4bf8fdae430a183cbebda/mycodo/config.py#L276), there's a MAX_AMPS option that [controller_relay.py](https://github.com/kizniche/Mycodo/blob/b7c5945ad18be782b4b4bf8fdae430a183cbebda/mycodo/controller_relay.py#L192-L202) checks every time a relay is requested to be turned on. This is all predicated on the notion you have set the current draw for each relay (if it's set to 0, this obviously won't work correctly). If the relay turning on will make the current draw greater than the max current allowed, the relay won't turn on and the event logged. This is a failsafe to ensure too much current isn't drawn that may pose a fire hazard because the wires may burn up.

I'll turn this into a database entry and make it editable in one of the next updates.

hideouspenguinboy, 2017-06-30:
That sounds great to have. Running a watt meter and just checking each
relay and print into a database should be easy and a good idea. Maybe a
warning about inductive loads would prevent people like me from fucking it
up with the wrong equipment. (I'm running high powered LED drivers for my
lights. I'm assuming those and ballasts are still resistive.

closed 2017-10-18 by kizniche