kizniche/Mycodo #250: 433mhz transmitter relay issues

mycol, 2017-05-29:
## Mycodo Issue Report:
Mycodo Version: 5.0.30
 
#### Problem Description
  - what were you trying to do:  

I was trying to add 433mhz relay but after adding I'm having trouble sending the on/off codes. Tested using codesend app to rule out transmitter and outlet. The settings for the wireless relay in mycodo are ...

WiringPi Pin #11 (this is a Pi 3 and the corresponding GPIO number should be 17)
Protocol 1
Pulse Length 189
Bit Length 24
On Command: 5575987
Off Command: 5575996
Current Draw (amps) : 0.0
Start State Off

### Errors
Does not turn outlet on/off.

### Additional Notes
Unlike the sensors the Relay setup asks for a WiringPi Pin instead of a GPIO, could the pin to GPIO # mapping be causing issues? There might be changes between Pi generations which might also affect the pin to GPIO number, I could be wrong? 

Any suggestions would be welcomed.

kizniche, 2017-05-31:
Have you executed the script mentioned in the [Manual](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.md#relays) to determine the code being transmitted from your remote (as well as the other parameters)?

kizniche, 2017-05-31:
pi_switch, the Python module being used, uses WiringPi pin numbering, not BCM numbering as do other modules used in Mycodo. You will need to reference your specific Pi version for the WiringPi pin, if they differ between boards.

kizniche, 2017-06-02:
What transmitter are you using and what receiver-relay product are you using?

mycol, 2017-06-02:
Thanks for the responses, Kyle, sorry for my delay in responding. I'm using the receiver/transmitter combo linked below. I know the transmitter/receiver work because I did log the codes from the outlets using the receiver which mapped each outlet. I am able to use the 'codesend' to send the On/Off codes with success so I think the pi and reciver/transmitter setup are OK.

So, I'm fairly certain my pin 11 sends to GPIO 17 on my pi3, but I don't believe the previous gen Pis had a GPIO17 so that could be my problem. I'll try finding another unused pin that's available on both Pi models and I'll let you know how that worked out.

433mhz radios
https://www.amazon.com/gp/product/B00M2CUALS/ref=oh_aui_detailpage_o03_s01?ie=UTF8&psc=1

kizniche, 2017-06-02:
If you go to the ```System Information``` page in Mycodo, you can see the specific pin layout for your Pi.

Also, I have the same outlets/remote and my codes are shorter numbers (though that doesn't necessarily mean yours should be).

mycol, 2017-06-02:
Oh, nice! Here's the output.
` +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |  OUT | 0 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 0 | IN   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | IN   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |  OUT | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 1 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 1 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |  OUT | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 0 | OUT  | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |  OUT | 0 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |  OUT | 0 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+`

kizniche, 2017-06-02:
Great! By "Works" do you mean everything works now?

mycol, 2017-06-02:
Yes, all is working properly now, thanks for the help and thanks for making this excellent application!

kizniche, 2017-06-02:
You're welcome. Let me know if there are any specific features you'd like to see added. I always value user feedback for new ideas.

closed 2017-06-02 by kizniche