kizniche/Mycodo #251: SHT75/71 cant activate

wangine, 2017-05-30:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 Mycodo Version: 5.0.30 
#### Problem Description
Please list:
The fields with CLK pin and voltage cant be filled. After save the fields 
remains blank and clock pin as 0. 
 
### Errors

If i try to activate i get a error because the fields remain empty. 0 for clk pin and no voltage reference. 
First message is weird .
xxxSuccess: Sensor controller activated in SQL database
Second message with the error (that's normal )
xxxError: Could not activate Sensor controller with ID 2: 0.0

### Steps to Reproduce the issue:
I didn't use my controller in last 2 months and i just say to update first to 
last version . And after that that error appear. I put a new image on my new SD card
because on last one (class 10) system keep crashing after 3 months of
continue job. I buy a new micro SD SONY UHS-I SDHC 96MB/s because the last one was a low quality SD card. After flashing a new image of fresh RPi image , i found that issue.
 
### Additional Notes
I send you on email my pass and user , also the adress for my RPi. Maybe help
Thank you

kizniche, 2017-05-31:
I'll investigate this later today. Should be an easy fix. Also, I never received an email.

wangine, 2017-05-31:
Yes , i've seen that . I cant send you a email , the error is strange.
Diagnostic-Code: smtp; 550 High probability of spam
Last-Attempt-Date: Tue, 30 May 2017 12:04:33 -0700 (PDT)
In email is , my RPi adress redirected on port 80,  user and pass. 
Also i dont know if you push a update , because i can not update. I put again a fresh system and new
Mycodo , but the same thing.

kizniche, 2017-05-31:
I pushed two commits with fixes, but I haven't made a new release. If you want to install Mycodo with the latest code, you'll have to clone the repo with git:

```git clone https://github.com/kizniche/Mycodo```

Then follow the instructions by executing the setup.sh script. I haven't had a chance to test the fixes so I haven't created a release. I'll be able to rest this afternoon, but I'm fairly sure it will work.

wangine, 2017-06-01:
Yes, the sensor is fixed now and working . Still now the relays can not be activate .
The error is listed below.
Error: Error: Actuate Relay: Invalid value: Channel must be an integer or list/tuple of integers

========= Remote Traceback (1) =========
Traceback (most recent call last):
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 305, in _dispatch_request
    res = self._HANDLERS[handler](self, *args)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 535, in _handle_call
    return self._local_objects[oid](*args, **dict(kwargs))
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 205, in exposed_relay_on
    return mycodo.relay_on(relay_id, duration, min_off_duration)
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 580, in relay_on
    min_off_duration=min_off_duration)
  File "/var/www/mycodo/mycodo/controller_relay.py", line 196, in relay_on_off
    current_amps = self.current_amp_load()
  File "/var/www/mycodo/mycodo/controller_relay.py", line 644, in current_amp_load
    if self.is_on(each_relay_id):
  File "/var/www/mycodo/mycodo/controller_relay.py", line 707, in is_on
    self._is_setup(relay_id, self.relay_pin[relay_id])):
  File "/var/www/mycodo/mycodo/controller_relay.py", line 725, in _is_setup
    GPIO.setup(pin, GPIO.OUT)
ValueError: Channel must be an integer or list/tuple of integers
//~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
I will wait a update when you have free time to push. Also i will put again a clean RPi system
because everything is new and i want to be stable for next 3 months.
Thanks
[image]

kizniche, 2017-06-01:
Is that the only relay that's been created?

wangine, 2017-06-01:
No , i have 8 relays , i tested two of them . The same error. Also i delete all relays and i created only one ,the error become when i try to test or i try to turn ON. If i turn off daemon will return succes.
Strange. Maybe i messed up something when i update . I will try tomorrowto put a clean system.

kizniche, 2017-06-01:
That's an unexpected error.

Could you add a log line to figure out what the ```pin``` variable it's complaining about it?

To do that, edit and paste this line above line 725 of ```~/Mycodo/mycodo/mycodo/controller_relay.py```:

```
            self.logger.error("ER01: {}".format(pin))
```

So it looks like:

```
        if self.relay_type[relay_id] == 'wired':
            GPIO.setmode(GPIO.BCM)
            self.logger.error("ER01: {}".format(pin))
            GPIO.setup(pin, GPIO.OUT)
        return True
```

Then restart mycodo: ```sudo service mycodo restart```

Then attempt to turn the relay on, then read the daemon log and look for that log line.

kizniche, 2017-06-01:
I suspect you have a relay that hasn't been configured with a pin. That is a scenario that might cause this situation.

kizniche, 2017-06-01:
That's it. I just replicated the issue. Working on a fix.

wangine, 2017-06-01:
Right now i'm not at home to testing that . I will be able after some hours. Thanks for your time .

kizniche, 2017-06-01:
Just released [v5.0.31](https://github.com/kizniche/Mycodo/releases/tag/v5.0.31) that should fix these issues. Let me know if it doesn't. Thanks for the help.

closed 2017-06-01 by kizniche