kizniche/Mycodo #252: Graphs not showing PID Values - Output/Setpoint

Ryobidelacroix, 2017-06-01:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.31
 
#### Problem Description
Please list: 

Navigate to: 
Data --> Live Graphs.

Graphs will not show PID values for Setpoint / Output. 

 
### Errors
 
Nothing appears erroneous in logs. 

### Steps to Reproduce the issue:
As above. 

Attempted to delete graphs, then added them again. No change. 
Deleted PID's. Re added PID's. No change.
Added separate graph, including ONLY PID data. No change. 

Note: ALL PID's are fully functional, they're just not displaying on a graph.
This feature is super handy for PID tuning. 

Cheers! :)

kizniche, 2017-06-02:
I'll look into this today when I get a chance to get on my development Pi.

kizniche, 2017-06-02:
I was able to replicate the issue and find a solution. I just released v5.0.32 that should fix the issue. Let me know if it doesn't work for you. Thanks for the bug report.

closed 2017-06-02 by kizniche