kizniche/Mycodo #254: "method raised IOError: [Errno 24] Too many open files"

Ryobidelacroix, 2017-06-07:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.33
 
#### Problem Description
All (temperature) sensors showing '0.00' 

### Errors
**(I have attached a full copy of the Daemon Log as a txt file)** 

2017-06-07 11:03:18,941 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-06-07 11:03:20,138 - mycodo.pid_1 - ERROR - Run Error: (sqlite3.OperationalError) unable to open database file

========= Remote Traceback (1) =========
Traceback (most recent call last):
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 329, in _dispatch_request
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 583, in _handle_call
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 201, in exposed_relay_off
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 553, in relay_off
  File "/var/www/mycodo/mycodo/controller_relay.py", line 348, in relay_on_off
    self.check_conditionals(relay_id, duration)
  File "/var/www/mycodo/mycodo/controller_relay.py", line 397, in check_conditionals
    for each_conditional in conditionals.all():

(continues) 

[DaemonLog.txt](https://github.com/kizniche/Mycodo/files/1057167/DaemonLog.txt)

### Steps to Reproduce the issue:
Unable to replicate so far.
Issue appears to be **resolved by restarting the Pi**, but submitting logs just as an FYI.

kizniche, 2017-06-07:
I've never seen that error before. I'll investigate it.

kizniche, 2017-06-08:
Are you running anything else besides Mycodo? I've never experienced this on any of my systems.

Ryobidelacroix, 2017-06-08:
No i'm not - this Pi is dedicated to Mycodo. 
It hasn't happened since either. It's a strange one! 

Feel free to close this issue, it appears to be a once off occurrence and I was just submitting to get your thoughts! :)

Ryobidelacroix, 2017-06-08:
I'll check the logs when I get home and see if there is anything else remotely odd.

Ryobidelacroix, 2017-06-08:
Same issue again. This time the Daemon crashed. No temperature sensor readings saved in the last ~7 hours. Interestingly though, by logging back in, temperature readings began working again, but without the daemon running.

kizniche, 2017-06-08:
Is this the only sensor you have enabled?

kizniche, 2017-06-08:
If you can, can you disable all parts of Mycodo (sensors, conditionals, timers, etc.) except the Pi temperature sensor? I'd like to isolate what's causing the issue, then we can add logging lines with open file counts to see what's going on.

kizniche, 2017-06-08:
You may have a very specific configuration that might be causing this issue, or there may be something specific about your system (corrupt file perhaps) that's causing it. I've ran every feature of mycodo, in all sorts of configurations, for extended periods of time and never came across this issue. I'd like to try all other options before just increasing the number of open files at one time, as I think that would only be a temporary bandaid to the issue and merely delay the inevitable (if there is something continuouslt opening files and not closing them).

kizniche, 2017-06-08:
Can you paste the output of the following command that will indicate how many open files your system permits at a time?

```ulimit -n```

kizniche, 2017-06-08:
You can edit ~/Mycodo/mycodo/sensors/raspi.py and edit after Line 45 so the ```get_measurement()``` function looks like this:

```python
    def get_measurement():
        """ Gets the Raspberry pi's temperature in Celsius by reading the temp file and div by 1000 """
        import psutil
        import resource
        open_files_count = 0
        for proc in psutil.process_iter():
            if proc.open_files():
                open_files_count += 1
        logger.info("Open files: {of}".format(of=open_files_count))
        soft, hard = resource.getrlimit(resource.RLIMIT_NOFILE)
        logger.info("LIMIT: Soft: {sft}, Hard: {hrd}".format(sft=soft, hrd=hard))
        with open('/sys/class/thermal/thermal_zone0/temp') as cpu_temp_file:
            return float(cpu_temp_file.read()) / 1000
```

Then save the file, and restart the daemon:

```sudo service mycodo restart```

The new log line will print the number of open files and the limits every temperature read. This may not exactly be the right information we're lookig for, but it could be a start to understanding what's going on.

Ryobidelacroix, 2017-06-09:
Sorry for the delay, I have been away from my Pi. 

The `ulimit -n` shows me 65536, same as your system.

I'll edit my raspi.py and send you the results...

Ryobidelacroix, 2017-06-09:
Current log after editing raspi.py

2017-06-09 18:22:07,442 - mycodo - INFO - Mycodo daemon v5.0.33 starting
2017-06-09 18:22:07,616 - mycodo - INFO - Anonymous statistics enabled
2017-06-09 18:22:07,825 - mycodo.relay - INFO - Relay 1 setup on pin 21 and turned OFF (OFF=HIGH)
2017-06-09 18:22:07,834 - mycodo.relay - INFO - Relay controller activated in 73.9 ms
2017-06-09 18:22:07,834 - mycodo - INFO - All activated timer controllers started
2017-06-09 18:22:08,007 - mycodo.sensor_1 - INFO - Activated in 133.0 ms
2017-06-09 18:22:08,009 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: No module named psutil
2017-06-09 18:22:08,009 - mycodo - INFO - All activated sensor controllers started
2017-06-09 18:22:08,010 - mycodo - INFO - All activated PID controllers started
2017-06-09 18:22:08,010 - mycodo - INFO - All activated LCD controllers started
2017-06-09 18:22:08,011 - mycodo - INFO - Mycodo daemon v5.0.33 started in 0.569 seconds
2017-06-09 18:22:08,018 - mycodo - INFO - 41.884 MB ram in use
2017-06-09 18:22:13,026 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: No module named psutil
2017-06-09 18:22:18,043 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: No module named psutil
2017-06-09 18:22:18,044 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-06-09 18:22:23,065 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: No module named psutil
2017-06-09 18:22:28,084 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: No module named psutil
2017-06-09 18:22:33,103 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: No module named psutil
2017-06-09 18:22:33,104 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-06-09 18:22:38,121 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: No module named psutil
2017-06-09 18:22:43,164 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: No module named psutil
2017-06-09 18:22:48,181 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: No module named psutil
2017-06-09 18:22:48,182 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.

Ryobidelacroix, 2017-06-09:
(All other sensors, PID's, conditionals were deactivated)

kizniche, 2017-06-09:
Looks like you don't have psutil installed. Use this command to install and restart the daemon.

```~/Mycodo/env/bin/pip install psutil```

Ryobidelacroix, 2017-06-09:
Done. 

2017-06-09 19:55:30,293 - mycodo - INFO - Mycodo daemon v5.0.33 starting
2017-06-09 19:55:30,519 - mycodo - INFO - Anonymous statistics enabled
2017-06-09 19:55:30,730 - mycodo.relay - INFO - Relay 1 setup on pin 21 and turned OFF (OFF=HIGH)
2017-06-09 19:55:30,731 - mycodo.relay - INFO - Relay controller activated in 62.4 ms
2017-06-09 19:55:30,731 - mycodo - INFO - All activated timer controllers started
2017-06-09 19:55:30,909 - mycodo.sensor_1 - INFO - Activated in 138.3 ms
2017-06-09 19:55:31,276 - mycodo.sensor_2 - INFO - Activated in 154.8 ms
2017-06-09 19:55:31,280 - mycodo - INFO - All activated sensor controllers started
2017-06-09 19:55:31,471 - mycodo.pid_1 - INFO - Activated in 150.0 ms
2017-06-09 19:55:31,477 - mycodo - INFO - All activated PID controllers started
2017-06-09 19:55:31,478 - mycodo - INFO - All activated LCD controllers started
2017-06-09 19:55:31,479 - mycodo - INFO - Mycodo daemon v5.0.33 started in 1.186 seconds
2017-06-09 19:55:31,490 - mycodo - INFO - 49.404 MB ram in use
2017-06-09 19:55:32,911 - mycodo.sensors.raspi - INFO - Open files: 57
2017-06-09 19:55:32,912 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-09 19:55:37,965 - mycodo.sensors.raspi - INFO - Open files: 57
2017-06-09 19:55:37,965 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-09 19:55:42,574 - mycodo.sensors.raspi - INFO - Open files: 57
2017-06-09 19:55:44,203 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-09 19:55:47,767 - mycodo.sensors.raspi - INFO - Open files: 57
2017-06-09 19:55:47,768 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-09 19:55:53,866 - mycodo.sensors.raspi - INFO - Open files: 57
2017-06-09 19:55:53,867 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-09 19:55:58,837 - mycodo.sensors.raspi - INFO - Open files: 57
2017-06-09 19:55:59,184 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-09 19:56:02,925 - mycodo.sensors.raspi - INFO - Open files: 57
2017-06-09 19:56:02,926 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-09 19:56:08,833 - mycodo.sensors.raspi - INFO - Open files: 57
2017-06-09 19:56:08,835 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096

kizniche, 2017-06-09:
Now we wait until the error occurs again.

Ryobidelacroix, 2017-06-09:
Brilliant. Thanks for your help!

kizniche, 2017-06-09:
You have a very short sensor period, so I would keep an eye on your log size, unless you're using a large SD card.

Ryobidelacroix, 2017-06-10:
A few interesting lines:  

2017-06-10 22:27:18,113 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/proc/2/stat'
2017-06-10 22:27:26,254 - mycodo.sensors.raspi - INFO - Open files: 29
2017-06-10 22:27:26,255 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096

2017-06-10 22:29:32,858 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-10 22:29:37,627 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp raised an exception when taking a reading: psutil.NoSuchProcess process no longer exists (pid=1121)
2017-06-10 22:29:42,489 - mycodo.sensors.raspi - INFO - Open files: 29
2017-06-10 22:29:42,490 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096

2017-06-10 22:32:45,297 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-10 22:32:49,723 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/proc/1121/fdinfo/863'
2017-06-10 22:32:53,621 - mycodo.sensors.raspi - INFO - Open files: 29
2017-06-10 22:32:53,621 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096

2017-06-10 22:37:31,706 - mycodo.sensors.raspi - INFO - LIMIT: Soft: 1024, Hard: 4096
2017-06-10 22:37:36,201 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too m
any open files: '/proc/1121/fdinfo/693'
2017-06-10 22:37:41,998 - mycodo.sensors.raspi - INFO - Open files: 29

I've attached a larger chunk of the daemon log: [Log2.txt](https://github.com/kizniche/Mycodo/files/1065858/Log2.txt)

kizniche, 2017-06-10:
Strange. It's not what I was expecting, and it doesn't seem like the code I mentioned is helping identify the issue. Can you run the following command at a fresh bootup and then once the Pi starts giving the "too many open files" error. I would remove (or comment out) the code you inserted so it's back to the original code.

```bash
sudo lsof | awk '{ print $2 " " $1; }' | sort -rn | uniq -c | sort -rn | head -20
```

This will output with three lines, ```open files | PID | process name```

Here is the output on my Pi that's been running for a few days:

```

 137671 720 influxd
   8584 29806 python
   2124 3745 apache2
     90 1154 fail2ban-
     66 4606 apache2
     66 4591 apache2
     66 3690 apache2
     66 28846 apache2
     66 28252 apache2
     66 25515 apache2
     66 17494 apache2
     66 17414 apache2
     66 1308 apache2
     66 11371 apache2
     63 1053 apache2
     60 469 pigpiod
     56 479 rngd
     54 16676 sshd
     50 16670 sshd
     46 1 systemd
```

kizniche, 2017-06-30:
This is indeed a strange issue that I've never seen or experienced before. Do you have any more info? Is the issue still happening? Have you tried different hardware (RPi, SD card), fresh Raspbian image?

supajay, 2017-07-01:
i have been getting the same error when running 5.0.39
first it showed the following in the Daemon log:

2017-07-01 14:19:49,786 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:19:49,790 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:20:04,803 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:20:19,809 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:20:34,819 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:20:34,843 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:20:49,876 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:21:04,958 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:21:07,036 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:21:20,050 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:21:20,059 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:21:35,053 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:21:50,145 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:05,168 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:05,171 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:22:20,174 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:35,186 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:37,094 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:22:50,216 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:50,248 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:23:05,298 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:23:20,345 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:23:35,348 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:23:35,364 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:23:50,376 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:24:05,475 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:24:07,207 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:24:20,563 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:24:20,591 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:24:35,593 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'

**now it shows this:**

[... cut ...]
    data = self._recv(timeout, wait_for_lock = True)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 369, in _recv
    data = self._channel.recv()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/channel.py", line 50, in recv
    header = self.stream.read(self.FRAME_HEADER.size)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/stream.py", line 213, in read
    raise EOFError("connection closed by peer")
EOFError: connection closed by peer

supajay, 2017-07-01:
also if i try and restart the daemon i get the following:

sudo service mycodo restart
Job for mycodo.service failed. See 'systemctl status mycodo.service' and 'journalctl -xn' for details.

supajay, 2017-07-01:
while the daemon appears to be running normally the following is found in the daemon log

2017-07-01 19:09:06,223 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:09:06,314 - mycodo.pid_4 - WARNING - No data returned from influxdb
2017-07-01 19:09:06,414 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:09:06,474 - mycodo.pid_3 - WARNING - No data returned from influxdb
2017-07-01 19:09:36,196 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:09:36,301 - mycodo.pid_4 - WARNING - No data returned from influxdb
2017-07-01 19:09:40,724 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:09:40,827 - mycodo.pid_2 - WARNING - No data returned from influxdb
2017-07-01 19:09:50,177 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 19:10:05,863 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:10:05,933 - mycodo.pid_4 - WARNING - No data returned from influxdb
2017-07-01 19:10:06,519 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:10:06,618 - mycodo.pid_3 - WARNING - No data returned from influxdb
2017-07-01 19:10:30,797 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:10:30,816 - mycodo.pid_4 - WARNING - No data returned from influxdb
2017-07-01 19:10:41,363 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:10:41,409 - mycodo.pid_2 - WARNING - No data returned from influxdb
2017-07-01 19:11:04,074 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:11:04,119 - mycodo.pid_4 - WARNING - No data returned from influxdb
2017-07-01 19:11:05,024 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:11:05,086 - mycodo.pid_3 - WARNING - No data returned from influxdb
2017-07-01 19:11:20,330 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 19:11:32,787 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:11:32,839 - mycodo.pid_4 - WARNING - No data returned from influxdb
2017-07-01 19:11:36,598 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:11:36,614 - mycodo.pid_2 - WARNING - No data returned from influxdb
2017-07-01 19:12:06,871 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:12:06,934 - mycodo.pid_4 - WARNING - No data returned from influxdb
2017-07-01 19:12:07,133 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-01 19:12:07,176 - mycodo.pid_3 - WARNING - No data returned from influxdb

kizniche, 2017-07-01:
The daemon doesn't appear to be running. What is the error when performing these commands? 

```
sudo service mycodo stop
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py -d
```

supajay, 2017-07-02:
I get the following when running those commands

sudo service mycodo stop
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py -d
Lock file present. Ensure the daemon isn't already running and delete /var/lock/mycodo.pid

kizniche, 2017-07-02:
Delete that file and try those commands again.

```
sudo rm /var/lock/mycodo.pid
```

supajay, 2017-07-02:
I have deleted the lock file and ran the commands again and it returns nothing, just back to the command prompt.

see below:

pi@PiDry:~ $ sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py -d
Lock file present. Ensure the daemon isn't already running and delete /var/lock/mycodo.pid
pi@PiDry:~ $ sudo rm /var/lock/mycodo.pid
pi@PiDry:~ $ sudo service mycodo stop
pi@PiDry:~ $ sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py -d
pi@PiDry:~ $

Ryobidelacroix, 2017-07-11:
It happened again today. The above command shows me:

 7277 1120 python
   2160 1241 apache2
   1701 1243 apache2
   1701 1242 apache2
   1372 549 influxd
    372 708 llvmpipe-
    175 989 menu-cach
    175 989 lxpanel
    175 989 gmain
    175 989 gdbus
    132 972 gvfsd-fus
    122 990 pcmanfm
    122 990 gmain
    122 990 gdbus
    102 1061 gvfs-afc-
     93 708 Xorg
     85 984 openbox
     84 927 lxsession
     84 927 gmain
     84 927 gdbus

I'll try a fresh install on a spare SD card and see how it goes...

Ryobidelacroix, 2017-07-11:
After reboot, Daemon now running... 

2756 1120 python
   2205 743 apache2
   1755 745 apache2
   1701 744 apache2
   1188 560 influxd
    372 708 llvmpipe-
    175 984 menu-cach
    175 984 lxpanel
    175 984 gmain
    175 984 gdbus
    132 968 gvfsd-fus
    122 985 pcmanfm
    122 985 gmain
    122 985 gdbus
    102 1053 gvfs-afc-
     93 708 Xorg
     85 979 openbox
     84 923 lxsession
     84 923 gmain
     84 923 gdbus

Theoi-Meteoroi, 2017-10-21:
Getting this issue [ too many open files ] with a TSL2591 as the only sensor connected.  I have two Raspi3 systems, this has followed the sensor. In this case, only the CPU temp and the TSL2591 are configured and activated. 

2017-10-21 01:54:59,176 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys
/class/thermal/thermal_zone0/temp'
2017-10-21 01:54:59,464 - mycodo.sensors.tsl2591_3_41 - ERROR - TSL2591Sensor raised an exception when taking a reading: 24
2017-10-21 01:55:14,201 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys
/class/thermal/thermal_zone0/temp'
pi@Control2:~ $ sudo lsof | awk '{ print $2 " " $1; }' | sort -rn | uniq -c | sort -rn | head -20
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
   7314 1187 python   
   2124 5878 apache2
   1785 1214 SimpleCac
   1728 5914 apache2
   1701 5879 apache2
   1626 1398 WorkerPoo
   1084 1398 Composito
   1071 1214 BrowserBl
    980 1306 llvmpipe-
    976 1274 Composito
    714 1214 chromium-
    714 1214 Chrome_Fi
    488 1274 Chrome_li
    440 727 llvmpipe-
    357 1214 WorkerPoo
    357 1214 sandbox_i
    357 1214 renderer_
    357 1214 NetworkCh
    357 1214 inotify_r
    357 1214 IndexedDB
pi@Control2:~ $ ulimit -a
core file size          (blocks, -c) 0
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 7314
max locked memory       (kbytes, -l) unlimited
max memory size         (kbytes, -m) unlimited
open files                      (-n) 65536
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 95
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 7314
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited

looking at all the limits, I note 'pending signals' and 'max user procs' are the same as the number of python procs. Coincidence?

Theoi-Meteoroi, 2017-10-21:
So I looked at the lsof output and found many python procs are holding /dev/i2c-3 open  ( I use the SDL 9545A mux )

A sample of three lines:

python    1187 1659             root 1021u      CHR       89,3       0t0       9430 /dev/i2c-3
python    1187 1659             root 1022u      CHR       89,3       0t0       9430 /dev/i2c-3
python    1187 1659             root 1023u      CHR       89,3       0t0       9430 /dev/i2c-3

pi@Control2:/etc/security $ sudo lsof | grep python | grep '/dev/i2c-3' | wc -l
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
6066

So there are over 6k device opens that has consumed up to the max_user_proc limit.

kizniche, 2017-10-21:
Can you test a potential fix:

Edit ~/Mycodo/mycodo/sensors/tsl2591_sensor.py

Change __init___() at have an additional line at the end so it looks like:

```python
    def __init__(self, address, bus):
        super(TSL2591Sensor, self).__init__()
        self.logger = logging.getLogger(
            "mycodo.sensors.tsl2591_{bus}_{add}".format(bus=bus, add=address))
        self.i2c_address = address
        self.i2c_bus = bus
        self._lux = 0.0
        self.tsl = tsl2591.Tsl2591(i2c_bus=self.i2c_bus, sensor_address=self.i2c_address)
```

And change get_measurement() by removing the first line so it looks like:

```python
    def get_measurement(self):
        """ Gets the TSL2591's lux """
        full, ir = self.tsl.get_full_luminosity()  # read raw values (full spectrum and ir spectrum)
        lux = self.tsl.calculate_lux(full, ir)  # convert raw values to lux
        return lux
```

Save, then restart the daemon with ```sudo service mycodo restart```

Let me know if that fixes the "Too many files open" error and I'll include it in the next release.

Theoi-Meteoroi, 2017-10-21:
Done.   
So far, I'm not seeing the fd leak.  It usually didn't take too long to go to town.

pi@Control2:~ $ sudo lsof | grep python | grep '/dev/i2c-3' | wc -l
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
11
pi@Control2:~ $

I'll let this one run awhile - still awaiting parts from china for that build.

Theoi-Meteoroi, 2017-10-21:
shoulda mentioned my version for others - 5.2.5

Given how fast this bug would present, I think this fixed it.  Holding at 11 fd.  Nice work!

Theoi-Meteoroi, 2017-10-23:
Fix confirmed after longer runtime.

kizniche, 2017-10-23:
Great. Thanks. The fix will be included in the next release.

kizniche, 2017-10-23:
@Ryobidelacroix, @supajay, Were there any other sensors besides the Raspberry Pi CPU temperature in use on your systems when you began seeing your issue?

kizniche, 2017-10-24:
I'm going to close this. If the issue persists after you upgrade to the next release that includes fixes for this error, please open a new issue.

closed 2017-10-24 by kizniche