kizniche/Mycodo #255: An error occurred that prevented Mycodo from being installed

Model76, 2017-06-07:
## Mycodo Issue Report:
  - Specific Mycodo Version:
Latest version as of the 7th of june 2017
 
#### Problem Description
Please list:

  - what were you trying to do
Installing Mycodo

  - specific setup details that are involved
I tried twice with the same result, following the instructions meticulously.
I was using 2017-04-10-raspbian-jessie-lite.
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

I have isolated 2 errors in ~/Mycodo/install/setup.log :
1: 

```
#### Creating InfluxDB database and user
Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: getsockopt: connection refused
Please check your connection settings and ensure 'influxd' is running.
```

And 2, at the end of the log :

```
#### Starting the Mycodo daemon and web server
Restarting apache2 (via systemctl): apache2.service.

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!

Please contact the developer by submitting a bug report at
https://github.com/kizniche/Mycodo/issues with the pertinent
excerpts from the setup log located at Mycodo/install/setup.log

An error occurred. Exiting... 
```

### Steps to Reproduce the issue:
How can this issue be reproduced?

Follow the installation instructions at [https://github.com/kizniche/Mycodo/](https://github.com/kizniche/Mycodo/).
The only step I had added was setting the correct keyboard layout in raspi-config.
 
### Additional Notes

I will now try installing again with an older version of raspian (2016-11-25-raspbian-jessie-lite) that I have laying around from when I successfully installed Mycodo last year. I'll report back if that solves it.

kizniche, 2017-06-07:
Did influxdb install properly during the setup?

Model76, 2017-06-07:
It did say "Incorrect version of influxDB" when it was checking for the correct version, but then it also said it was installing another version. No error or failing was mentioned at this point in the log, so I can only assume that all went well.

Model76, 2017-06-07:
OK, I just finished trying to install on 2016-11-25-raspbian-jessie-lite and that failed, too. What can I do to help you fix this?

kizniche, 2017-06-07:
What version Pi is it? I suspect that influxdb didn't have enough time to start before the connection to create the user/database was attempted. You can just rerun the setup.sh script to see if it can finish, now that influxdb has probably started.

Model76, 2017-06-07:
It's a version 3

Model76, 2017-06-07:
I re-ran setup.sh (with `sudo /bin/bash ./setup.sh`) and am still getting an error after `Restarting apache2....`

kizniche, 2017-06-07:
Did influxdb start without errors after it was installed?

Do you have any errors restarting it? ```service influxdb restart```

Can you connect to influxdb from the terminal? ```influx```

kizniche, 2017-06-07:
Strange issue.

Do you have any peripherals attached to your Pi?

Do you have another Pi to test?

Do you have another SD card to test?

kizniche, 2017-06-07:
So the web server is not starting as well? That means when you go to https://your.Pi.IP.address/ it doesn't connect?

kizniche, 2017-06-07:
just type ```influx``` into the terminal.

kizniche, 2017-06-07:
And just to be sure, this is how you're executing setup.sh, correct?

```sudo /bin/bash ./setup.sh```

Model76, 2017-06-07:
That is correct.

influx says "Connected to http://localhost:8086 version 1.2.4" so seems fine to me.

kizniche, 2017-06-07:
try adding the user and database manually:

```influx -execute "CREATE DATABASE mycodo_db"```

```influx -database mycodo_db -execute "CREATE USER mycodo WITH PASSWORD 'mmdu77sj3nIoiajjs'"```

kizniche, 2017-06-07:
Then restart your Pi and everything should work

kizniche, 2017-06-07:
This is the code for configuring influxdb, from setup.sh:

```bash
printf "\n#### Creating InfluxDB database and user\n"
# Attempt to connect to influxdb 5 times, sleeping 60 seconds every fail
for i in {1..5}; do
influx -execute "CREATE DATABASE mycodo_db" &&
influx -database mycodo_db -execute "CREATE USER mycodo WITH PASSWORD 'mmdu77sj3nIoiajjs'" &&
break || sleep 60; done
```

This should have attempted to execute the commands 5 times, with 60 seconds between attempts. Your log shows only attempting once. Can you paste the output of ```/bin/bash --version``` for me?

Model76, 2017-06-07:
I am afraid it doesn't, though.

As for other Pi's to test with, I have a Pi 1 with sd card, and another Pi 3, but no other sd card for that.

kizniche, 2017-06-07:
What is indicating to you it isn't working?

Model76, 2017-06-07:
I still get Internal Server Error when I try to connect via browser.

The version of bash is 4.3.30(1)-release

kizniche, 2017-06-07:
Do this to start the HTTP server in debug mode to get a detailed error:

```
sudo /etc/init.d/apache2 stop
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/start_flask_ui.py -d
```

Then go to the PI web address and paste the error.

kizniche, 2017-06-07:
I have to leave for an appointment, but I'll be back later to help diagnose the issue further.

Model76, 2017-06-07:
OK, thank you. I also have an appointment later, and will return after that.

Now I am getting Unable to connect - so it seems the apache server is not back up... Or is there a particular port other than 80 I should use for debugging?

kizniche, 2017-06-07:
Debugging should still be using SSL, so https. Is there any output in the terminal when you run start_flask_ui.py?

Model76, 2017-06-07:
Yes, I get:
`Running on https://0.0.0.0:443/`
`Restarting with stat`
`Debugger is active!`
And then a debugger PIN that you probably don't care about.

kizniche, 2017-06-07:
That's indicating the server is running in debug mode, so you should be able to connect with your web browser.

Model76, 2017-06-07:
Yeah, but that was when I got "Unable to connect".

kizniche, 2017-06-07:
Do you happen to have a monitor/keyboard connected to your Pi that you can log into its GUI and open the web browser to https://127.0.0.1?

Model76, 2017-06-07:
Yes, except I installed the lite version of Raspbian.
Do you happen to know a package I can install to easily get the desktop on there?

kizniche, 2017-06-07:
```sudo apt-get install xfce4 lightdm``` should install all the prerequisites to login to xfce.

Model76, 2017-06-07:
Wait... It magically works now! What the hell???

kizniche, 2017-06-07:
Good to hear. Did you do anything in the meantime that would have affected anything?

Model76, 2017-06-07:
Maybe I had to power the pi completely down and not just reboot for some reason?
I did absolutely nothing besides turn the Pi off for 6 hours. So odd.

Model76, 2017-06-07:
Well, I unplugged the and reinserted the keyboard when it was off, but that REALLY shouldn't matter...

I don't mind starting all over again to see if we can localize the error. Would you like that?

kizniche, 2017-06-07:
I'm not really sure how to even begin diagnosing this issue. It appears that the ```for``` loop in the bash script wasn't working properly. It should have attempted to create the user/database 5 times with 60 seconds between attempts. It appears it didn't do this, and just exited the ```for``` loop at the first error. Let me research bash/```for``` before suggesting you should overwrite a working setup.

kizniche, 2017-06-07:
Was this the entire error:

```
#### Creating InfluxDB database and user
Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: getsockopt: connection refused
Please check your connection settings and ensure 'influxd' is running.
```

before showing this?

```
#### Installing and configuring apache2 web server
```

Model76, 2017-06-07:
Just give me the word, I'd like to contribute what little I can. I am a developer, too, so I understand how helpful it can be to have someone test for you.

Model76, 2017-06-07:
The log seems to have been overwritten the second time I ran the setup script.
My offer still stands. I will gladly run through the procedure one more time.

Model76, 2017-06-07:
To clarify: There seemingly was no failure with influxDB the second time the script was run.

It still said
```

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!
```
at the end of the script, though.

Model76, 2017-06-07:
I had a look at your for loop, and it did look a little strange to me. Bear in mind that I am not at all a bash person, and actually never looked at it before today, but shouldn't the loop look more like this:
```
for i in {1..5}; do
 if [influx -execute "CREATE DATABASE mycodo_db" && influx -database mycodo_db -execute "CREATE USER mycodo WITH PASSWORD 'mmdu77sj3nIoiajjs'"]
 then
  break
 fi
 sleep 60
done
```
Please bear with me if I am completely wrong...

kizniche, 2017-06-07:
> The log seems to have been overwritten the second time I ran the setup script.

It should append the log, and not overwrite. I'll correct that in the next release.

> There seemingly was no failure with influxDB the second time the script was run

Yeah, I'm trying to figure this out as well. That error should stop the install script wherever the error occurs. So, it seems it's happening on one of these three commands:

```bash
printf "\n#### Starting the Mycodo daemon and web server\n"
/etc/init.d/apache2 restart
wget --quiet --no-check-certificate -p http://127.0.0.1 -O /dev/null
service mycodo start
```

kizniche, 2017-06-07:
> shouldn't the loop look more like this

There are probably a number of ways to do it, but what's there now is what I've tested to work. Perhaps influxdb changed the exit code for an inability to connect.

kizniche, 2017-06-07:
I believe [this](https://unix.stackexchange.com/a/82610) is where I got the bash loop solution.

Model76, 2017-06-07:
Oh, yes, I see how it is supposed to work now. A bit hacky to use the order of operations like that if you ask me, but hey, I come from C++.

So the assumptions in this code are as follows:
1: Each command will evaluate to `true` on success or `false` on failure.

2: The second command will always succeed when run in rapid succession of the first command
 **OR** the first command can succeed more than once and the second command _can_ succeed when run in rapid succession of the first command.

Are those true?

kizniche, 2017-06-07:
Yes, if the frist command succeeds, the ```&&``` will allow the next to be executed. If the first command succeeds, the second command should succeed. If the second command succeeds, the ```break``` will be executed, leaving the loop. If any command before the break fails, then ```sleep 60``` will be executed and the loop will run again.

kizniche, 2017-06-07:
I'm open to changing the code to something that works better. I just haven't put much effort into testing to notice any situation where it would fail. I'll stop influx and run some tests with this loop isolated.

kizniche, 2017-06-07:
My test was successful when influx wasn't running and the command returned "connection refused", in that it waited and tried again. I suspect for some reason your influx didn't start. Why? I haven't a clue.

Model76, 2017-06-07:
Well, I couldn't leave it alone and just reinstalled the system a third time. I am still getting the same errors. On the up-side I now have a complete install log. Here is the influx part you were asking about earlier:
```
#### Ensuring compatible version of influxdb is installed ####
Incorrect version of InfluxDB installed: v. Installing 1.2.4-1
Selecting previously unselected package influxdb.
(Reading database ... 46730 files and directories currently installed.)
Preparing to unpack influxdb_1.2.4_armhf.deb ...
Unpacking influxdb (1.2.4-1) ...
Setting up influxdb (1.2.4-1) ...
Created symlink from /etc/systemd/system/influxd.service to /lib/systemd/system/influxdb.service.
Created symlink from /etc/systemd/system/multi-user.target.wants/influxdb.service to /lib/systemd/system/influxdb.service.
Processing triggers for man-db (2.7.0.2-5) ...

#### Creating InfluxDB database and user
Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: getsockopt: connection refused
Please check your connection settings and ensure 'influxd' is running.

#### Installing and configuring apache2 web server
```
Is there anything else you would like to see?

kizniche, 2017-06-07:
Strange. Can you create a file with just this in it:

```bash
#!/bin/bash
for i in {1..5}; do
influx -execute "CREATE DATABASE mycodo_db" &&
influx -database mycodo_db -execute "CREATE USER mycodo WITH PASSWORD 'mmdu77sj3nIoiajjs'" &&
printf "success" &&
break || sleep 60; done
```

then make it executable and execute it...

```
chmod +x test.sh
./test.sh
```

Model76, 2017-06-07:
Sure. What is the name of the command line text editor on the Pi, do you know?

Also, influx _is_ running, I just checked.

kizniche, 2017-06-07:
Influx takes a while to start after an install or upgrade, which is why I created the loop to try for up to 5 minutes. However it seems your version of bash or influx isn't behaving properly with the for loop. To properly test the script, you'll have to get the PID from influx (```ps aux | grep influx```) and kill it (```kill -9 xxxx``` where xxxx is the PID), before running the test script.

I believe ```vi``` is the default editor, but I use ```vim``` (```sudo apt-get install vim```). Use the Insert key to start editing. Then Escape to exit insert mode. Then use the colon (```:```), then ```wq``` to write and quit the editor.

Model76, 2017-06-08:
Sorry, my internet went out...

The script returned "success", and that was without killing influx. I will try it now with influx down.

Model76, 2017-06-08:
I tried killing influx, but wasn't able to. Not for long, anyway. Every time I killed it, it just came back up with a different PID, so something must be making sure it runs.

Anyway, I decided to try running the original install script again, and low and behold, it finished without any errors this time. Super strange, since that is the first and only time it has finished without errors, during these 3 separate install attempts, from scratch. It would be nice if we could at least have consistent errors, huh? ;-)

I still have the original failing log, is you are interested, and now I can also pull the log from the successful run. Let me know if you want them.

Do you have any idea what is causing this odd behavior? Are we sure influx is the culprit?

kizniche, 2017-06-08:
> I tried killing influx, but wasn't able to. Not for long, anyway. Every time I killed it, it just came back up with a different PID, so something must be making sure it runs.

Yeah, I forgot to mention you should run the script immediately after killing influx. I also shortened the sleep to 5 seconds for my testing.

> Anyway, I decided to try running the original install script again, and low and behold, it finished without any errors this time.

This is what I expected to happen when I asked you to rerun the setup.sh script. Once influxdb has started, the influx commands should work fine.

> I still have the original failing log, is you are interested, and now I can also pull the log from the successful run. Let me know if you want them.

I don't think they'll be that useful, since I think we've narrowed the issue down to the loop.

> Do you have any idea what is causing this odd behavior? Are we sure influx is the culprit?

I'm pretty sure the issue was influx returning the "connection refused". I'm still not sure why the loop isn't properly working for you but it is for me. I'm 99% sure the issue is the influx loop.

```
kiz@tnccave:~ $ ./test.sh 
Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: getsockopt: connection refused
Please check your connection settings and ensure 'influxd' is running.
test Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: getsockopt: connection refused
Please check your connection settings and ensure 'influxd' is running.
test Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: getsockopt: connection refused
Please check your connection settings and ensure 'influxd' is running.
test Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: getsockopt: connection refused
Please check your connection settings and ensure 'influxd' is running.
test Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: getsockopt: connection refused
Please check your connection settings and ensure 'influxd' is running.
```

rngkay, 2017-06-11:
Thanks kizniche for you help. I have a similar problem and I have managed to solve it with several installs and repairs.

Looking forward for the fix.

kizniche, 2017-06-12:
I don't really know what the issue for @Model76 and @rngkay installs are, as my bash for loop works as expected. I'll put some time this upcoming weekend (or sooner) to attempt to develop a more robust method to detect when influx has started to be able to properly connect and create the user/database.

If anyone has a suggestion, please let me know, but as of now I'm thinking I should put a loop that continually tries to connect to the influx port until it opens, rather than merely attempting to create the user/database.

kizniche, 2017-06-28:
I just tested installing Mycodo on a fresh Raspbian Lite image (6/21/2017) and was able to finally replicate the issue with the proper install of influxdb. I used [```curl``` to test the connection](https://github.com/kizniche/Mycodo/blob/b76d4464f1d16816e71ad4856b8fa2fa91a7b98f/install/setup.sh#L77-L89) and it appears to work for me.

```bash
printf "\n#### Creating InfluxDB database and user\n"
# Attempt to connect to influxdb 5 times, sleeping 60 seconds every fail
for i in {1..5}; do
    # Check if influxdb has successfully started and be connected to
    curl -sL -I localhost:8086/ping > /dev/null &&
    influx -execute "CREATE DATABASE mycodo_db" &&
    influx -database mycodo_db -execute "CREATE USER mycodo WITH PASSWORD 'mmdu77sj3nIoiajjs'" &&
    printf "Influxdb database and user successfully created\n" &&
    break ||
    # Else wait 60 seconds if the influxd port is not accepting connections
    printf "Could not connect to Influxdb. Waiting 60 seconds then trying again...\n" &&
    sleep 60
done
```

I'm going to get one more thing added before making the release. I should have it out within the hour, and I believe that will close this issue.

kizniche, 2017-06-28:
Release [v3.5.36](https://github.com/kizniche/Mycodo/releases/tag/v5.0.36) should fix this issue.

closed 2017-06-28 by kizniche