kizniche/Mycodo #257: TSL2591 Digital Lux Sensor

rngkay, 2017-06-12:
Hi,

Amazing job with this, been following for a while now 👍 

I have the TSL2591 and tried to connect it to Mycodo, but it seems incompatible. I get reading from it through ssh, but not on Mycodo - I assumed it would work if I choose TSL2561.

Any chance you can add support for it? :)

Thanks again!

kizniche, 2017-06-12:
I could, so long as you could test the code I add ;)

rngkay, 2017-06-13:
Gladly :)

kizniche, 2017-06-13:
I'll put together some code for you to test in the coming days.

Could you point me to the code you've already confirmed to work with the device?

itzik16, 2017-06-13:
https://github.com/maxlklaxl/python-tsl2591
I'll just leave it here... @kizniche  👍

rngkay, 2017-06-13:
Yep, that's what I did. Getting a simple reading:

tsl = tsl2591.Tsl2591()  # initialize
full, ir = tsl.get_full_luminosity()  # read raw values (full spectrum and ir spectrum)
lux = tsl.calculate_lux(full, ir)  # convert raw values to lux
print lux, full, ir

>> 10.6488 19 10

kizniche, 2017-06-19:
I just released version 5.0.34 that includes the TSL2591 sensor. Update when you can and let me know if it works.

kizniche, 2017-06-30:
Just checking if you had a chance to test the sensor w Mycodo.

rngkay, 2017-06-30:
Not yet, it takes about two hours to install everything fresh, so I will do it probably this weekend. Thanks!

kizniche, 2017-08-05:
Any update on whether this works or not?

Jhutterer, 2017-08-28:
alright the TSL2591 sensor still does not work out of the box on Mycodo. I am able to use the code in the link above and get a reading in the CLI : 
```
import tsl2591

tsl = tsl2591.Tsl2591()  # initialize
full, ir = tsl.get_full_luminosity()  # read raw values (full spectrum and ir spectrum)
lux = tsl.calculate_lux(full, ir)  # convert raw values to lux
print lux, full, ir
```

I would be more than willing to do any testing needed, just let me know.

kizniche, 2017-08-28:
Can you look for any pertinent errors in the daemon log?

Jhutterer, 2017-08-28:
The below just keeps repeating.

> 2017-08-28 17:24:32,902 - mycodo.sensor_4 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
> 2017-08-28 17:24:47,931 - mycodo.sensors.tsl2591 - ERROR - TSL2591Sensor raised an exception when taking a reading: 2
> 2017-08-28 17:25:02,969 - mycodo.sensors.tsl2591 - ERROR - TSL2591Sensor raised an exception when taking a reading: 2
> 2017-08-28 17:25:18,012 - mycodo.sensors.tsl2591 - ERROR - TSL2591Sensor raised an exception when taking a reading: 2

kizniche, 2017-08-28:
Can you edit ~/Mycodo/mycodo/sensors/tsl2591_sensor.py and change line 67, the word "error" to "exception"

from:

```
            logger.error("{cls} raised an exception when taking a reading: "
                         "{err}".format(cls=type(self).__name__, err=e))
```

To:

```
            logger.error("{cls} raised an exception when taking a reading: "
                         "{err}".format(cls=type(self).__name__, err=e))
```

Save, then restart the daemon:

```sudo service mycodo restart```

Jhutterer, 2017-08-28:
I think above your From: and To: are the same thing so this is what I changed it to : 

```
 logger.exception("{cls} raised an exception when taking a reading: "
                         "{err}".format(cls=type(self).__name__, err=e))
```
Now I get this repeating in daemon log : 

> 2017-08-28 17:37:41,666 - mycodo.sensors.tsl2591 - ERROR - TSL2591Sensor raised an exception when taking a reading: 2
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/tsl2591_sensor.py", line 64, in read
    self._lux = self.get_measurement()
  File "/var/www/mycodo/mycodo/sensors/tsl2591_sensor.py", line 52, in get_measurement
    tsl = tsl2591.Tsl2591(self.i2c_address, "/dev/i2c-" + str(self.i2c_bus))
  File "/home/pi/Downloads/Mycodo/env/src/tsl2591/tsl2591/read_tsl.py", line 69, in __init__
    self.bus = smbus.SMBus(i2c_bus)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/smbus.py", line 56, in __init__
    self.open(bus)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/smbus.py", line 83, in open
    raise IOError(e.errno)
IOError: 2

kizniche, 2017-08-29:
> I think above your From: and To: are the same thing

I edited it directly after commenting to correct the code.

Try changing line 52 to below and see if that fixes it.

```
tsl = tsl2591.Tsl2591(i2c_bus=self.i2c_bus, sensor_address=self.i2c_address)
```

Jhutterer, 2017-08-29:
That did it!

kizniche, 2017-08-29:
Great! I'll incorporate the fix into the next release. Thanks for testing.

closed 2017-08-29 by kizniche