kizniche/Mycodo #258: Error installing latest mycodo version rpi2

reymart081678, 2017-06-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.033
 
#### Problem Description
Please list:
    Installing Mycodo using RPI Jessie with pixel.   kernel 4.4  ( April 2017)
 
### Errors
  copying pi_switch/__init__.py -> build/lib.linux-armv6l-2.7/pi_switch
  running build_ext
  building '_pi_switch' extension
  creating build/temp.linux-armv6l-2.7
  creating build/temp.linux-armv6l-2.7/pi_switch
  creating build/temp.linux-armv6l-2.7/pi_switch/wrapper
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c pi_switch/wrapper/pi_switch.cpp -o build/temp.linux-armv6l-2.7/pi_switch/wrapper/pi_switch.o -Wno-write-strings
  cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
  arm-linux-gnueabihf-gcc: internal compiler error: Killed (program cc1plus)
  Please submit a full bug report,
  with preprocessed source if appropriate.
  See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 4
  
  ----------------------------------------
  Failed building wheel for pi-switch
  Running setup.py clean for pi-switch
Failed to build pi-switch
Installing collected packages: argparse, pi-switch, pyserial, plumbum, rpyc, sht-sensor, tentacle-pi, w1thermsensor, spidev, adafruit-pureio, Adafruit-GPIO, Adafruit-ADS1x15, Adafruit-BMP, Adafruit-TMP, smbus-cffi, MCP342x
  Found existing installation: argparse 1.2.1
    Not uninstalling argparse at /usr/lib/python2.7, outside environment /home/pi/Mycodo/env
  Running setup.py install for pi-switch: started
    Running setup.py install for pi-switch: still running...
    Running setup.py install for pi-switch: finished with status 'error'
    Complete output from command /home/pi/Mycodo/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-GjvD4S/pi-switch/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-Q960Qt-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/Mycodo/env/include/site/python2.7/pi-switch:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv6l-2.7
    creating build/lib.linux-armv6l-2.7/pi_switch
    copying pi_switch/__init__.py -> build/lib.linux-armv6l-2.7/pi_switch
    running build_ext
    building '_pi_switch' extension
    creating build/temp.linux-armv6l-2.7
    creating build/temp.linux-armv6l-2.7/pi_switch
    creating build/temp.linux-armv6l-2.7/pi_switch/wrapper
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c pi_switch/wrapper/pi_switch.cpp -o build/temp.linux-armv6l-2.7/pi_switch/wrapper/pi_switch.o -Wno-write-strings
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    arm-linux-gnueabihf-gcc: internal compiler error: Killed (program cc1plus)
    Please submit a full bug report,
    with preprocessed source if appropriate.
    See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 4
    
    ----------------------------------------
Command "/home/pi/Mycodo/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-GjvD4S/pi-switch/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-Q960Qt-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/Mycodo/env/include/site/python2.7/pi-switch" failed with error code 1 in /tmp/pip-build-GjvD4S/pi-switch/

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!

Please contact the developer by submitting a bug report at
https://github.com/kizniche/Mycodo/issues with the pertinent
excerpts from the setup log located at Mycodo/install/setup.log

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Follow steps to install Mycodo
 2. Call sudo bin/bash ./setup.sh
 3.  Failed as shown above
 4. execute sudo apt-get update and upgrade for RPI 
5. redo install
6. same error 
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-06-19:
It looks like there was a build problem with pi_switch. I'm not sure what could be causing this. It may be something wrong with the pi_switch build (works for me), your compiler, or you may have a bad linux install (corrupt files, bad SD card or sectors, etc.).

This is the also the first issue with pi_switch encountered, so I'm leaning toward you may have had a bad write of Raspbian to yoru SD card (or a bad SD card). Could you test redownloading the raspbian image and/or trying another SD card?

kizniche, 2017-06-19:
Ah, now that you mention it, I think I may have had to do this when installing a module, though I'm not sure it was pi_switch. I'll look into it.

reymart081678, 2017-06-19:
Hi kizniche, okay thank you

kizniche, 2017-06-19:
It looks like [this may be a fix](https://github.com/lexruee/pi-switch-python/issues/13). Can you use that method and let me know if pi_switch will properly compile? Try ```CONF_SWAPSIZE=512``

reymart081678, 2017-06-19:
Hi kizniche, okay i'll do that later.

kizniche, 2017-06-19:
I just released version 5.0.35 that incorporates a check of the swap size and increases if it needs, during install and upgrade. This removes a step that the user has to edit the swap config and restart the dphys-swapfile.

Grab the latest version and try to install it normally and see if it gets past the pi_switch compilation.

kizniche, 2017-06-19:
I went ahead and incorporated this into Mycodo and released the next version, before you had a chance to test because I found this exact swap size edit on my development Pi that I used to install/test the pi_switch dependency.

This may also explain why i've seen the number fo active Mycodo installs drop from ~50 to ~35 when I returned home today, when it has remained at a constant ~50-60 for the past year. I suspect it may have to do with the latest upgrade to using pi_switch breaking some installs/upgrades.

reymart081678, 2017-06-19:
Hi Kizniche,  It continued but when it tries to start the apache , it failed.

1.   Executed the command below, but it did not resolve it.
      sudo ~/Mycodo/env/bin/pip install -r ~/Mycodo/install/requirements.txt --upgrade --force-reinstall --no-deps

2. Check apache logs and got this error.
     [Mon Jun 19 22:23:00.603713 2017] [wsgi:error] [pid 1882:tid 3037312048] [remote      127.0.0.1:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/w1thermsen     sor/core.py", line 294, in load_kernel_modules
    [Mon Jun 19 22:23:00.604665 2017] [wsgi:error] [pid 1882:tid 3037312048] [remote      127.0.0.1:0]     raise KernelModuleLoadError()
    [Mon Jun 19 22:23:00.605255 2017] [wsgi:error] [pid 1882:tid 3037312048] [remote      127.0.0.1:0] KernelModuleLoadError: Cannot load w1 therm kernel modules
pi

3. Execute raspi-config to enable w1
   -- apache error.log
         [Mon Jun 19 22:46:58.837587 2017] [wsgi:error] [pid 652:tid 3011929136] [remote 192.168.0.110:0] OperationalError: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'\\nCREATE TABLE remote (\\n\\tid INTEGER NOT NULL, \\n\\tis_activated BOOLEAN, \\n\\thost TEXT, \\n\\tusername TEXT, \\n\\tpassword_hash TEXT, \\n\\tPRIMARY KEY (id), \\n\\tUNIQUE (id), \\n\\tCHECK (is_activated IN (0, 1))\\n)\\n\\n']

Log entry:
#### Updating crontab entry
Crontab entry does not exist, nothing to do.
no crontab for root
Appending to crontab:
-----------------------------------------------
@reboot /usr/local/bin/pigpiod &-----------------------------------------------
To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh mycodo --remove

#### Creating proper users, directories, and permissions
Adding user `mycodo' to group `gpio' ...
Adding user mycodo to group gpio
Done.
Adding user `mycodo' to group `adm' ...
Adding user mycodo to group adm
Done.
Adding user `mycodo' to group `video' ...
Adding user mycodo to group video
Done.
Adding user `mycodo' to group `dialout' ...
Adding user mycodo to group dialout
Done.

#### Starting the Mycodo daemon and web server
Restarting apache2 (via systemctl): apache2.service.

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!

Please contact the developer by submitting a bug report at
https://github.com/kizniche/Mycodo/issues with the pertinent
excerpts from the setup log located at Mycodo/install/setup.log

An error occurred. Exiting...

reymart081678, 2017-06-19:
Hi Kizniche,   after redoing the installation the system is okay now.
I think the issue is not enabled w1 and disabled pi camera by default.

#### Creating proper users, directories, and permissions
useradd: user 'mycodo' already exists
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `adm'.
The user `mycodo' is already a member of `video'.
The user `mycodo' is already a member of `dialout'.

#### Starting the Mycodo daemon and web server
Restarting apache2 (via systemctl): apache2.service.

***************************************************
** Mycodo successfully installed without errors! **
***************************************************

Go to https://192.168.0.100/, or whatever your Pi's
IP address is, to create an admin user and log in.

kizniche, 2017-06-28:
I just fixed a few issues withthe install procedure and just successfully tested a fresh install of Mycodo with the latest Raspbian Jessie Lite image available.

closed 2017-06-28 by kizniche

kizniche, 2017-08-11:
This issue was actualy just fixed with release v5.1.3. I had overlooked that I had actually not fixed it until now.