kizniche/Mycodo #259: opencv camera error

hamster007, 2017-06-19:
## Mycodo Issue Report:
  - Specific Mycodo Version:5.0.30
 
#### Problem Description
Please list:when taking picture

  - what were you trying to do
  - specific setup details that are involved
 
### Errors

Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 118, in page_camera
    camera_record('photo', mod_camera)
  File "/var/www/mycodo/mycodo/devices/camera.py", line 172, in camera_record
    img_edited = img_orig.copy()
AttributeError: 'NoneType' object has no attribute 'copy'

  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-06-19:
What camera are you using? Does linux recognize it when you plug it in? What are your camera settings in Mycodo?

kizniche, 2017-06-19:
I have a hunch it's one of [these settings](https://github.com/kizniche/Mycodo/blob/67cd79fb3048aaf9b73eb3206764d4c77b54ee4f/mycodo/devices/camera.py#L160-L167) that's causing the issue:

```python
        cap.set(cv2.cv.CV_CAP_PROP_FRAME_WIDTH, settings.width)
        cap.set(cv2.cv.CV_CAP_PROP_FRAME_HEIGHT, settings.height)
        cap.set(cv2.cv.CV_CAP_PROP_EXPOSURE, settings.exposure)
        cap.set(cv2.cv.CV_CAP_PROP_GAIN, settings.gain)
        cap.set(cv2.cv.CV_CAP_PROP_BRIGHTNESS, settings.brightness)
        cap.set(cv2.cv.CV_CAP_PROP_CONTRAST, settings.contrast)
        cap.set(cv2.cv.CV_CAP_PROP_HUE, settings.hue)
        cap.set(cv2.cv.CV_CAP_PROP_SATURATION, settings.saturation)
```

Because when the image is [read a few lines prior](https://github.com/kizniche/Mycodo/blob/67cd79fb3048aaf9b73eb3206764d4c77b54ee4f/mycodo/devices/camera.py#L153-L158) to determine if the camera can be read, there wasn't an error:

```python
        # Check if image can be read
        if not cap.read():
            logger.error(
                "Cannot detect USB camera with device '{dev}'".format(
                    dev=settings.opencv_device))
            return
```

kizniche, 2017-06-19:
Do you have a light on the camera that turns on when an image is acquired (in whatever operating system you can acquire an image with)? If so, do you see that light go on at all during setup or image acquisition?

hamster007, 2017-06-20:
Hi,

Logitech HD webcam.

Was working fine for weeks now this error every time 

Thanks,
Mike

kizniche, 2017-06-20:
If nothing has changed and the error spontaneously started occurring, it's probably either your cam or linux to blame. I would first confirm the cam still works on another computer. Then, try another USB port on the Pi. I would also reboot the Pi if you haven't done so already. If all that fails, make a backup image of your SD card and try a fresh install of Raspbian and Mycodo and see if that fixes it.

kizniche, 2017-06-20:
Have you recently performed an upgrade? Can you check your upgrade log to see if a newer version of opencv was installed during the upgrade?

kizniche, 2017-06-30:
Just checking if there's any new info/developments with this issue. Have you tried a fresh install of Raspbian/Mycodo to rule out software/corruption as the cause? Tried with another computer to verify the camera hardware works?

kizniche, 2017-10-18:
I finally have my only Pi camera free from the long timelapse that it's been used for, so I can begin testing this issue. Sorry, I would have gotten to this earlier if I hadn't had my camera in use.

kizniche, 2017-11-10:
Any new developments on this issue?

kizniche, 2017-11-15:
I just had my long-time running development Pi (version 2) stop working. I took the opportunity to, instead of just flashing a new OS, replace it with a Pi 3 I have. In doing so, I transferred over my Pi camera and a USB camera. The Pi camera works fine, but the USB camera is having issues. Interestingly, it is displaying the same behavior in this issue. Hence, the reopening.

kizniche, 2017-11-30:
I moved my USB camera that was working with Mycodo on my Pi2, back to my Pi2 that I installed a fresh version of Raspbian Stretch and Mycodo, and now it is not working, when it was working previously.

I will try to replicate my setup with an image of Raspbian Jessie and Mycodo to determine if the new Stretch is to blame for the change in opencv behavior.

closed 2017-12-20 by kizniche

gaborlorincz88, 2023-03-01:
Hello Kyle, Im also facing a problem with opencv and Logitech hd webcam.
The camera works as it captures still image via fswebcam (/dev/video0) but opencv states there are not devices found..
I have also installed Motion and i get an image and video stream from there so my Pi is recognising the camera , but opencv does not,
Do you have any idea how to fix this? im using Mycodo version 8.15.6

Thank you in advance :)