kizniche/Mycodo #261: BMP and DS18B20 sensors error

cohe4ko, 2017-06-22:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.35
 
#### Problem Description
I reboot my device every day at 3 AM. Mycodo is used for greenhouse automation. there is 7 DS18B20 sensors and one BMP sensor. 
I did not find any logic but sometimes random DS18B20 and BMP sensors disappear from Mycodo. After cycle of Deactivate/Activate for the sensors - everything becomes ok.

kizniche, 2017-06-22:
Am I interpreting this correctly? When Mycodo starts there are certain sensors not acquiring measurements?

Are there any errors in the daemon log?

cohe4ko, 2017-06-24:
`
2017-06-24 03:05:52,856 - mycodo - INFO - Mycodo daemon v5.0.35 starting
2017-06-24 03:05:53,233 - mycodo - INFO - Anonymous statistics enabled
2017-06-24 03:05:53,673 - mycodo.relay - INFO - Relay 1 setup on pin 24 and turned OFF (OFF=HIGH)
2017-06-24 03:05:53,675 - mycodo.relay - INFO - Relay 2 setup on pin 23 and turned OFF (OFF=HIGH)
2017-06-24 03:05:53,676 - mycodo.relay - INFO - Relay 4 setup on pin 18 and turned OFF (OFF=HIGH)
2017-06-24 03:05:53,678 - mycodo.relay - INFO - Relay 5 setup on pin 16 and turned OFF (OFF=HIGH)
2017-06-24 03:05:53,679 - mycodo.relay - INFO - Relay 6 setup on pin 26 and turned OFF (OFF=HIGH)
2017-06-24 03:05:53,681 - mycodo.relay - INFO - Relay 7 setup on pin 19 and turned OFF (OFF=HIGH)
2017-06-24 03:05:53,682 - mycodo.relay - INFO - Relay 8 setup on pin 25 and turned OFF (OFF=HIGH)
2017-06-24 03:05:53,684 - mycodo.relay - INFO - Relay 9 setup on pin 20 and turned OFF (OFF=HIGH)
2017-06-24 03:05:53,688 - mycodo.relay - INFO - Relay controller activated in 160.6 ms
2017-06-24 03:05:53,877 - mycodo.timer_3 - INFO - Activated in 137.6 ms
2017-06-24 03:05:54,135 - mycodo.timer_5 - INFO - Activated in 171.8 ms
2017-06-24 03:05:54,141 - mycodo - INFO - All activated timer controllers started
2017-06-24 03:05:54,388 - mycodo.sensor_4 - INFO - Activated Conditional (9)
2017-06-24 03:05:54,390 - mycodo.sensor_4 - INFO - Activated Conditional (19)
2017-06-24 03:05:54,649 - mycodo.sensor_4 - INFO - Activated in 323.8 ms
2017-06-24 03:05:55,361 - mycodo.sensor_5 - INFO - Activated in 472.3 ms
2017-06-24 03:05:56,276 - mycodo.sensor_6 - INFO - Activated in 797.6 ms
2017-06-24 03:05:57,882 - mycodo.sensor_7 - INFO - Activated in 1157.2 ms
2017-06-24 03:05:59,399 - mycodo.sensor_8 - INFO - Activated in 1113.3 ms
2017-06-24 03:06:01,745 - mycodo.sensor_9 - INFO - Activated in 1649.9 ms
2017-06-24 03:06:03,520 - mycodo.sensor_12 - INFO - Activated in 1129.6 ms
2017-06-24 03:06:05,898 - mycodo.sensor_13 - INFO - Activated in 1669.1 ms
2017-06-24 03:06:08,507 - mycodo.sensor_14 - INFO - Activated in 2210.6 ms
2017-06-24 03:06:09,838 - mycodo.sensor_15 - INFO - Activated Conditional (14)
2017-06-24 03:06:09,841 - mycodo.sensor_15 - INFO - Activated Conditional (15)
2017-06-24 03:06:09,886 - mycodo.sensor_15 - INFO - Activated Conditional (16)
2017-06-24 03:06:11,578 - mycodo.sensor_15 - INFO - Activated in 2681.2 ms
2017-06-24 03:06:11,597 - mycodo - INFO - All activated sensor controllers started
2017-06-24 03:06:11,598 - mycodo - INFO - All activated PID controllers started
2017-06-24 03:06:15,642 - mycodo.lcd_1 - INFO - Activated in 3529.8 ms
2017-06-24 03:06:19,393 - mycodo.lcd_2 - INFO - Activated in 3237.8 ms
2017-06-24 03:06:19,402 - mycodo - INFO - All activated LCD controllers started
2017-06-24 03:06:19,403 - mycodo - INFO - Mycodo daemon v5.0.35 started in 26.547 seconds
2017-06-24 03:06:19,419 - mycodo - INFO - 44.312 MB ram in use
2017-06-24 03:06:55,366 - mycodo.sensor_4 - ERROR - Could not connect to influxdb. Check that it is running and accepting connections
2017-06-24 03:07:10,775 - mycodo.influxdb - ERROR - No measurement available in the past 22 seconds.
2017-06-24 03:07:12,570 - mycodo.influxdb - ERROR - No measurement available in the past 22 seconds.
2017-06-24 06:35:45,046 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 06:35:45,749 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 06:35:46,019 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 06:35:47,176 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 06:35:48,598 - mycodo.influxdb - ERROR - No measurement available in the past 22 seconds.
2017-06-24 06:35:51,219 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 06:35:53,470 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 06:35:59,897 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 07:35:59,501 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 07:36:00,413 - mycodo.influxdb - ERROR - No measurement available in the past 22 seconds.
2017-06-24 07:36:01,581 - mycodo.influxdb - ERROR - No measurement available in the past 22 seconds.
2017-06-24 07:36:02,239 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 07:36:05,445 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 07:36:06,568 - mycodo.sensors.ds18b20 - ERROR - DS18B20Sensor raised an exception when taking a reading: Sensor is not yet ready to read temperature
2017-06-24 11:22:40,858 - mycodo.sensor_4 - INFO - Activated Conditional (9)
2017-06-24 11:22:40,872 - mycodo.sensor_4 - INFO - Activated Conditional (19)
2017-06-24 11:22:42,061 - mycodo.sensor_4 - INFO - Activated in 1657.0 ms`

cohe4ko, 2017-06-26:
I found some logic. Errors occurred   only with sensors which has conditionals with relays.

kizniche, 2017-06-28:
I don't see the "Deactivated" commands in the log. You're deactivating sensors from the web UI during this log period you pasted?

kizniche, 2017-06-28:
Also, could you try to explain in more detail what exactly is the issue? What does "sensors disappear from Mycodo" mean?

cohe4ko, 2017-06-28:
I mean that there is no sensor data at /live page.

cohe4ko, 2017-06-28:
Last rows - it's deactivating/activating 3 sensors
```
2017-06-28 14:33:47,731 - mycodo - INFO - Mycodo daemon v5.0.35 starting
2017-06-28 14:33:48,088 - mycodo - INFO - Anonymous statistics enabled
2017-06-28 14:33:48,501 - mycodo.relay - INFO - Relay 1 setup on pin 24 and turned OFF (OFF=HIGH)
2017-06-28 14:33:48,504 - mycodo.relay - INFO - Relay 2 setup on pin 23 and turned OFF (OFF=HIGH)
2017-06-28 14:33:48,506 - mycodo.relay - INFO - Relay 4 setup on pin 18 and turned OFF (OFF=HIGH)
2017-06-28 14:33:48,507 - mycodo.relay - INFO - Relay 5 setup on pin 16 and turned OFF (OFF=HIGH)
2017-06-28 14:33:48,509 - mycodo.relay - INFO - Relay 6 setup on pin 26 and turned OFF (OFF=HIGH)
2017-06-28 14:33:48,510 - mycodo.relay - INFO - Relay 7 setup on pin 19 and turned OFF (OFF=HIGH)
2017-06-28 14:33:48,512 - mycodo.relay - INFO - Relay 8 setup on pin 25 and turned OFF (OFF=HIGH)
2017-06-28 14:33:48,514 - mycodo.relay - INFO - Relay 9 setup on pin 20 and turned OFF (OFF=HIGH)
2017-06-28 14:33:48,517 - mycodo.relay - INFO - Relay controller activated in 157.0 ms
2017-06-28 14:33:48,712 - mycodo.timer_3 - INFO - Activated in 141.2 ms
2017-06-28 14:33:49,119 - mycodo.timer_5 - INFO - Activated in 315.0 ms
2017-06-28 14:33:49,128 - mycodo - INFO - All activated timer controllers started
2017-06-28 14:33:49,553 - mycodo.sensor_4 - INFO - Activated Conditional (9)
2017-06-28 14:33:49,554 - mycodo.sensor_4 - INFO - Activated Conditional (19)
2017-06-28 14:33:50,324 - mycodo.sensor_4 - INFO - Activated in 975.5 ms
2017-06-28 14:33:51,776 - mycodo.sensor_5 - INFO - Activated in 1230.3 ms
2017-06-28 14:33:53,375 - mycodo.sensor_6 - INFO - Activated in 1449.4 ms
2017-06-28 14:33:55,952 - mycodo.sensor_7 - INFO - Activated in 2086.5 ms
2017-06-28 14:33:58,556 - mycodo.sensor_8 - INFO - Activated in 2101.4 ms
2017-06-28 14:34:01,397 - mycodo.sensor_9 - INFO - Activated in 2223.1 ms
2017-06-28 14:34:02,236 - mycodo.sensor_12 - INFO - Activated Conditional (20)
2017-06-28 14:34:04,494 - mycodo.sensor_12 - INFO - Activated in 2632.0 ms
2017-06-28 14:34:08,701 - mycodo.sensor_13 - INFO - Activated in 3446.6 ms
2017-06-28 14:34:12,182 - mycodo.sensor_14 - INFO - Activated in 2624.2 ms
2017-06-28 14:34:14,952 - mycodo.sensor_15 - INFO - Activated Conditional (14)
2017-06-28 14:34:14,954 - mycodo.sensor_15 - INFO - Activated Conditional (15)
2017-06-28 14:34:14,992 - mycodo.sensor_15 - INFO - Activated Conditional (16)
2017-06-28 14:34:17,259 - mycodo.sensor_15 - INFO - Activated in 3776.4 ms
2017-06-28 14:34:17,261 - mycodo - INFO - All activated sensor controllers started
2017-06-28 14:34:17,273 - mycodo - INFO - All activated PID controllers started
2017-06-28 14:34:22,816 - mycodo.lcd_1 - INFO - Activated in 4991.4 ms
2017-06-28 14:34:27,618 - mycodo.lcd_2 - INFO - Activated in 4018.2 ms
2017-06-28 14:34:27,642 - mycodo - INFO - All activated LCD controllers started
2017-06-28 14:34:27,647 - mycodo - INFO - Mycodo daemon v5.0.35 started in 39.916 seconds
2017-06-28 14:34:27,789 - mycodo - INFO - 44.492 MB ram in use
2017-06-28 14:34:52,133 - mycodo.sensor_4 - ERROR - Could not connect to influxdb. Check that it is running and accepting connections
2017-06-28 14:35:04,722 - mycodo.sensor_12 - ERROR - Could not connect to influxdb. Check that it is running and accepting connections
2017-06-28 14:35:16,956 - mycodo.sensor_15 - ERROR - Could not connect to influxdb. Check that it is running and accepting connections
2017-06-28 14:38:55,633 - mycodo.sensor_15 - INFO - Activated Conditional (14)
2017-06-28 14:38:55,644 - mycodo.sensor_15 - INFO - Activated Conditional (15)
2017-06-28 14:38:55,668 - mycodo.sensor_15 - INFO - Activated Conditional (16)
2017-06-28 14:38:56,830 - mycodo.sensor_12 - INFO - Activated Conditional (20)
2017-06-28 14:38:57,031 - mycodo.sensor_4 - INFO - Activated Conditional (9)
2017-06-28 14:38:57,036 - mycodo.sensor_4 - INFO - Activated Conditional (19)
2017-06-28 14:38:57,756 - mycodo.sensor_15 - INFO - Activated in 2532.1 ms
2017-06-28 14:38:58,772 - mycodo.sensor_4 - INFO - Activated in 2214.8 ms
2017-06-28 14:38:58,900 - mycodo.sensor_12 - INFO - Activated in 2527.7 ms
```

cohe4ko, 2017-06-28:
[image]

kizniche, 2017-06-30:
What is the period of sensor 4?

kizniche, 2017-06-30:
It appears you are showing a log of the startup that has only a few errors due to influx starting up slowly (cannot connect to influx database), after a few minutes the errors go away because influx has started. Additionally, your screenshot shows only a few minutes difference between the sensor 4 not updating and the actual time (34 vs 36 min). I suspect this issue is only occurring within the first few minutes of starting up and resolves itself once influx has properly started.

Is this the case?

kizniche, 2017-08-10:
This was most likely an issue with influxdb, and should be resolved with the latest version of Mycodo (v.5.1.1)

closed 2017-08-10 by kizniche