kizniche/Mycodo #264: Multiple sensor per relay actuation (enhancement)

reymart081678, 2017-06-30:
Enhancement request:
   - get the average value of multiple sensor ( TEMP,HUMIDITY,etc) then actuate the relay to control it.

kizniche, 2017-06-30:
Are you referring to use with PID controllers?

reymart081678, 2017-06-30:
Hi kizniche, 
Yes

reymart081678, 2017-07-25:
Hi Kizniche,
  Instead of updating PID to acquire multiple sensor reading, Can you incorporate similar function in relay that it can call a script. You can include it in the sensor drop down list and return value of the scripts can be the "average" reading value to emulate sensor readings.
The script can communicate to arduino through USB and the arduino can return the average sensor readings like humidity and/or temperature.

kizniche, 2017-10-18:
I like the idea of being able to use the return value from a script as the input to a PID. That will actually be very easy to incorporate into Mycodo and would add a huge amount of potential uses. Since this appears fairly easy to do, I'll put this as a priority.

kizniche, 2017-10-21:
I wasted several hours developing this feature in a very non-logical way. I completely redesigned it and now the working iteration is a new sensor that allows a command to to be used that returns a value for the measurement storage. This will act like any other sensor and can be used as such. I'll get this update released by the end of the weekend once I've had a little more testing completed.

reymart081678, 2017-10-30:
Thank you so much this will greatly add more potential use of mycodo