kizniche/Mycodo #265: Internal Server Error, Python

RynFlutsch, 2017-07-01:
## Mycodo Issue Report:
  5.0.39 on clean Raspbian Lite @ Raspberry Pi 3B
 
#### Problem Description

Followed Readme.md instal Instructions.
Choosed de_DE-UTF-8. It can be a Problem ?
Installation aborted first time at the final Step (Restart Apache / Mycodo). 
On the second try it worked without errors, but It is not possible to visit the Webinterface.
 
### Errors
`pi@raspberrypi:~ $ less /var/log/apache2/error.log

[Sat Jul 01 09:39:50.076111 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1481, in _run_visitor
[Sat Jul 01 09:39:50.076214 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     **kwargs).traverse_single(element)
[Sat Jul 01 09:39:50.076257 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/sql/visitors.py", line 121, in traverse_single
[Sat Jul 01 09:39:50.077653 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     return meth(obj, **kw)
[Sat Jul 01 09:39:50.077718 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/sql/ddl.py", line 730, in visit_metadata
[Sat Jul 01 09:39:50.080999 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     _is_metadata_operation=True)
[Sat Jul 01 09:39:50.081104 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/sql/visitors.py", line 121, in traverse_single
[Sat Jul 01 09:39:50.081213 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     return meth(obj, **kw)
[Sat Jul 01 09:39:50.081256 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/sql/ddl.py", line 764, in visit_table
[Sat Jul 01 09:39:50.081325 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     include_foreign_key_constraints=include_foreign_key_constraints
[Sat Jul 01 09:39:50.081366 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 914, in execute
[Sat Jul 01 09:39:50.081436 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     return meth(self, multiparams, params)
[Sat Jul 01 09:39:50.081475 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/sql/ddl.py", line 68, in _execute_on_connection
[Sat Jul 01 09:39:50.081542 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     return connection._execute_ddl(self, multiparams, params)
[Sat Jul 01 09:39:50.081610 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 968, in _execute_ddl
[Sat Jul 01 09:39:50.081677 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     compiled
[Sat Jul 01 09:39:50.081716 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1146, in _execute_context
[Sat Jul 01 09:39:50.081782 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     context)
[Sat Jul 01 09:39:50.081821 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1341, in _handle_dbapi_exception
[Sat Jul 01 09:39:50.081888 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     exc_info
[Sat Jul 01 09:39:50.081927 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 202, in raise_from_cause
[Sat Jul 01 09:39:50.082350 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     reraise(type(exception), exception, tb=exc_tb, cause=cause)
[Sat Jul 01 09:39:50.082395 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1139, in _execute_context
[Sat Jul 01 09:39:50.082465 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     context)
[Sat Jul 01 09:39:50.082511 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 450, in do_execute
[Sat Jul 01 09:39:50.083638 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0]     cursor.execute(statement, parameters)
[Sat Jul 01 09:39:50.083826 2017] [wsgi:error] [pid 794:tid 1962865712] [remote 192.168.178.32:0] OperationalError: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'\\nCREATE TABLE remote (\\n\\tid INTEGER NOT NULL, \\n\\tis_activated BOOLEAN, \\n\\thost TEXT, \\n\\tusername TEXT, \\n\\tpassword_hash TEXT, \\n\\tPRIMARY KEY (id), \\n\\tUNIQUE (id), \\n\\tCHECK (is_activated IN (0, 1))\\n)\\n\\n']
[Sat Jul 01 10:05:24.512445 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0] mod_wsgi (pid=794): Target WSGI script '/var/www/mycodo/mycodo_flask.wsgi' cannot be loaded as Python module.
[Sat Jul 01 10:05:24.512593 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0] mod_wsgi (pid=794): Exception occurred processing WSGI script '/var/www/mycodo/mycodo_flask.wsgi'.
[Sat Jul 01 10:05:24.512682 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0] Traceback (most recent call last):
[Sat Jul 01 10:05:24.512821 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]   File "/var/www/mycodo/mycodo_flask.wsgi", line 8, in <module>
[Sat Jul 01 10:05:24.512931 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]     from mycodo.start_flask_ui import app as application
[Sat Jul 01 10:05:24.512977 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]   File "/var/www/mycodo/mycodo/start_flask_ui.py", line 11, in <module>
[Sat Jul 01 10:05:24.513050 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]     app = create_app()  # required by the wsgi config and main()
[Sat Jul 01 10:05:24.513094 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]   File "/var/www/mycodo/mycodo/mycodo_flask/app.py", line 66, in create_app
[Sat Jul 01 10:05:24.513165 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]     register_extensions(app)
[Sat Jul 01 10:05:24.513209 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]   File "/var/www/mycodo/mycodo/mycodo_flask/app.py", line 104, in register_extensions
[Sat Jul 01 10:05:24.513310 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]     db.create_all()
[Sat Jul 01 10:05:24.513350 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_sqlalchemy/__init__.py", line 1007, in create_all
[Sat Jul 01 10:05:24.513420 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]     self._execute_for_all_tables(app, bind, 'create_all')
[Sat Jul 01 10:05:24.513460 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_sqlalchemy/__init__.py", line 999, in _execute_for_all_tables
[Sat Jul 01 10:05:24.513527 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]     op(bind=self.get_engine(app, bind), **extra)
[Sat Jul 01 10:05:24.513566 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/sql/schema.py", line 3695, in create_all
[Sat Jul 01 10:05:24.513635 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]     tables=tables)
[Sat Jul 01 10:05:24.513673 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1856, in _run_visitor
[... cut ...]
[Sat Jul 01 10:05:24.515283 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0]     cursor.execute(statement, parameters)
[Sat Jul 01 10:05:24.515424 2017] [wsgi:error] [pid 794:tid 1954477104] [remote 192.168.178.32:0] OperationalError: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'\\nCREATE TABLE remote (\\n\\tid INTEGER NOT NULL, \\n\\tis_activated BOOLEAN, \\n\\thost TEXT, \\n\\tusername TEXT, \\n\\tpassword_hash TEXT, \\n\\tPRIMARY KEY (id), \\n\\tUNIQUE (id), \\n\\tCHECK (is_activated IN (0, 1))\\n)\\n\\n']
(END)
`

kizniche, 2017-07-01:
"attempt to write a readonly database" 

What are the permissions of ~/Mycodo/databases/mycodo.db?

RynFlutsch, 2017-07-01:
-rw-r--r-- 1 root   root      0 Jun 30 14:24 mycodo.db

kizniche, 2017-07-01:
Can you attach the entire ```~/Mycodo/install/setup.log```?

kizniche, 2017-07-01:
Use this command and then visit the web interface again.

```sudo chown mycodo.mycodo ~/Mycodo/databases/mycodo.db```

RynFlutsch, 2017-07-02:
Thanks, worked great.

Sorry no log.

kizniche, 2017-07-02:
```~/Mycodo/install/setup.log``` Doesn't exist? Can you paste the output from:

```
ls ~/Mycodo/install
```

kizniche, 2017-07-02:
Also note this file would be in a backup if you had performed any upgrades to Mycodo.

```
find /var/Mycodo-backups -name "setup.log"
```

RynFlutsch, 2017-07-03:
Okay i tried only 1 command:
```
cat /Mycodo/install/setup.log
/Mycodo/install/setup.log: Datei oder Verzeichnis nicht gefunden
(not found)
```
Weird: checked with `ls` and i've found it.
Setup.log:
`
### Mycodo installation beginning at 06-30-2017 14:12:37

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap currently set to 100 MB. Changing to 512 MB and restarting
Stopping dphys-swapfile (via systemctl): dphys-swapfile.service.
Starting dphys-swapfile (via systemctl): dphys-swapfile.service.

#### Uninstalling apt version of pip (if installed)
OK   http://mirrordirector.raspbian.org jessie InRelease
OK   http://archive.raspberrypi.org jessie InRelease
OK   http://mirrordirector.raspbian.org jessie/main armhf Packages
OK   http://mirrordirector.raspbian.org jessie/contrib armhf Packages
OK   http://archive.raspberrypi.org jessie/main armhf Packages
OK   http://mirrordirector.raspbian.org jessie/non-free armhf Packages
OK   http://mirrordirector.raspbian.org jessie/rpi armhf Packages
OK   http://archive.raspberrypi.org jessie/ui armhf Packages
Ign http://archive.raspberrypi.org jessie/main Translation-de_DE
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-de_DE
Ign http://archive.raspberrypi.org jessie/main Translation-de
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-de
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-de_DE
Ign http://mirrordirector.raspbian.org jessie/main Translation-de
Ign http://archive.raspberrypi.org jessie/ui Translation-de_DE
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-de_DE
Ign http://archive.raspberrypi.org jessie/ui Translation-de
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-de
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-de_DE
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-de
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Paketlisten werden gelesen...
Paketlisten werden gelesen...
Abhängigkeitsbaum wird aufgebaut....
Statusinformationen werden eingelesen....
Paket »python-pip« ist nicht installiert, wird also auch nicht entfernt.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

#### Installing prerequisite apt packages and update pip
OK   http://archive.raspberrypi.org jessie InRelease
OK   http://mirrordirector.raspbian.org jessie InRelease
OK   http://archive.raspberrypi.org jessie/main armhf Packages
OK   http://mirrordirector.raspbian.org jessie/main armhf Packages
OK   http://mirrordirector.raspbian.org jessie/contrib armhf Packages
OK   http://mirrordirector.raspbian.org jessie/non-free armhf Packages
OK   http://archive.raspberrypi.org jessie/ui armhf Packages
OK   http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-de_DE
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-de
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-de_DE
Ign http://mirrordirector.raspbian.org jessie/main Translation-de
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-de_DE
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-de
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://archive.raspberrypi.org jessie/main Translation-de_DE
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-de_DE
[... 1405 lines cut ...]
Fertig.
Füge Benutzer »mycodo« der Gruppe »dialout« hinzu ...
Benutzer mycodo wird zur Gruppe dialout hinzugefügt.
Fertig.

#### Restarting the Mycodo web server
Restarting apache2 (via systemctl): apache2.service.

#### Creating Mycodo database if it doesn't exist

#### Restarting the Mycodo daemon
Job for mycodo.service failed. See 'systemctl status mycodo.service' and 'journalctl -xn' for details.

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!

Please contact the developer by submitting a bug report at
https://github.com/kizniche/Mycodo/issues with the pertinent
excerpts from the setup log located at Mycodo/install/setup.log

An error occurred. Exiting...
`

kizniche, 2017-07-03:
You forgot the tilde (```~```) that indicates your user home directory. ```~``` is equivalent to ```/home/user```, so ```~/Mycodo``` in bash translates to ```/home/user/Mycodo```.

kizniche, 2017-07-03:
Thanks for the log. The error occurred at the very end during the restart of the daemon. I'm not sure why your database had the incorrect permissions. I'll investigate these issues.

kizniche, 2017-07-03:
I just found one of the issues. The database is created after initialization (which sets proper permissions). I need to add another initialization command after the database creation.

kizniche, 2017-07-04:
This install issue should be resolved in the latest release.

closed 2017-07-04 by kizniche