kizniche/Mycodo #266: Daemon Crashes Unexpectedly

supajay, 2017-07-01:
## Mycodo Issue Report:
  - Specific Mycodo Version:
5.0.39
 
#### Problem Description
Daemon Crashes unexpectedly
as a side note, the setpoint control via method does not work in this release

normal running of mycodo
 
### Errors
Extract from daemon log:

2017-07-01 14:19:49,786 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:19:49,790 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:20:04,803 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:20:19,809 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:20:34,819 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:20:34,843 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:20:49,876 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:21:04,958 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:21:07,036 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:21:20,050 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:21:20,059 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:21:35,053 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:21:50,145 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:05,168 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:05,171 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:22:20,174 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:35,186 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:37,094 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:22:50,216 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:22:50,248 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:23:05,298 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:23:20,345 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:23:35,348 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:23:35,364 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:23:50,376 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:24:05,475 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:24:07,207 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:24:20,563 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'
2017-07-01 14:24:20,591 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-01 14:24:35,593 - mycodo.sensors.raspi - ERROR - RaspberryPiCPUTemp.get_measurement() method raised IOError: [Errno 24] Too many open files: '/sys/class/thermal/thermal_zone0/temp'

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1.running mycodo results in this issue once it has run for approximately 2 hours
 
### Additional Notes
n/a

kizniche, 2017-07-01:
This is a duplicate of #254

Since you're now the second to experience this issue, I'll return to investigating it. Can you please post in the other thread from here on out? I'll ask you some questions about your system/setup a little later when I have more time, but in the meantime, can you post in #254 what version Raspberry Pi, anything you enabled after a fresh install of Mycodo? Thanks.

supajay, 2017-07-01:
thanks I have now logged the error in the same thread and also added some
more details from the Daemon log that shows something different now but
also related to no receiving any sensor readings.

the daemon crashes after some time

it seems that the issue is the PID controller since the sensor readings are
still being logged it seems.

on a side note the live graphs are also not autoupdating.

is there anyway to revert to a previous build?

thanks Jaryd

kizniche, 2017-07-03:
I'm going to focus on the method not working with this issue. This seems like a top priority right now, but I'm not sure when in the next few days I'll be able to work on it.

kizniche, 2017-07-04:
The method issue should be resolved in the latest release.

closed 2017-07-04 by kizniche