kizniche/Mycodo #269: Ramp Soak PID feature request

eap904, 2017-07-08:
Not sure if there is a way to do this currently, but wanting the ability to ramp/soak with the PID controller. 

This means making a setpoint, and once it hits that, it will maintain for a specified period of time before shutting off.

kizniche, 2017-07-08:
You can do this already, but it's not immediately obvious. Create a PID method (duration) that will be set at your setpoint for a specific duration. Then select/save that method on the PID. Now, when you activate that PID, it will regulate to the method's setpoint until the duration ends, then stop.

To be safe, I believe you must also set the actual PID setpoint very low (below activation threshold) so when the method ends and falls back to the PID setpoint, it doesn't activate. I can't exactly remember the behavior when the method ends, so you'll have to test to truly find out.

eap904, 2017-07-08:
I do see a way to set a duration as you describe, but I do not know what the duration will be to get to that temp

kizniche, 2017-07-08:
If the time to setpoint is unknown, and you need a specific duration until it turns off after hitting the setpoint, then I don't think it is possible. However, if you can test and find the time to setpoint is consistent, then you can account for that time.

Could you describe your desired feature a little more, if it's something you would like to see added?

eap904, 2017-07-08:
The logic seems easy, but I don't know the programming. Have a setpoint, turn heating on until that setpoint is reached, then use the PID to maintain that temp for a specified duration.

From an auberins PID description: (http://www.auberins.com/index.php?main_page=product_info&cPath=1&products_id=4)

"This controller is for applications where it is desirable to have adjusted set point automatically over time. Up to 30 steps can be programmed to A) ramp the temperature up and down at controlled rate, B) maintain the temperature for a specific period time, C) turn on/off the relays (alarm relay) after process reach a specific step, D) jump (for loops), run, hold or stop after process reaches a specific step."

kizniche, 2017-10-18:
I've been planning how to do this recently, and encapsulate all the desired features (from your link): A) ramp the temperature up and down at controlled rate, B) maintain the temperature for a specific period time, C) turn on/off the relays (alarm relay) after process reach a specific step, D) jump (for loops), run, hold or stop after process reaches a specific step. It appears I'll have to create a new type of PID method (ramp/soak) to accomplish this and it will be more involved than I anticipated. I'm not sure when I'll be able to get to tackling this, but it seems like a very useful feature.

eap904, 2017-10-19:
That sounds great Kyle.

I am attempting to use it to control a water heater element in a 55 gal
barrel to maintain 210F or so in a sealed barrel. This way instead of
losing water and energy exhausting the steam, it just keeps it heated to
the right temp.

Using the rPi as a controller works ok, but the PID option is obviously the
best. I have a contactor controlled by a relay from the rPi, and I have
read that for PID the best option is a SSR.

Currently trying to recover data from a non-booting pi (probably corrupt
SD due to power loss). Aside from a UPS, have any tips for avoiding SD card
corruption? Hopefully I can still get the timestamp before power loss from
the mycodo logs on the card.

Keep up the great work, hopefully I will one day be able to contribute. I'm
still seeking an idiot proof way to connect and run sensors all over the
place, using a breadboard with dupont pin connectors is really a PITA and
seems fragile, even with a fancy crimp tool I bought

-Eric

jacopienaar, 2019-11-01:
I see this is an old issue but I am also looking for the same function. I posted on the forum about something similar and Kyle gave a bit of a workaround - never really got it working properlyas the system shuts off when the set point is reached instead of starting the timer to shut it off after predetermined period

chakahamilton, 2021-05-23:
> The logic seems easy, but I don't know the programming. Have a setpoint, turn heating on until that setpoint is reached, then use the PID to maintain that temp for a specified duration.
>
> From an [...]

I just came across this issue, I am four years in the future. I am familiar with the mentioned PID ramp and soak functions. It's a bit tedious to program which is why I decided to write a console program to help program in the steps AKA do the equations for each step. I made something similar in Excel which was fairly straightforward. I've built beer brewing machines around the Aubur instruments syl-2352P.