kizniche/Mycodo #270: Updating from 5.0.39 -> 5.0.45 broke permissions

lucasvickers, 2017-07-09:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.45
 
#### Problem Description
Please list:
  - after update, when I select update again (to verify version), I see errors on permissions. 
 
### Errors
  - Last error in update from 0.39 -> 0.45

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

lucasvickers, 2017-07-09:
Apologies for incomplete original post, it got submitted by accident.

kizniche, 2017-07-09:
> No such file or directory

Not sure how this happened. 

> Permission denied

Try this to fix the permissions:

```sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions```

kizniche, 2017-07-09:
Try running the post upgrade script and see if it fixes the issues:

```sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh```

lucasvickers, 2017-07-09:
Those steps corrected it.

kizniche, 2017-07-09:
Can you take the excerpt from the upgrade log (entire upgrade part from start to end of the last upgrade), zip the text file, and attach it here for me to review for issues?

lucasvickers, 2017-07-09:
FYI replicated it on my second setup.

[mycodoupgrade.log.zip](https://github.com/kizniche/Mycodo/files/1133898/mycodoupgrade.log.zip)

kizniche, 2017-07-09:
Okay, I figured it out. The previous versions aren't running the new permissions command of the latest releases. I'll have to push a quick update/release when I get home to fix this issue. Thanks.