kizniche/Mycodo #271: Just tried to Upgrade - Box has been dying on and off for days now

jeffyboz, 2017-07-10:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 *.45
#### Problem Description
Please list:

  - what were you trying to do
Check upgrade

 
Just tried to Upgrade - Box has been dying on and off for days now

Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/admin_routes.py", line 150, in admin_upgrade
    with open(INSTALL_DIRECTORY + '/.upgrade', 'w') as f:
IOError: [Errno 13] Permission denied: '/home/pi/Mycodo/mycodo/../.upgrade'

kizniche, 2017-07-10:
I see from your other issue you're running 5.0.46, which is the version right before the current (5.0.47) which I fixed this issue. You happened to upgrade before I fixed the issue. Use the following command and let me know if it fixes the issue.

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_mycodo_release.sh
```

jeffyboz, 2017-07-10:
Will do - on the road now

kizniche, 2017-07-10:
Also commands to try is that doesn't fix it:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions
```

reboot

If it is still not working, then edit ```~/Mycodo/mycodo/config.py``` and change ```MYCODO_VERSION = '5.0.47'``` to ```MYCODO_VERSION = '5.0.46'```, save, then issue these commands:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_mycodo_release.sh
```

jeffyboz, 2017-07-10:
> -------- Original Message --------
> Local Time: July 10, 2017 1:56 PM
> UTC Time: July 10, 2017 5:56 PM

***Running Now****

jeffyboz, 2017-07-11:
> -------- Original Message --------
> Local Time: July 10, 2017 7:20 PM
> UTC Time: July 10, 2017 11:20 PM

It wasn't running this morning - tried your last suggestion of reverting the distro and then re-running the install. I'll let you know how it goes.

closed 2017-10-24 by kizniche