kizniche/Mycodo #272: Can save pid change

jeffyboz, 2017-07-10:
## Mycodo Issue Report:
  - Specific Mycodo Version:
5.0.46
 
#### Problem Description
Please list:
Can't save max on max off for PID ID 1

 
### Errors
[image]

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
PID - change rise min on or max on or min off, then save.

 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-08-06:
I wasn't able to reproduce this. This may be a lingering effect of an install gone wrong. Try this command to make your database writable again:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
```

kizniche, 2017-08-09:
I believe this issue was fixed by the command in the previous comment. Please open if the issue persists.

closed 2017-08-09 by kizniche