kizniche/Mycodo #273: stop sensor

gadem35, 2017-07-11:
2017-07-11 17:50:24,828 - mycodo.sensors.am2315 - ERROR - All measurements returned failed CRC
2017-07-11 17:50:24,829 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-11 17:50:28,872 - mycodo.influxdb - ERROR - No measurement available in the past 22 seconds.
2017-07-11 17:50:29,574 - mycodo.influxdb - ERROR - No measurement available in the past 22 seconds.
2017-07-11 17:51:25,046 - mycodo.sensors.am2315 - ERROR - All measurements returned failed CRC
2017-07-11 17:51:29,266 - mycodo.influxdb - ERROR - No measurement available in the past 22 seconds.
2017-07-11 17:51:29,888 - mycodo.influxdb - ERROR - No measurement available in the past 22 seconds.
2017-07-11 17:51:55,242 - mycodo.sensors.am2315 - ERROR - All measurements returned failed CRC

sensors are stop every time.i use dht22 and am2315.same issue all sensors.help me pls

kizniche, 2017-07-11:
These errors are only showing an AM2315 sensor issue. Make sure your sensor is properly connected.

Does any other test code retrieve a measurement? If not, then it's not Mycodo that's the issue.

kizniche, 2017-07-12:
The AM2315 sometimes hangs in certain configurations, and requires it to be disconnected and reconnected (power wire).

This is similar to what happens with the DHT22. With the DHT22 I solved this issue by enabling a relay to be selected in the sensor configuration that is connected to the sensor power pin. This is configured as a relay (on the relay page) but it is in fact a transistor (which acts similar to a relay). If the DHT22 has 3 consecutive misreads, the power is turned off for 2 seconds by actuation this relay, then turned back on. This successfully resets the sensor and allows it to be read again. I'll add this functionality to the AM2315 and that should fix this issue for good.

You'll have to wire a transistor to your AM2315 power wire and have a Pi pin wired to trigger it, and then add that pin as a relay in Mycodo, then set that relay as the power relay in the sensor configuration.

I'll see if I can push an update/release tonight that incorporates this feature. You should see a reference to the commit show up in this issue when it happens.

kizniche, 2017-07-12:
I just pushed a commit that adds the "Power Relay" option to the AM2315 sensor options, and released version 5.0.48. 

Whether this was your issue or not remains to be seen, but if unplugging and replugging the power of the sensor fixes the issue, then this fix will most likely keep your sensor working from now on, so long as you use the new feature to automatically power-cycle it when 3 consecutive readings fail.

Ericktz, 2017-07-13:
Could you explain step by step how to use this feature? Or some example with images or text if possible? For both cases, with normal relays or with transistors please ...

I think that in a few cases, the same problem occurs with DHT11, it could be a solution for that sensor to add this feature too

thank you very much

kizniche, 2017-07-13:
Ah, you're right, I don't have this added for the DHT11. I had built in part of the code for it to work for the DHT11, but didn't fully implement it. I'll get to that later today or this weekend. I also need to add that option description to the [manual](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.md).

As for how to use it, I've had success with the DHT22 using an NPN transistor. Here's a diagram for boh the NPN and PNP transistor circuits.

[image]

In this diagram, the load will be the sensor, and the transistor will be switching on/off the ground connection to the sensor, signaled by the Raspberry Pi. I'm not home to check what NPN transistor I used or the resistor (R1) value, but a 10k resistor should be sufficient. The Pi pin you connect to the transistor will need to be set up as a new relay on the relay page. This will be the relay you set as the "Power Relay" for the sensor. For the relay settings, you can set the start state to "Off" so the sensor is off when the Pi starts, as the relay will be turned on by the sensor controller when the sensor is activated.

You can also use a relay to power the sensor. In this case, use the relay to break the power line or ground line by connecting it to the switched leads. That is, remove the power line from the sensor and connect it to one of the switched leads of the relay, then connect from the other switched lead to the sensor power pin. This will allow the relay to open and close the power or the ground (whichever you choose to use) circuit to the sensor power or ground, respectively.

Let me know if you need any more clarification. I can get you the transistor part later when I get home.

kizniche, 2017-07-13:
Just checked, the NPN transistor is the [BC548](https://en.wikipedia.org/wiki/BC548).

closed 2017-10-24 by kizniche