kizniche/Mycodo #274: Suggestion: Gauges for temp and humidity at the live page

pewi81, 2017-07-17:
This might seem like a cosmetic suggestion but as I use a 7" hdmi touchscreen mounted at my orchid vivarium it would be an improvement to be able to toggle on/off gauges for sensor data at the live page. The issue is that its hard to read the data unless I get really close to the screen. Gauges fix this problem. Its easy to use for example a Google Gauge or Thingspeak but it would be nice to have it integrated into Mycodo.

Regards
Pete

kizniche, 2017-07-18:
Great idea. This would be a nice addition. I'll see if I can roll out something basic in the near future.

molexuse, 2017-08-18:
Great! It will be really convenient, There is such an option, i used it for sensors on esp266
https://github.com/Freeboard/freeboard

pewi81, 2017-09-15:
Is there any plans for this?

kizniche, 2017-09-16:
I'm working on something this weekend I'll try to release by Sunday.

kizniche, 2017-09-16:
Progress:

[image]

kizniche, 2017-09-16:
I'm currently working with this gauge: https://www.highcharts.com/demo/gauge-solid/dark-unica
Although there are several ways to configure it. The menu on the left demonstrates a few different styles. 

What options are desirable? I currently have max_measure_age (will only display a measurement if it was acquired in the past x seconds), min and max (determines lowest and highest measurements on the gauge). I'm going to add the ability to add and remove color zones, so you can define as many of any color at any point along the gauge. Knowing what peoples' uses/desires are can help me add the appropriate options if you have something specific you would like to see.

pewi81, 2017-09-16:
That looks great! I'm satisfied to be able to see exactly what you described above; a value acquired from the past seconds, a max & min on the gauge, and the value itself in the middle of the gauge like pictured above! Color zones is also a good idea, it gives a better and quicker view from a distance!

kizniche, 2017-09-16:
Second gauge style added

[image]

pewi81, 2017-09-17:
Very nice!

kizniche, 2017-09-17:
Colors and ranges are now editable, but until I add a way to add/remove colors, the number is currently stuck at 4.

[image]

kizniche, 2017-09-17:
I'm going to release it as it currently is and add the ability to add/remove colors soon.

molexuse, 2017-09-21:
I change gauge to solid and Error 500: Internal Server Error

Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 350, in page_graph
    flaskutils.graph_mod(form_mod_gauge, request.form)
  File "/var/www/mycodo/mycodo/flaskutils.py", line 1026, in graph_mod
    colors_hex[i]['stop'],
KeyError: 'stop'

kizniche, 2017-09-22:
Did you modify flaskutils.py? I ask because the line numbers your exception references does not align with what I pushed for the latest release.

kizniche, 2017-09-22:
I just released v5.2.1 that incorporates a fix. It's untested, so please let me know if ti works after you perform an upgrade.

molexuse, 2017-09-22:
Hi, i don`t modify flaskutils.py.
Just upgrade, and solid works fine!

kizniche, 2017-09-22:
Great. Thanks! I'm going to close this issue now that gauges are fully implemented. I'll be adding new gauges, options, and measurement options (relay, PID, etc.) soon.

closed 2017-09-22 by kizniche